rusefi/rusefi #4671: How long till we want to execute code from QSPI

rusefillc, 2022-10-17:
Ethernet not fitting on flash makes me wonder how long till we play with code execution from QSPI as supported by F42x and newer (not F40x)

https://github.com/rusefi/rusefi/commit/c7b13337958d696574e2345aa99891d7da3133da#diff-cc7e02cb67a42c85073313459e81e05bce9bfbfc19723b7e1cf22da3c44ffc4d

STM32F746G-DISCO is $70 shipped at https://www.newark.com/stmicroelectronics/stm32f746g-disco/dev-board-cortex-m7-discovery/dp/49Y1253

rusefillc, 2022-10-17:
@dron0gus https://community.st.com/s/question/0D50X00009XkWENSA3/how-to-execute-code-from-external-flash-on-stm32f407 :)

mck1117, 2022-10-17:
Execution from qspi also has some pretty serious performance impacts with regards to bus load instead of internal flash.

rusefillc, 2022-10-17:
> I've used ethernet on my F429 nucleo.
https://github.com/rusefi/rusefi/issues/4671#issuecomment-1280243713

rusefillc, 2022-10-17:
> Execution from qspi also has some pretty serious performance impacts with regards to bus load instead of internal flash.

Copy to RAM? (yes, external RAM)

mck1117, 2022-10-17:
> Copy to RAM? (yes, external RAM)

Potentially better latency, but same bus contention problem. The STM32 has an extra little pipe for instruction fetch from flash, but not to anywhere else.

mck1117, 2022-10-17:
@rusefillc #4163 is maybe a lowish hanging fruit with huge flash memory savings