rusefi/rusefi #4688: GM ETB bench test in NJ

rusefillc, 2022-10-20:
Trying to get GM ETB with MRE in NJ

at the moment use master, MRE 0.6.0 with C54 removed

```
2022-10-20_14_53_50_590: EngineState: WARNING: Injector 2 fault: open_load 
2022-10-20_14_53_50_591: EngineState: WARNING: Injector 3 fault: open_load 
2022-10-20_14_53_58_648: EngineState: ---- SENT ch 0 ----
2022-10-20_14_53_58_648: EngineState: Unit time 272 timer ticks
2022-10-20_14_53_58_648: EngineState: Total pulses 1052358
2022-10-20_14_53_58_648: EngineState: Last fast msg Status 0x0 Sig0 0x000 Sig1 0x000
2022-10-20_14_53_58_649: EngineState: Slow channels:
2022-10-20_14_53_58_654: EngineState: Restarts 0
2022-10-20_14_53_58_655: EngineState: Interval errors 50 short, 862 long
2022-10-20_14_53_58_657: EngineState: Total frames 115302 with crc erorr 115302
2022-10-20_14_53_58_659: EngineState: Sync errors 50
2022-10-20_14_53_58_660: EngineState: --------------------
2022-10-20_14_53_58_661: EngineState: confirmation_sentinfo:8

```

dron0gus, 2022-10-20:
Good output:
```
2022-10-20_22_04_34_104: EngineState: confirmation_sentinfo:8
2022-10-20_22_05_08_337: EngineState: ---- SENT ch 0 ----
2022-10-20_22_05_08_338: EngineState: Unit time 275 timer ticks
2022-10-20_22_05_08_339: EngineState: Total pulses 65666
2022-10-20_22_05_08_339: EngineState: Last fast msg Status 0x0 Sig0 0x0C6 Sig1 0x0CB
2022-10-20_22_05_08_340: EngineState: Slow channels:
2022-10-20_22_05_08_340: EngineState: Restarts 1
2022-10-20_22_05_08_340: EngineState: Interval errors 21 short, 316 long
2022-10-20_22_05_08_340: EngineState: Total frames 6709 with crc erorr 6709
2022-10-20_22_05_08_341: EngineState: Sync errors 21
2022-10-20_22_05_08_341: EngineState: --------------------
```

dron0gus, 2022-10-21:
Definitely good output from fuel pressure sensor:
```
 2022-10-21_11_41_16_865: EngineState: ---- SENT ch 0 ----
2022-10-21_11_41_16_866: EngineState: Unit time 218 timer ticks
2022-10-21_11_41_16_866: EngineState: Total pulses 121848
2022-10-21_11_41_16_866: EngineState: Last fast msg Status 0xC Sig0 0x0C5 Sig1 0x0CB
2022-10-21_11_41_16_867: EngineState: Slow channels:
2022-10-21_11_41_16_867: EngineState:  ID 22: 714
2022-10-21_11_41_16_867: EngineState:  ID 16: 715
2022-10-21_11_41_16_868: EngineState: Restarts 0
2022-10-21_11_41_16_868: EngineState: Interval errors 119 short, 953 long
2022-10-21_11_41_16_868: EngineState: Total frames 11757 with crc erorr 38
2022-10-21_11_41_16_869: EngineState: Sync errors 119
2022-10-21_11_41_16_869: EngineState: --------------------
2022-10-21_11_41_16_870: EngineState: confirmation_sentinfo:8
```

dron0gus, 2022-10-21:
Fuel pressure waveform
[image]

dron0gus, 2022-10-21:
ETB waveform
[image]

dron0gus, 2022-10-21:
So this looks like "another" CRC. J7216 has defines "New" and "Legacy" CRC calculation methods.
Which one do we have implemented now?

dron0gus, 2022-10-22:
New CRC calculation compatible with GM ETB is implemented in https://github.com/rusefi/rusefi/pull/4692

rusefillc, 2022-10-22:
@dron0gus do you have CRC auto-detection or do you want a new configuration setting for CRC mode choice?

rusefillc, 2022-10-23:
idle position

```
2022-10-22_21_04_23_873: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_23_873: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_23_874: EngineState: Total pulses 815553
2022-10-22_21_04_23_875: EngineState: Last fast msg Status 0x0 Sig0 0xADC Sig1 0x523
2022-10-22_21_04_23_879: EngineState: Restarts 0
2022-10-22_21_04_23_880: EngineState: Interval errors 91 short, 1223 long
2022-10-22_21_04_23_882: EngineState: Total frames 88313 with crc error 53 (0.060013816 %)
2022-10-22_21_04_23_884: EngineState: Sync errors 91
2022-10-22_21_04_23_888: EngineState: --------------------
2022-10-22_21_04_23_889: EngineState: confirmation_sentinfo:8
2022-10-22_21_04_25_184: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_25_184: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_25_184: EngineState: Total pulses 833835
2022-10-22_21_04_25_185: EngineState: Last fast msg Status 0x0 Sig0 0xADD Sig1 0x522
2022-10-22_21_04_25_187: EngineState: Restarts 0
2022-10-22_21_04_25_191: EngineState: Interval errors 95 short, 1256 long
2022-10-22_21_04_25_192: EngineState: Total frames 90280 with crc error 54 (0.059813912 %)
2022-10-22_21_04_25_193: EngineState: Sync errors 95
2022-10-22_21_04_25_194: EngineState: --------------------
2022-10-22_21_04_25_195: EngineState: confirmation_sentinfo:8
2022-10-22_21_04_27_039: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_27_039: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_27_040: EngineState: Total pulses 859947
2022-10-22_21_04_27_040: EngineState: Last fast msg Status 0x0 Sig0 0xADC Sig1 0x523
2022-10-22_21_04_27_040: EngineState: Restarts 0
2022-10-22_21_04_27_042: EngineState: Interval errors 96 short, 1295 long
2022-10-22_21_04_27_046: EngineState: Total frames 93109 with crc error 55 (0.059070548 %)
2022-10-22_21_04_27_047: EngineState: Sync errors 96
2022-10-22_21_04_27_050: EngineState: --------------------
2022-10-22_21_04_27_051: EngineState: confirmation_sentinfo:8
2022-10-22_21_04_28_453: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_28_453: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_28_454: EngineState: Total pulses 879859
2022-10-22_21_04_28_460: EngineState: Last fast msg Status 0x0 Sig0 0xADC Sig1 0x523
2022-10-22_21_04_28_461: EngineState: Restarts 0
2022-10-22_21_04_28_463: EngineState: Interval errors 97 short, 1335 long
2022-10-22_21_04_28_464: EngineState: Total frames 95248 with crc error 55 (0.057743996 %)
2022-10-22_21_04_28_465: EngineState: Sync errors 97
2022-10-22_21_04_28_466: EngineState: --------------------
2022-10-22_21_04_28_468: EngineState: confirmation_sentinfo:8
2022-10-22_21_04_29_542: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_29_543: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_29_543: EngineState: Total pulses 895191
2022-10-22_21_04_29_544: EngineState: Last fast msg Status 0x0 Sig0 0xADC Sig1 0x523
2022-10-22_21_04_29_550: EngineState: Restarts 0
2022-10-22_21_04_29_550: EngineState: Interval errors 100 short, 1364 long
2022-10-22_21_04_29_552: EngineState: Total frames 96897 with crc error 57 (0.058825352 %)
2022-10-22_21_04_29_553: EngineState: Sync errors 100
2022-10-22_21_04_29_554: EngineState: --------------------
2022-10-22_21_04_29_556: EngineState: confirmation_sentinfo:8
2022-10-22_21_04_30_631: EngineState: ---- SENT ch 0 ----
2022-10-22_21_04_30_631: EngineState: Unit time 267 timer ticks
2022-10-22_21_04_30_631: EngineState: Total pulses 910502
2022-10-22_21_04_30_631: EngineState: Last fast msg Status 0x0 Sig0 0xADC Sig1 0x523
2022-10-22_21_04_30_632: EngineState: Restarts 0
2022-10-22_21_04_30_632: EngineState: Interval errors 104 short, 1389 long
2022-10-22_21_04_30_636: EngineState: Total frames 98550 with crc error 60 (0.060882800 %)
2022-10-22_21_04_30_637: EngineState: Sync errors 104
2022-10-22_21_04_30_638: EngineState: --------------------
2022-10-22_21_04_30_639: EngineState: confirmation_sentinfo:8
```

ETB held CLOSED
```
2022-10-22_21_05_04_776: EngineState: ---- SENT ch 0 ----
2022-10-22_21_05_04_776: EngineState: Unit time 267 timer ticks
2022-10-22_21_05_04_776: EngineState: Total pulses 1390627
[... 114 lines cut ...]
2022-10-22_21_05_54_950: EngineState: Sync errors 236
2022-10-22_21_05_54_956: EngineState: --------------------
2022-10-22_21_05_54_957: EngineState: confirmation_sentinfo:8
2022-10-22_21_05_55_720: EngineState: ---- SENT ch 0 ----
2022-10-22_21_05_55_721: EngineState: Unit time 267 timer ticks
2022-10-22_21_05_55_721: EngineState: Total pulses 2097728
2022-10-22_21_05_55_723: EngineState: Last fast msg Status 0x0 Sig0 0x1B7 Sig1 0xE48
2022-10-22_21_05_55_731: EngineState: Restarts 0
2022-10-22_21_05_55_732: EngineState: Interval errors 238 short, 2932 long
2022-10-22_21_05_55_732: EngineState: Total frames 227568 with crc error 128 (0.056246924 %)
2022-10-22_21_05_55_733: EngineState: Sync errors 238
2022-10-22_21_05_55_734: EngineState: --------------------
2022-10-22_21_05_55_734: EngineState: confirmation_sentinfo:8
2022-10-22_21_05_56_375: EngineState: ---- SENT ch 0 ----
2022-10-22_21_05_56_375: EngineState: Unit time 267 timer ticks
2022-10-22_21_05_56_375: EngineState: Total pulses 2106538
2022-10-22_21_05_56_387: EngineState: Last fast msg Status 0x0 Sig0 0x1B6 Sig1 0xE49
2022-10-22_21_05_56_390: EngineState: Restarts 0
2022-10-22_21_05_56_393: EngineState: Interval errors 239 short, 2939 long
2022-10-22_21_05_56_394: EngineState: Total frames 228533 with crc error 129 (0.056446992 %)
2022-10-22_21_05_56_397: EngineState: Sync errors 239
2022-10-22_21_05_56_399: EngineState: --------------------
2022-10-22_21_05_56_401: EngineState: confirmation_sentinfo:8

```

rusefillc, 2022-10-23:
@dron0gus see https://github.com/rusefi/rusefi/commit/6e24020ab58d9fefad53905d79f6e6b4059216ab

rusefillc, 2022-10-23:
larger 12617792 is also happy

idle position
```
2022-10-22_22_50_59_588: EngineState: WARNING: Injector 3 fault: open_load 
2022-10-22_22_51_00_340: EngineState: ---- SENT ch 0 ----
2022-10-22_22_51_00_340: EngineState: Unit time 272 timer ticks
2022-10-22_22_51_00_341: EngineState: Total pulses 2123425
2022-10-22_22_51_00_341: EngineState: Last fast msg Status 0x0 Sig0 0xAE9 Sig1 0x516
2022-10-22_22_51_00_344: EngineState: Restarts 1
2022-10-22_22_51_00_345: EngineState: Interval errors 183 short, 2568 long
2022-10-22_22_51_00_346: EngineState: Total frames 231111 with crc error 103 (0.044567328 %)
2022-10-22_22_51_00_348: EngineState: Sync errors 183
2022-10-22_22_51_00_349: EngineState: --------------------
2022-10-22_22_51_00_349: EngineState: confirmation_sentinfo:8
2022-10-22_22_51_00_888: EngineState: ---- SENT ch 0 ----
2022-10-22_22_51_00_888: EngineState: Unit time 272 timer ticks
2022-10-22_22_51_00_890: EngineState: Total pulses 2130497
2022-10-22_22_51_00_896: EngineState: Last fast msg Status 0x0 Sig0 0xAEA Sig1 0x515
2022-10-22_22_51_00_899: EngineState: Restarts 1
2022-10-22_22_51_00_901: EngineState: Interval errors 184 short, 2580 long
2022-10-22_22_51_00_904: EngineState: Total frames 231874 with crc error 104 (0.044851944 %)
2022-10-22_22_51_00_907: EngineState: Sync errors 184
2022-10-22_22_51_00_907: EngineState: --------------------
2022-10-22_22_51_00_907: EngineState: confirmation_sentinfo:8
2022-10-22_22_51_01_432: EngineState: ---- SENT ch 0 ----
2022-10-22_22_51_01_433: EngineState: Unit time 272 timer ticks
2022-10-22_22_51_01_433: EngineState: Total pulses 2138130
2022-10-22_22_51_01_437: EngineState: Last fast msg Status 0x0 Sig0 0xAEA Sig1 0x515
2022-10-22_22_51_01_440: EngineState: Restarts 1
2022-10-22_22_51_01_441: EngineState: Interval errors 184 short, 2589 long
2022-10-22_22_51_01_443: EngineState: Total frames 232706 with crc error 104 (0.044691584 %)
2022-10-22_22_51_01_443: EngineState: Sync errors 184
2022-10-22_22_51_01_444: EngineState: --------------------
2022-10-22_22_51_01_445: EngineState: confirmation_sentinfo:8

```

forced closed
```
2022-10-22_22_50_42_237: EngineState: ---- SENT ch 0 ----
2022-10-22_22_50_42_237: EngineState: Unit time 272 timer ticks
2022-10-22_22_50_42_238: EngineState: Total pulses 1876921
2022-10-22_22_50_42_240: EngineState: Last fast msg Status 0x0 Sig0 0xE3A Sig1 0x1C5
2022-10-22_22_50_42_240: EngineState: Restarts 1
2022-10-22_22_50_42_246: EngineState: Interval errors 155 short, 2248 long
2022-10-22_22_50_42_248: EngineState: Total frames 204323 with crc error 84 (0.041111376 %)
2022-10-22_22_50_42_250: EngineState: Sync errors 155
2022-10-22_22_50_42_253: EngineState: --------------------
2022-10-22_22_50_42_256: EngineState: confirmation_sentinfo:8
2022-10-22_22_50_42_890: EngineState: ---- SENT ch 0 ----
2022-10-22_22_50_42_890: EngineState: Unit time 272 timer ticks
2022-10-22_22_50_42_891: EngineState: Total pulses 1885722
2022-10-22_22_50_42_894: EngineState: Last fast msg Status 0x0 Sig0 0xE3A Sig1 0x1C5
2022-10-22_22_50_42_901: EngineState: Restarts 1
2022-10-22_22_50_42_905: EngineState: Interval errors 157 short, 2258 long
2022-10-22_22_50_42_908: EngineState: Total frames 205281 with crc error 85 (0.041406656 %)
2022-10-22_22_50_42_909: EngineState: Sync errors 157
2022-10-22_22_50_42_912: EngineState: --------------------
2022-10-22_22_50_42_913: EngineState: confirmation_sentinfo:8
2022-10-22_22_50_43_540: EngineState: ---- SENT ch 0 ----
2022-10-22_22_50_43_541: EngineState: Unit time 272 timer ticks
2022-10-22_22_50_43_541: EngineState: Total pulses 1894761
2022-10-22_22_50_43_541: EngineState: Last fast msg Status 0x0 Sig0 0xE39 Sig1 0x1C6
2022-10-22_22_50_43_542: EngineState: Restarts 1
2022-10-22_22_50_43_545: EngineState: Interval errors 158 short, 2265 long
2022-10-22_22_50_43_546: EngineState: Total frames 206272 with crc error 86 (0.041692520 %)
2022-10-22_22_50_43_547: EngineState: Sync errors 158
2022-10-22_22_50_43_566: EngineState: --------------------
2022-10-22_22_50_43_576: EngineState: confirmation_sentinfo:8
[... 30 lines cut ...]
2022-10-22_22_51_31_268: EngineState: Total frames 276636 with crc error 126 (0.045547216 %)
2022-10-22_22_51_31_269: EngineState: Sync errors 219
2022-10-22_22_51_31_271: EngineState: --------------------
2022-10-22_22_51_31_272: EngineState: confirmation_sentinfo:8
2022-10-22_22_51_31_809: EngineState: ---- SENT ch 0 ----
2022-10-22_22_51_31_809: EngineState: Unit time 272 timer ticks
2022-10-22_22_51_31_810: EngineState: Total pulses 2549552
2022-10-22_22_51_31_811: EngineState: Last fast msg Status 0x0 Sig0 0x279 Sig1 0xD86
2022-10-22_22_51_31_815: EngineState: Restarts 1
2022-10-22_22_51_31_816: EngineState: Interval errors 219 short, 3131 long
2022-10-22_22_51_31_817: EngineState: Total frames 277406 with crc error 126 (0.045420792 %)
2022-10-22_22_51_31_819: EngineState: Sync errors 219
2022-10-22_22_51_31_821: EngineState: --------------------
2022-10-22_22_51_31_822: EngineState: confirmation_sentinfo:8
2022-10-22_22_51_32_461: EngineState: ---- SENT ch 0 ----
2022-10-22_22_51_32_461: EngineState: Unit time 272 timer ticks
2022-10-22_22_51_32_462: EngineState: Total pulses 2558526
2022-10-22_22_51_32_462: EngineState: Last fast msg Status 0x0 Sig0 0x279 Sig1 0xD86
2022-10-22_22_51_32_462: EngineState: Restarts 1
2022-10-22_22_51_32_465: EngineState: Interval errors 222 short, 3139 long
2022-10-22_22_51_32_467: EngineState: Total frames 278387 with crc error 127 (0.045619944 %)
2022-10-22_22_51_32_468: EngineState: Sync errors 222
2022-10-22_22_51_32_468: EngineState: --------------------
2022-10-22_22_51_32_469: EngineState: confirmation_sentinfo:8
```

rusefillc, 2022-10-24:
@dron0gus 
```
setTickRate(100)
function onTick()
	print(getSentValue(1))
end

```

rusefillc, 2022-10-26:
```

tpsSensor = Sensor.new("Tps1")

setTickRate(100)
function onTick()
	sentValue = getSentValue(1)
	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)

	tpsSensor : set(tpsValue )
	print('sentValue ' ..sentValue..' tps ' ..tpsValue)
end

```

rusefillc, 2022-10-28:
lua
```
restartEtb
```

rusefillc, 2022-11-03:
Ford part https://rusefi.com/forum/viewtopic.php?p=46584#p46584

idle state
```
2022-11-02_23_36_52_173: EngineState: ---- SENT ch 0 ----
2022-11-02_23_36_52_173: EngineState: Unit time 506 CPU ticks 3.011904716 uS
2022-11-02_23_36_52_174: EngineState: Total pulses 1905744
2022-11-02_23_36_52_174: EngineState: Last valid fast msg Status 0x0 Sig0 0x231 Sig1 0xDAC
2022-11-02_23_36_52_174: EngineState: Slow channels:
2022-11-02_23_36_52_174: EngineState:  ID 1: 0
2022-11-02_23_36_52_174: EngineState:  ID 149: 0
2022-11-02_23_36_52_174: EngineState:  ID 3: 100
2022-11-02_23_36_52_174: EngineState:  ID 150: 0
2022-11-02_23_36_52_176: EngineState:  ID 6: 3
2022-11-02_23_36_52_176: EngineState:  ID 9: 248
2022-11-02_23_36_52_180: EngineState:  ID 35: 790
2022-11-02_23_36_52_182: EngineState:  ID 43: 1541
2022-11-02_23_36_52_185: EngineState:  ID 5: 66
2022-11-02_23_36_52_187: EngineState:  ID 7: 0
2022-11-02_23_36_52_189: EngineState:  ID 148: 17
2022-11-02_23_36_52_191: EngineState:  ID 147: 1625
2022-11-02_23_36_52_193: EngineState:  ID 41: 992
2022-11-02_23_36_52_193: EngineState:  ID 145: 2387
2022-11-02_23_36_52_196: EngineState:  ID 8: 85
2022-11-02_23_36_52_197: EngineState:  ID 42: 2192
2022-11-02_23_36_52_199: EngineState: Restarts 0
2022-11-02_23_36_52_200: EngineState: Interval errors 9686 short, 181529 long
2022-11-02_23_36_52_201: EngineState: Total frames 176519 with crc error 23968 (13.578142144 %)
2022-11-02_23_36_52_203: EngineState: Sync errors 176504
2022-11-02_23_36_52_205: EngineState: --------------------
2022-11-02_23_36_52_206: EngineState: confirmation_sentinfo:8

2022-11-02_23_38_33_286: EngineState: ---- SENT ch 0 ----
2022-11-02_23_38_33_287: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-02_23_38_33_287: EngineState: Total pulses 3035948
2022-11-02_23_38_33_288: EngineState: Last valid fast msg Status 0x0 Sig0 0x232 Sig1 0xD9D
2022-11-02_23_38_33_288: EngineState: Slow channels:
2022-11-02_23_38_33_288: EngineState:  ID 1: 0
2022-11-02_23_38_33_288: EngineState:  ID 149: 0
2022-11-02_23_38_33_291: EngineState:  ID 3: 100
2022-11-02_23_38_33_291: EngineState:  ID 150: 0
2022-11-02_23_38_33_292: EngineState:  ID 6: 3
2022-11-02_23_38_33_292: EngineState:  ID 9: 248
2022-11-02_23_38_33_293: EngineState:  ID 35: 790
2022-11-02_23_38_33_294: EngineState:  ID 43: 1541
2022-11-02_23_38_33_295: EngineState:  ID 5: 66
2022-11-02_23_38_33_296: EngineState:  ID 7: 0
2022-11-02_23_38_33_296: EngineState:  ID 148: 17
2022-11-02_23_38_33_296: EngineState:  ID 147: 1625
2022-11-02_23_38_33_296: EngineState:  ID 41: 992
2022-11-02_23_38_33_296: EngineState:  ID 145: 2387
2022-11-02_23_38_33_297: EngineState:  ID 8: 85
2022-11-02_23_38_33_297: EngineState:  ID 42: 2192
2022-11-02_23_38_33_297: EngineState: Restarts 0
2022-11-02_23_38_33_297: EngineState: Interval errors 15322 short, 289270 long
2022-11-02_23_38_33_297: EngineState: Total frames 281386 with crc error 38264 (13.598402048 %)
2022-11-02_23_38_33_297: EngineState: Sync errors 281370
2022-11-02_23_38_33_297: EngineState: --------------------
2022-11-02_23_38_33_297: EngineState: confirmation_sentinfo:8

```
Note 13% CRC error rate?

fully closed causes restats
```
2022-11-02_23_39_41_121: EngineState: ---- SENT ch 0 ----
2022-11-02_23_39_41_122: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-02_23_39_41_122: EngineState: Total pulses 5722
2022-11-02_23_39_41_124: EngineState: Last valid fast msg Status 0x8 Sig0 0x0F8 Sig1 0xF32
2022-11-02_23_39_41_124: EngineState: Restarts 55
[... 71 lines cut ...]
2022-11-02_23_41_08_593: EngineState: Restarts 75
2022-11-02_23_41_08_594: EngineState: Interval errors 1487 short, 15201 long
2022-11-02_23_41_08_594: EngineState: Total frames 14435 with crc error 2054 (14.229303360 %)
2022-11-02_23_41_08_594: EngineState: Sync errors 14432
2022-11-02_23_41_08_594: EngineState: --------------------
2022-11-02_23_41_08_594: EngineState: confirmation_sentinfo:8
2022-11-02_23_41_09_114: EngineState: WARNING: interpolatelua: why X
2022-11-02_23_41_09_444: EngineState: ---- SENT ch 0 ----
2022-11-02_23_41_09_444: EngineState: Unit time 509 CPU ticks 3.029761792 uS
2022-11-02_23_41_09_444: EngineState: Total pulses 176224
2022-11-02_23_41_09_446: EngineState: Last valid fast msg Status 0xC Sig0 0xF00 Sig1 0x021
2022-11-02_23_41_09_452: EngineState: Slow channels:
2022-11-02_23_41_09_453: EngineState:  ID 1: 0
2022-11-02_23_41_09_456: EngineState:  ID 147: 1625
2022-11-02_23_41_09_458: EngineState:  ID 44: 533
2022-11-02_23_41_09_461: EngineState:  ID 7: 0
2022-11-02_23_41_09_463: EngineState:  ID 146: 2457
2022-11-02_23_41_09_465: EngineState: Restarts 75
2022-11-02_23_41_09_465: EngineState: Interval errors 1590 short, 16079 long
2022-11-02_23_41_09_466: EngineState: Total frames 15259 with crc error 2170 (14.221115120 %)
2022-11-02_23_41_09_466: EngineState: Sync errors 15254
2022-11-02_23_41_09_466: EngineState: --------------------
2022-11-02_23_41_09_466: EngineState: confirmation_sentinfo:8

```

dron0gus, 2022-11-03:
Time to implement CRC check on slow channels...

dron0gus, 2022-11-03:
[image]
I need a hint

dron0gus, 2022-11-03:
LUA code:
```
tpsSensor = Sensor.new("Tps1")
ppsSensor = Sensor.new("AcceleratorPedal")
cnt = 10
restartEtb()
setTickRate(100)
function onTick()
	cnt = cnt + 1

	sentValue = getSentValue(1)
	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)

	tpsSensor : set(tpsValue )
	ppsSensor : set(cnt / 10)
	print('sentValue ' ..sentValue..' tps ' ..tpsValue .. ' pps ' .. (cnt/10))
	if (cnt >= 1000) then
		cnt = 10
	end
end
```

```
2022-11-03_12_34_16_741: EngineState: CRITICAL error: Pin "PC8" required by "ETB Disable" but is used by "ETB Disable" MINIMAL_PINS 20221103@microRusEFI
2022-11-03_12_34_16_742: EngineState: LUA: sentValue 0.3233211 tps 27.3488655 pps 67.8000030
2022-11-03_12_34_16_836: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 67.9000015
2022-11-03_12_34_16_837: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.0000000
2022-11-03_12_34_16_837: EngineState: LUA: sentValue 0.3233211 tps 27.3488655 pps 68.0999984
2022-11-03_12_34_16_837: EngineState: LUA: sentValue 0.3233211 tps 27.3488655 pps 68.1999969
2022-11-03_12_34_16_838: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.3000030
2022-11-03_12_34_16_838: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.4000015
2022-11-03_12_34_16_838: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.5000000
2022-11-03_12_34_16_838: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.5999984
2022-11-03_12_34_16_839: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.6999969
2022-11-03_12_34_16_840: EngineState: LUA: sentValue 0.3230769 tps 27.3175583 pps 68.8000030

```

`restartEtb()` cause second init of motor at electronic_throttle.cpp:967

dron0gus, 2022-11-03:
```
Thread 13 hit Breakpoint 2, brain_pin_markUsed (brainPin=<optimized out>, msg=0x80799f0 "ETB Disable") at ./hw_layer/pin_repository.cpp:66
66			firmwareError(CUSTOM_ERR_PIN_ALREADY_USED_1, "Pin \"%s\" required by \"%s\" but is used by \"%s\" %s",
(gdb) bt
#0  brain_pin_markUsed (brainPin=<optimized out>, msg=0x80799f0 "ETB Disable") at ./hw_layer/pin_repository.cpp:66
#1  0x08064d76 in efiSetPadMode (msg=msg@entry=0x80799f0 "ETB Disable", brainPin=brainPin@entry=Gpio::C8, mode=mode@entry=1) at ./hw_layer/io_pins.cpp:54
#2  0x0802fda6 in OutputPin::initPin (this=0x20006230 <dcHardware+48>, msg=0x80799f0 "ETB Disable", brainPin=<optimized out>, outputMode=outputMode@entry=0x2001b10f <DEFAULT_OUTPUT>, 
    forceInitWithFatalError=forceInitWithFatalError@entry=false) at ./controllers/system/efi_gpio.cpp:616
#3  0x0803018a in OutputPin::initPin (this=<optimized out>, msg=<optimized out>, brainPin=<optimized out>) at ./controllers/system/efi_gpio.cpp:543
#4  0x0803ea98 in DcHardware::start (frequency=800, executor=<optimized out>, isInverted=<optimized out>, pinDisable=<optimized out>, pinDir2=Gpio::Unassigned, pinDir1=Gpio::A8, pinEnable=Gpio::C7, 
    useTwoWires=<optimized out>, this=0x20006200 <dcHardware>) at ./controllers/actuators/dc_motors.cpp:61
#5  initDcMotor (io=..., index=index@entry=0, useTwoWires=<optimized out>) at ./controllers/actuators/dc_motors.cpp:120
#6  0x0803ebfc in doInitElectronicThrottle () at ./controllers/actuators/electronic_throttle.cpp:967
#7  0x0806106c in operator() (l=<optimized out>, __closure=0x0) at ./controllers/lua/lua_hooks.cpp:697
#8  _FUN () at ./controllers/lua/lua_hooks.cpp:697
#9  0x0805dc56 in luaD_precall (L=0x2002000c, func=0x20020370, nresults=0) at ./ext/lua/ldo.c:532
#10 0x080103b4 in luaV_execute (L=0x2002000c, ci=0x20020e18) at ./ext/lua/lvm.c:1624
#11 0x0805d9f8 in ccall (L=0x2002000c, func=<optimized out>, nResults=<optimized out>, inc=inc@entry=65537) at ./ext/lua/ldo.c:577
#12 0x0805da14 in luaD_callnoyield (L=<optimized out>, func=<optimized out>, nResults=<optimized out>) at ./ext/lua/ldo.c:595
#13 0x0805833c in f_call (L=<optimized out>, ud=<optimized out>) at ./ext/lua/lapi.c:1031
#14 0x0805bdf6 in luaD_rawrunprotected (L=L@entry=0x2002000c, f=0x8058333 <f_call(lua_State*, void*)>, ud=0x20003eb0 <luaThread+4336>) at ./ext/lua/ldo.c:144
#15 0x0805cc8a in luaD_pcall (L=0x2002000c, func=<optimized out>, u=<optimized out>, old_top=8, ef=0) at ./ext/lua/ldo.c:892
#16 0x08009a0c in lua_pcallk (L=0x2002000c, nargs=<optimized out>, nresults=-1, errfunc=<optimized out>, ctx=0, k=0x0) at ./ext/lua/lapi.c:1057
#17 0x08011f5e in loadScript (ls=..., 
    scriptStr=scriptStr@entry=0x1000163c <persistentState+5692> "tpsSensor = Sensor.new(\"Tps1\")\nppsSensor = Sensor.new(\"AcceleratorPedal\")\ncnt = 10\nrestartEtb()\nsetTickRate(100)\nfunction onTick()\n\tcnt = cnt + 1\n\n\tsentValue = getSentValue(1)\n\ttpsValue = interpolate("...) at ./controllers/lua/rusefi_lua.h:38
#18 0x080138b0 in runOneLua (alloc=alloc@entry=0x8011b8d <myAlloc(void*, void*, size_t, size_t)>, 
    script=script@entry=0x1000163c <persistentState+5692> "tpsSensor = Sensor.new(\"Tps1\")\nppsSensor = Sensor.new(\"AcceleratorPedal\")\ncnt = 10\nrestartEtb()\nsetTickRate(100)\nfunction onTick()\n\tcnt = cnt + 1\n\n\tsentValue = getSentValue(1)\n\ttpsValue = interpolate("...) at ./controllers/lua/lua.cpp:292
#19 0x08013994 in LuaThread::ThreadTask (this=<optimized out>) at ./controllers/lua/lua.cpp:320
#20 0x08011792 in ThreadController<4096>::main (this=<optimized out>) at ./controllers/system/thread_controller.h:30
#21 0x08060a02 in chibios_rt::_thd_start (arg=<optimized out>) at ChibiOS/os/various/cpp_wrappers/ch.cpp:41
#22 0x08001316 in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201

```

dron0gus, 2022-11-03:
SENT ETB starts to make some noise https://github.com/rusefi/rusefi/pull/4735

rusefillc, 2022-11-12:
Great news @dron0gus @mck1117 

master as is works perfectly with the following magic:

```
tpsSensor = Sensor.new("Tps1")
ppsSensor = Sensor.new("AcceleratorPedal")
Tps1Primary = Sensor.new("Tps1Primary")
AcceleratorPedalPrimary = Sensor.new("AcceleratorPedalPrimary")
ppsSensor : setRedundant(true)
tpsSensor : setRedundant(true)
```

that set of Lua sensors works with the current logic as in master

rusefillc, 2022-11-12:
Full script (please ignore CAN bus, please note timeout/tickrate weirdness)

```

tpsSensor = Sensor.new("Tps1")
ppsSensor = Sensor.new("AcceleratorPedal")
Tps1Primary = Sensor.new("Tps1Primary")
AcceleratorPedalPrimary = Sensor.new("AcceleratorPedalPrimary")

ppsSensor : setRedundant(true)

tpsSensor : setRedundant(true)

setTickRate(100)
-- huh? with 100Hz tick rate why do we need such huge timeout?
ppsSensor : setTimeout(1000)
tpsSensor : setTimeout(1000)
restartEtb()

function setTwoBytes(data, offset, value)
	value = math.floor(value)
	data[offset + 2] = value >> 8
	data[offset + 1] = value & 0xff
end

R52_RPM = 0x316

data = { 0, 0, 0, 00, 0, 0 , 0, 0 }

rpm = 7000

i = 0

function onTick()
	rpm = getSensor("RPM")

	if i == 3 then
		print('Postponed restart')
		restartEtb()
--		i = 0
	end
	-- 	print('Period ' ..i)

	i = i + 1

	setTwoBytes(data, 2, rpm * 6.39)

	txCan(1, R52_RPM, 0, data)

	-- 1 ABS and warm-up light but no tach
	-- 3 blinker and alive
	-- 4 ABS and alive
	--
	status2 = 2

	-- status3 = 0xC0
	status3 = 0x0

	status4 = 0x0
	-- status4 = 0xC0
	-- with ABS with blinker with warm-up light
	-- status4 = 0xFF

	txCan(1, 0x61f, 0, { 0, 0, status2, status3, status, 0 , 0, 0 })

	sentValue = getSentValue(1)
	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)

	tpsSensor : set(tpsValue )
	Tps1Primary : set(tpsValue )
	ppsSensor : set(50)

	print('sentValue ' ..sentValue..' tps ' ..tpsValue)
end

```

rusefillc, 2022-11-13:
Oooops I've broken GM, just fixed it

```
2022-11-13_11_42_29_677: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_29_677: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2022-11-13_11_42_29_677: EngineState: Total pulses 1941691
2022-11-13_11_42_29_678: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD3 Sig1 0x52C
2022-11-13_11_42_29_681: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_29_687: EngineState: Restarts 0
2022-11-13_11_42_29_689: EngineState: Interval errors 19520 short, 1616 long
2022-11-13_11_42_29_692: EngineState: Total frames 193264 with CRC error 15 (0.007761404%)
2022-11-13_11_42_29_694: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_29_695: EngineState: Sync errors 19520
2022-11-13_11_42_29_696: EngineState: --------------------
2022-11-13_11_42_29_698: EngineState: confirmation_sentinfo:8
2022-11-13_11_42_33_328: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_33_328: EngineState: Unit time 538 CPU ticks 3.202380896 uS
2022-11-13_11_42_33_330: EngineState: Total pulses 1990861
2022-11-13_11_42_33_333: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD3 Sig1 0x52C
2022-11-13_11_42_33_335: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_33_337: EngineState: Restarts 0
2022-11-13_11_42_33_338: EngineState: Interval errors 22219 short, 1654 long
2022-11-13_11_42_33_339: EngineState: Total frames 195963 with CRC error 15 (0.007654506%)
2022-11-13_11_42_33_341: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_33_341: EngineState: Sync errors 22219
2022-11-13_11_42_33_341: EngineState: --------------------
2022-11-13_11_42_33_342: EngineState: confirmation_sentinfo:8
2022-11-13_11_42_34_210: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_34_210: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2022-11-13_11_42_34_210: EngineState: Total pulses 2003197
2022-11-13_11_42_34_210: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD4 Sig1 0x52B
2022-11-13_11_42_34_210: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_34_210: EngineState: Restarts 0
2022-11-13_11_42_34_211: EngineState: Interval errors 22899 short, 1660 long
2022-11-13_11_42_34_215: EngineState: Total frames 196643 with CRC error 15 (0.007628036%)
2022-11-13_11_42_34_216: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_34_217: EngineState: Sync errors 22899
2022-11-13_11_42_34_218: EngineState: --------------------
2022-11-13_11_42_34_219: EngineState: confirmation_sentinfo:8
2022-11-13_11_42_34_544: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_34_544: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2022-11-13_11_42_34_544: EngineState: Total pulses 2007215
2022-11-13_11_42_34_544: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD4 Sig1 0x52B
2022-11-13_11_42_34_544: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_34_545: EngineState: Restarts 0
2022-11-13_11_42_34_546: EngineState: Interval errors 23118 short, 1665 long
2022-11-13_11_42_34_551: EngineState: Total frames 196862 with CRC error 15 (0.007619551%)
2022-11-13_11_42_34_552: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_34_553: EngineState: Sync errors 23118
2022-11-13_11_42_34_554: EngineState: --------------------
2022-11-13_11_42_34_555: EngineState: confirmation_sentinfo:8
2022-11-13_11_42_34_555: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_34_556: EngineState: Unit time 538 CPU ticks 3.202380896 uS
2022-11-13_11_42_34_556: EngineState: Total pulses 2008357
2022-11-13_11_42_34_557: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD4 Sig1 0x52B
2022-11-13_11_42_34_558: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_34_558: EngineState: Restarts 0
2022-11-13_11_42_34_559: EngineState: Interval errors 23181 short, 1666 long
2022-11-13_11_42_34_559: EngineState: Total frames 196925 with CRC error 15 (0.007617113%)
2022-11-13_11_42_34_560: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_34_560: EngineState: Sync errors 23181
2022-11-13_11_42_34_561: EngineState: --------------------
2022-11-13_11_42_34_561: EngineState: confirmation_sentinfo:8
2022-11-13_11_42_34_882: EngineState: ---- SENT ch 0 ----
2022-11-13_11_42_34_882: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2022-11-13_11_42_34_882: EngineState: Total pulses 2011733
2022-11-13_11_42_34_883: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD4 Sig1 0x52B
2022-11-13_11_42_34_883: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_34_883: EngineState: Restarts 0
2022-11-13_11_42_34_883: EngineState: Interval errors 23366 short, 1668 long
[... 311 lines cut ...]
2022-11-13_11_42_40_412: EngineState: Unit time 539 CPU ticks 3.208333248 uS
2022-11-13_11_42_40_412: EngineState: Total pulses 2086105
2022-11-13_11_42_40_413: EngineState: Last valid fast msg Status 0x0 Sig0 0xAD3 Sig1 0x52C
2022-11-13_11_42_40_420: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_40_421: EngineState: Restarts 0
2022-11-13_11_42_40_422: EngineState: Interval errors 27454 short, 1720 long
2022-11-13_11_42_40_423: EngineState: Total frames 201202 with CRC error 22 (0.010934285%)
2022-11-13_11_42_40_425: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_40_426: EngineState: Sync errors 27454
2022-11-13_11_42_40_427: EngineState: --------------------
[... cut ...]
2022-11-13_11_42_40_530: EngineState: valueErrorRate 0.000000000
2022-11-13_11_42_40_532: EngineState: Restarts 0
2022-11-13_11_42_40_532: EngineState: Interval errors 27575 short, 1723 long
2022-11-13_11_42_40_533: EngineState: Total frames 201323 with CRC error 22 (0.010927713%)
2022-11-13_11_42_40_534: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-13_11_42_40_537: EngineState: Sync errors 27575
2022-11-13_11_42_40_538: EngineState: --------------------
2022-11-13_11_42_40_538: EngineState: confirmation_sentinfo:8

```

rusefillc, 2022-11-13:
I think we are mission accomplished here see https://rusefi.com/forum/viewtopic.php?p=44982#p44982

[image]
[image]
[image]
[image]

closed 2022-11-13 by rusefillc