rusefi/rusefi #4689: review typo and overall sentinfo output

rusefillc, 2022-10-20:
100% CRC error or happy?

```
2022-10-20_22_05_08_337: EngineState: ---- SENT ch 0 ----
2022-10-20_22_05_08_338: EngineState: Unit time 275 timer ticks
2022-10-20_22_05_08_339: EngineState: Total pulses 65666
2022-10-20_22_05_08_339: EngineState: Last fast msg Status 0x0 Sig0 0x0C6 Sig1 0x0CB
2022-10-20_22_05_08_340: EngineState: Slow channels:
2022-10-20_22_05_08_340: EngineState: Restarts 1
2022-10-20_22_05_08_340: EngineState: Interval errors 21 short, 316 long
2022-10-20_22_05_08_340: EngineState: Total frames 6709 with crc erorr 6709
2022-10-20_22_05_08_341: EngineState: Sync errors 21
2022-10-20_22_05_08_341: EngineState: --------------------
```

rusefillc, 2022-10-20:
let's add  Unit time validation

for X <  Unit time < Y says "reasonable" otherwise say "does not look good"

dron0gus, 2022-10-22:
https://github.com/rusefi/rusefi/pull/4692

Now looks like:
```
2022-10-23_00_40_30_226: EngineState: ---- SENT ch 0 ----
2022-10-23_00_40_30_226: EngineState: Unit time 275 timer ticks
2022-10-23_00_40_30_227: EngineState: Total pulses 2998718
2022-10-23_00_40_30_227: EngineState: Last fast msg Status 0x0 Sig0 0xAD4 Sig1 0x52B
2022-10-23_00_40_30_228: EngineState: Restarts 0
2022-10-23_00_40_30_228: EngineState: Interval errors 544 short, 9998 long
2022-10-23_00_40_30_229: EngineState: Total frames 314564 with crc error 249 (0.079157184 %)
2022-10-23_00_40_30_230: EngineState: Sync errors 544
2022-10-23_00_40_30_230: EngineState: --------------------

```

closed 2022-10-22 by rusefillc