rusefi/rusefi #4706: RPM not reading, test units 4chan and hellen154

ElDominio, 2022-10-27:
last known good FW was 20221018, recent updates have broken RPM reading on test cars with 4B11 and Hyundai Theta 60-2. Downgrading back to 20221018 fixes RPM reading. RusEFI console can see all teeth without problem but RPM never shows up on Tunerstudio/car doesn't run

mck1117, 2022-10-27:
logs?

mck1117, 2022-10-27:
[image]

rusefillc, 2022-10-27:
It's past sunset here unfortunatelly and testing happens outside. No logs today unfortunatelly.

@mck1117 is it easy for you to update any vehicles in your visinity to latest?

mck1117, 2022-10-27:
My car runs and drives on FW from 10/21, I can update the other one to snapshot and see what happens.

mck1117, 2022-10-27:
no start on latest snapshot

mck1117, 2022-10-27:
what's up with those zeroish dwell spark events?

[image]

mck1117, 2022-10-27:
Trigger input DEBUG pins? What?

[image]

rusefillc, 2022-10-28:
Introduced by #4691

thank you @mck1117 for finding root cause!

closed 2022-10-28 by rusefillc