rusefi/rusefi #4707: Backup ecu state button

rusefillc, 2022-10-27:
Since we do not have resources available to have LTS we need a way to read FW (and tune?) From ECU

mck1117, 2022-10-27:
> do not have resources available to have LTS

do we think that's really true?

rusefillc, 2022-10-28:
> why do we need to read fw? The signature already gives a unique ID of what firmware we're running.
signature gives FW yes but that snapshot FW is not available anywhere

we need "transfer my favourite snapshot from one unit to another" functionality and "let me backup to a safer point"

both are hacks around the snapshot reliability/lack of LTS

mck1117, 2022-10-28:
> we need "transfer my favourite snapshot from one unit to another" functionality and "let me backup to a safer point"

This seems like an obscure enough need that one could use stm32cubeprogrammer instead of us writing custom code for this