rusefi/rusefi #4717: RPM freezes and drops to zero few seconds after engine dies

dron0gus, 2022-10-30:
Physical RPM gauge (and TS) drops to zero with some delay after actual engine stop.
[image]
Before it freezes on some value for two seconds.

rusefillc, 2022-11-06:
Somewhat relevant code is
```
	bool engineMovedRecently(efitick_t nowNt) const {
		constexpr float oneRevolutionLimitInSeconds = 60.0 / RPM_LOW_THRESHOLD;
		auto maxAverageToothTime = oneRevolutionLimitInSeconds / triggerShape.getSize();

		// Some triggers may have long gaps (with many teeth), don't count that as stopped!
		auto maxAllowedGap = maxAverageToothTime * 10;

		// Clamp between 0.1 seconds ("instant" for a human) and worst case of one engine cycle on low tooth count wheel
		maxAllowedGap = clampF(0.1f, maxAllowedGap, oneRevolutionLimitInSeconds);

		return getSecondsSinceTriggerEvent(nowNt) < maxAllowedGap;
	}

```

mck1117, 2022-11-06:
is this a problem? what are we supposed to do instead?

rusefillc, 2022-11-06:
I assume the question is "why does it takes is long to detect that engine has stopped"