rusefi/rusefi #4722: UI Suggestions for Tunerstudio

ElDominio, 2022-10-31:
Create a "Tuning" or "Popular" tab with the basics like VE, main ignition, main AFR, accel enrich and idle

Idle% adder for A/C should also be under "Air conditioning"
Fan adders should be under Fans
remove extra idle air and taper until they do something documented

Idle position is under "Sensors - Extra 1" instead of an "Idle" section, even though there's an IDLE PID section with everything but the actual output (for gauges)

I'm going to post a critique of each section individually so my brain doesn't fry

ElDominio, 2022-10-31:
[image]

Order should be from most important to least, with folders always at the bottom for visual cohesion
Ignition settings
________________
Ignition Advance
Ign CLT correction
Ign IAT correction (important to indicate IGN even though this is the tab because human brains short circuit)
________________
Dwell
Ignition Blend tables (what is an adder bias?)
Per Cylinder Trims

ElDominio, 2022-10-31:
Put Trigger Gap Override as a part of Advanced Trigger, it is literally an advanced trigger option
Why is CLT based RPM limit in its own section? imo make it a menu that enables if chosen in the limits category

All the small additional outputs should be clumped together in one or two "Outputs" section, no reason to have two clicks to set each output (more if you count burns every time)

ElDominio, 2022-10-31:
[image]

This one is a bit more involved, but I'd replace "Sensors" with "General I/O" and clump together inputs and puts (seperate with a horizontal lines for inputs and outputs) and keep Advanced for actual advanced functions, like traction control, launch control, flex fuel blending etc

[image]

Move I/O configuration to that tab, then tables that use that configuration (like VVT) live inside Advanced.

dron0gus, 2022-10-31:
https://github.com/rusefi/rusefi/pull/4723

ElDominio, 2022-11-03:
Another suggestion: remove deadzone on Closed Loop Ignition timing

rusefillc, 2022-11-07:
@ElDominio @dron0gus are we 100% done or less than 100% done here? how do we track what else needs to be changed?

@ElDominio do you have time and motivation to create a new one for remaining stuff? any other approach?

dron0gus, 2022-11-13:
@rusefillc suggestion from only two first messages are done.

ElDominio, 2022-11-13:
I have thought a bit more, I think Advanced should eb changed to "Outputs"  and take out
-cylinder offsets (group under base engine)
-gear detection (should be under Sensors - Vehicle speed since it's directly dependant on this)
-Lua Analog Inputs (Should be under Sensors, since it's literally a sensor)
--or, yank out all Lua stuff from "Advanced" and put it under a folder in Controller called "LUA", since this is quite related to the controller itself and a very not-begginer-level capability.

-make View the last button or next to last button, the things here are used almost exclusively for debugging. In fact, this sould be a folder under Controller since the views there are all debug functions.

[image]
o means output, i means input. Alternator and Battery should not be tied together.
Start Button should be renamed to Starter Control.

rusefillc, 2022-11-13:
@ElDominio "how do we track what else needs to be changed?" was me trying to say "please stop piling all small separate changes into one ticket it's already impossible to figure out what was done and what still needs to be done"

It would be amazing to separate all things which would be separated while all things relevant together :) "starter control" and "CLT warm-up curve" would be examples of things which seem SEPARATE to me.

rusefillc, 2022-11-13:
progress?

[image]

[image]

closed 2023-08-24 by ElDominio