rusefi/rusefi #4727: target lambda & afr log fields broken

mck1117, 2022-11-02:
these fields are always 0 in both SD and TS logs

mck1117, 2022-11-02:
We allow duplicate fields in output channels !?!?!?!?

[image]

dron0gus, 2022-11-02:
Similar to https://github.com/rusefi/rusefi/issues/4720

rusefillc, 2022-11-03:
@mck1117 I've disallowed duplicate fields in output channels, I hope that would fix target lambda in TS logs

@mck1117 can you help with SD card logs? specifically my problem is around current requirement to generate very static reference

BeerMoneyMotorsports, 2022-11-26:
related? https://github.com/rusefi/rusefi/issues/4826

mck1117, 2022-11-26:
No, it's not SD, it's everywhere. It looks like now everything in `fuel_computer_s` is stuck at 0.

rusefillc, 2022-11-26:
@BeerMoneyMotorsports yes does look pretty related

Just updated ticket title - the hope is that TS logs are working OK now but that was not really validated.

rusefillc, 2022-11-30:
@mck1117 sounds like closing this one as well

closed 2022-11-30 by rusefillc