rusefi/rusefi #4737: master does not compile on specific laptop

rusefillc, 2022-11-05:
My upstairs bedroom Windows 7 laptop fails to compile rusEFI master :(

[image]

```
Compiling tunerstudio.cpp
Compiling tunerstudio_commands.cpp
Compiling bluetooth.cpp
Compiling signature.cpp
Compiling pin_repository.cpp
Compiling microsecond_timer.cpp
./console/binary/tunerstudio.cpp: In member function 'int TunerStudio::handleCrcCommand(TsChannelBase*, char*, int)':
./console/binary/tunerstudio.cpp:710:32: error: this statement may fall through [-Werror=implicit-fallthrough=]
  710 |   EnableToothLoggerIfNotEnabled();
```

```
C:\stuff\rusefi>arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gcc.exe
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ arm\ embedded\ toolchain/10\ 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/mnt/wor
kspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw --libexecdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/li
b --infodir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-
pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-non
e-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disabl
e-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as -
-with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/arm-non
e-eabi --with-libiconv-prefix=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-gmp=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-p
ipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpc=/mnt/workspace/wo
rkspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/hos
t-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm
' --with-pkgversion='GNU Arm Embedded Toolchain 10.3-2021.10' --with-multilib-list=rmprofile,aprofile
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)

C:\stuff\rusefi>                                                                                                                                                                                                    ```

rusefillc, 2022-11-06:
And here is downstairs workstation where it compiles, gcc seems the same :(

```
   text    data     bss     dec     hex filename                                                                                                                                                                    521298    3932  191712  716942   af08e build/rusefi.elf                                                                                                                                                           bss Total size: 103060                                                                                                                                                                                             ram4 Total size: 0                                                                                                                                                                                                 text Total size: 398220                                                                                                                                                                                            data Total size: 3913                                                                                                                                                                                              rodata Total size: 124933                                                                                                                                                                                           20 .ram4         0000fc38  10000000  080803bc  000a0000  2**3                                                                                                                                                                                                                                                                                                                                                                        Done                                                                                                                                                                                                                                                                                                                                                                                                                                  C:\stuff\rusefi\firmware>arm-none-eabi-gcc -v                                                                                                                                                                      Using built-in specs.                                                                                                                                                                                              COLLECT_GCC=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gcc.exe
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ arm\ embedded\ toolchain/10\ 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/lto-wrapper.exe                                                                         Target: arm-none-eabi                                                                                                                                                                                              Configured with: /mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw --libexecdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/lib --infodir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/arm-none-eabi --with-libiconv-prefix=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-gmp=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-libelf= [... cut ...]                                                                                                                                                                                                                                                                                              C:\stuff\rusefi\firmware>                                                                                                                                                                                         ↑
```

rusefillc, 2022-11-06:
@mck1117 any ideas what could be messed up between different devices?

nmschulte, 2022-11-06:
I don't know much about ChibiOS, but I see in the Makefile this block (about `USE_OPT`/`USE_COPT`): https://github.com/rusefi/rusefi/blob/master/firmware/Makefile#L69

``` Makefile
# C specific options here (added to USE_OPT).
ifeq ($(USE_COPT),)
  USE_COPT = -fgnu89-inline -std=gnu99 -Wno-error=implicit-fallthrough
endif
```

Perhaps your build environment (Eclipse) is setting `USE_COPT` and in turn skipping this addition?  If that's not the case, I suppose you could add `-Wno-error=implicit-fallthrough` to `EXTRA_PARAMS`, or `EXTRA_2_PARAMS`, or via `CFLAGS` or `CXXFLAGS` to avoid the error from GCC.

rusefillc, 2022-11-06:
99% of the times I just build default primary configuration and that's what I do today

same issue without Eclipse just command line on the affected device:

[image]

nmschulte, 2022-11-06:
FWIW, `master` builds fine on my amd64 Linux machine:

`time make -j8`
```
No CCACHE_DIR
PROJECT_BOARD: f407-discovery
PROJECT_CPU:   ARCH_STM32F4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
Compiler Options
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -O2 -ggdb -g -fomit-frame-pointer -fsingle-precision-constant -fno-inline-functions -Werror -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=sign-compare -Wno-error=unused-parameter -DEFI_UNIT_TEST=0 -DEFI_PROD_CODE=1 -DEFI_SIMULATOR=0 -ffunction-sections -fdata-sections -fno-common -flto -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fgnu89-inline -std=gnu99 -Wno-error=implicit-fallthrough -Wall -Wextra -Wstrict-prototypes -Wa,-alms=build/lst/ -DSTM32F407xx -DFIRMWARE_ID="stm32f407vg" -DHW_FRANKENSO=1 -DSTM32_I2C_USE_I2C3=TRUE -DHAL_USE_EEPROM=TRUE -DSTM32F407xx -DCORTEX_USE_FPU=TRUE -MD -MP -MF .dep/build.d -I. -I./pch -IChibiOS/os/license -IChibiOS/os/common/portability/GCC -IChibiOS/os/common/startup/ARMCMx/compilers/GCC -IChibiOS/os/common/startup/ARMCMx/devices/STM32F4xx -IChibiOS/os/common/ext/ARM/CMSIS/Core/Include -IChibiOS/os/common/ext/ST/STM32F4xx -IChibiOS/os/hal/include -IChibiOS-Contrib/os/hal/include -IChibiOS/os/hal/ports/common/ARMCMx -IChibiOS/os/hal/ports/STM32/STM32F4xx -IChibiOS/os/hal/ports/STM32/LLD/ADCv2 -IChibiOS/os/hal/ports/STM32/LLD/CANv1 -IChibiOS/os/hal/ports/STM32/LLD/CRYPv1 -IChibiOS/os/hal/ports/STM32/LLD/DACv1 -IChibiOS/os/hal/ports/STM32/LLD/DMAv2 -IChibiOS/os/hal/ports/STM32/LLD/EXTIv1 -IChibiOS/os/hal/ports/STM32/LLD/GPIOv2 -IChibiOS/os/hal/ports/STM32/LLD/I2Cv1 -IChibiOS/os/hal/ports/STM32/LLD/MACv1 -IChibiOS/os/hal/ports/STM32/LLD/OTGv1 -IChibiOS/os/hal/ports/STM32/LLD/QUADSPIv1 -IChibiOS/os/hal/ports/STM32/LLD/RTCv2 -IChibiOS/os/hal/ports/STM32/LLD/SPIv1 -IChibiOS/os/hal/ports/STM32/LLD/SDIOv1 -IChibiOS/os/hal/ports/STM32/LLD/TIMv1 -IChibiOS/os/hal/ports/STM32/LLD/USARTv1 -IChibiOS/os/hal/ports/STM32/LLD/xWDGv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/CRCv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/DMA2Dv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/FSMCv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/TIMv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/LTDCv1 -IChibiOS-Contrib/os/hal/ports/STM32/LLD/USBHv1 -IChibiOS/os/hal/osal/rt-nil -IChibiOS/os/rt/include -IChibiOS/os/oslib/include -IChibiOS/os/common/ports/ARMCMx -IChibiOS/os/common/ports/ARMCMx/compilers/GCC -IChibiOS/os/ex/include -IChibiOS/os/ex/devices/ST -IChibiOS/os/hal/lib/streams -IChibiOS/os/various/cpp_wrappers -IChibiOS/os/hal/lib/complex/mfs -I./ext -I./ext/uzlib/src -I./controllers/lua -I./controllers/lua/luaaa -I./ext/lua -I./config/stm32f4ems -I./console/binary -I./console -I./console/binary_log -I./development -I./config/engines -I./config/boards/ -I./hw_layer/algo -I./init -I./ext_algo -I./hw_layer/drivers -I./hw_layer/drivers/gpio -I./hw_layer/drivers/can -I./hw_layer/drivers/sent -I./hw_layer/drivers/serial -I./hw_layer/drivers/i2c -I./hw_layer/drivers/lcd -I./hw_layer -I./hw_layer/adc -I./hw_layer/digital_input -I./hw_layer/digital_input/trigger -I./hw_layer/microsecond_timer -I./hw_layer/sensors -I./util -I./util/containers -I./util/math -I./controllers -I./controllers/system -I./controllers/system/timer -I./controllers/algo -I./controllers/algo/airmass -I./controllers/algo/defaults -I./controllers/algo/fuel -I./controllers/engine_cycle -I./controllers/trigger/decoders -I./controllers/tcu -I./controllers/trigger -I./controllers/sensors -I./controllers/sensors/converters -I./controllers/can -I./controllers/core -I./controllers/gauges -I./controllers/math -I./controllers/generated -I./controllers/actuators -I./controllers/actuators/gppwm -I./controllers/serial -I./console/binary/generated -I./bootloader/ -IChibiOS/os/various -IChibiOS/os/hal/lib/peripherals/sensors -I./libfirmware/util/include -I./libfirmware/pt2001/include -I./libfirmware/pt2001/project/rusefi/sample_code -I./hw_layer/ports/stm32/stm32f4/cfg -Iext/FatFS -I./hw_layer/mass_storage -I./hw_layer/serial_over_usb -I./hw_layer/lcd -I./hw_layer/mass_storage -Ihw_layer/ports/stm32/stm32f4 -I./hw_layer/ports -I./hw_layer/ports/stm32 -I./hw_layer/ports/stm32/serial_over_usb -Idevelopment/hw_layer -Idevelopment/test -IChibiOS-Contrib/os/various main.c -o main.o

Compiling pch.h
Compiling crt0_v7m.S
Compiling vectors.S
Compiling chcoreasm_v7m.S
Compiling rusEfiStartup.S

...

Compiling vr_pwm.cpp
Compiling tunerstudio_io.cpp
Compiling tunerstudio_io_serial.cpp
Compiling tunerstudio_io_serial_ports.cpp
Compiling live_data.cpp
Compiling ts_can_channel.cpp
Compiling serial_can.cpp
Compiling tunerstudio.cpp
Compiling tunerstudio_commands.cpp
Compiling bluetooth.cpp
Compiling signature.cpp

...

Done

real	0m24.685s
user	1m17.806s
sys	0m14.934s
```
`arm-none-eabi-gcc -v`
```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/11.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disabl [... cut ...] T_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip SED=/bin/sed SHELL=/bin/sh BASH=/bin/bash CONFIG_SHELL=/bin/bash
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.3.1 20220712 (15:11.3.rel1-1) 
```

dron0gus, 2022-11-06:
Relevant documentation https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wimplicit-fallthrough

dron0gus, 2022-11-06:
And https://stackoverflow.com/questions/45129741/gcc-7-wimplicit-fallthrough-warnings-and-portable-way-to-clear-them

rusefillc, 2022-11-07:
@dron0gus the part I am focused on is the fact that seems like to devices with similar setup produce different result

I would love for all workstations to produce same result :(

rusefillc, 2022-11-15:
Fun fact: this weirdness still happens even with gcc11 migration

[image]

mi-hol, 2023-04-01:
@rusefillc no activity since months, suggest to close