rusefi/rusefi #4744: new trigger wheel shape for Benelli 3-cylinder engines

mi-hol, 2022-11-06:
Benelli 3-cylinder engines use a trigger wheel shape not supported yet.
I don't have a supported ECU yet, hence can't provide a trigger log as described in https://github.com/rusefi/rusefi/wiki/Trigger-Configuration-Guide.
Is it feasible to get definition added with just the picture and maybe some measurements of its dimension?

[image]
More pictures and context are available on http://engenia.com.au/05_Engine/01_Camshaft/01_Remove/Removal.html

rusefillc, 2022-11-06:
@mck1117 does this look like a "normal" skipped wheel or not really due to extra wide tooth in addition to skipped tooth?

mck1117, 2022-11-06:
@mi-hol is that a VR or hall effect sensor?

mck1117, 2022-11-06:
> extra wide tooth in addition to skipped tooth?

depends on how brave we're willing to get with an angle grinder and file...

mi-hol, 2022-11-06:
> is that a VR or hall effect sensor?

I've relayed the question to Benelli experts.
In case nobody knows, is there maybe a simple test to identify them?

mck1117, 2022-11-06:
How many pins? If 2 pins, what's the resistance between the pins?

mi-hol, 2022-11-06:
> depends on how brave we're willing to get with an angle grinder and file...

Benelli spare part are rare, hence a physical modification if trigger wheel is out of scope

mck1117, 2022-11-06:
> Benelli spare parts are rare, hence a physical modification if trigger wheel is out of scope

That also looks like a very simple trigger wheel to copy - many places like SendCutSend will laser cut you a new one for very cheap (~$10).

mi-hol, 2022-11-06:
> looks like a very simple trigger wheel to copy

Are you aware of a simple process to copy? CAD is none of my skills :(

rusefillc, 2022-11-06:
@mck1117 if that's not a skipped wheel I think we should simply add it into the firmware

mck1117, 2022-11-06:
> @mck1117 if that's not a "normal" skipped wheel I think we should simply add it into the firmware

Well, it's maybe not that simple. If it's a hall sensor (or can be swapped to a hall sensor), then it is a conventional skipped wheel. It may also work as a skipped wheel with some creative gap ratios, so long as nothing important has to happen in the gap where you might get an inaccurate tooth. I'm not sure what the right move is here without additional information about the sensor.

mi-hol, 2022-11-06:
> How many pins? If 2 pins, what's the resistance between the pins?

will measure and get back to you tomorrow

mck1117, 2022-11-06:
> Are you aware of a simple process to copy? CAD is none of my skills :(

I've also heard good things about these guys: http://trigger-wheels.com/store/index.html

With some measurements (or send them the existing one), I'm sure they could copy it.

rusefillc, 2022-11-06:
[image]

mck1117, 2022-11-06:
yeah this is a "regular" 24-2 if you snipped off the part from 22-36 degrees

nmschulte, 2022-11-06:
Just from the photo, it's 24 tooth w/ a gap one-adjacent to the fill.  Quick research suggests it's a VR sensor; "phonic wheel" term and related patents, and this replacement part shows two-wire: https://www.aliexpress.us/item/2251832139717815.html

mi-hol, 2022-11-07:
> How many pins? If 2 pins, what's the resistance between the pins?

2 pins, resistance .506 kOhm

rusefillc, 2022-11-07:
2 wires is always VR right @dron0gus @mck1117 ?

dron0gus, 2022-11-07:
99% VR if 2 pins.

Engenia, 2022-11-07:
Definitely inductive pickup. Haven't checked, but I expect there is a magnet included. They usually do. The only issue to deal with is the orientation of the pins. Swapping them will change the slope direction and most ECUs work better in one direction than the other.

mck1117, 2022-11-08:
> 2 wires is always VR right @dron0gus @mck1117 ?

With 506 ohms between the pins, yes, that's a VR sensor.

rusefillc, 2022-11-13:
@mi-hol what is your timeline/order of actions for the whole project? Adding new trigger would take only two hours but my preference is to start once I see test mule with some rusEFI hardware next to it. Any chance you could start a build thread with your plans on the forum or shall we voice talk or else?

mi-hol, 2022-11-13:
realistically I'll try to have my test mule ready by April 2023.
I'll  start a build thread with my plans on the forum.
Biggest risk I see ATM is the potentially required "modification" of phonic wheel pointed out by [mck1117](https://github.com/mck1117).

rusefillc, 2022-11-13:
Please note that @mck1117 and myself have different points of view here: I am positive that new decoder is the way to go _once we see a viable test mule_, and it's 100% feasable.

Engenia, 2022-11-13:
If the decoder is firmware then that is definitely the way forward.
If the decoder was hardware, then it makes little difference whether the change is made at it or the wheel.

rusefillc, 2022-11-13:
@Engenia it's the firmware decoder I was referring two. Hardware-vise it's VR and all VR seem the same so far.

mi-hol, 2022-11-16:
wow this is fast turnaround :)

rusefillc, 2022-11-16:
it's definitely not done yet, give me a few more hours :)

mi-hol, 2022-11-16:
May I suggest to name the trigger "Benelli_Tre" instead of just "Benelli"?
Background: Benelli build engines ranging 1,  2, 3, 4 and 6-cylinders

rusefillc, 2022-11-16:
Now it's done as "Benelli_Tre"

At least done enough for real hardware tests

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#benelli-tre

mi-hol, 2022-11-16:
Will the change become available after the next build of TunerStudio or do I need to do something else?

mck1117, 2022-11-16:
@rusefillc I don't think that pattern will work.

VR sensors generate good edges on the centers of teeth.  VR sensors are also poorly behaved during very long teeth.

The edge marked "TDC" is in the wrong spot.

I think this one needs a logic analyzer trace before we can move forward. As-is, even if it does sync, it could result in unpredictable behavior.

[image]

rusefillc, 2022-11-16:
> Will the change become available after the next build of TunerStudio or do I need to do something else?

https://github.com/rusefi/rusefi/wiki/Download#faq

Really has nothing to do with TunerStudio builds.

rusefillc, 2022-11-16:
Huh? I've contributed FAQ content and responded with a link, that's not enough @mck1117 ? What's _your_ problem?

https://github.com/rusefi/rusefi_documentation/commit/b8cc5af30eccbf44e9a14986bccac7a127fe5707

mi-hol, 2022-11-17:
@rusefillc @mck1117 thanks for your help

Guess the confusion stems from statement "Really **has nothing** to do with TunerStudio **builds**." developers use the term "build" for compile & packaging of software and both of you pointed that out, just in different words.
My understanding of process improved :)

rusefillc, 2022-11-17:
My focus was on the term *TunerStudio*

mi-hol, 2022-11-17:
> I think this one needs a logic analyzer trace before we can move forward

@mck1117 would rusEfi console tooth logger be a good enough source for such a trace?

mck1117, 2022-11-18:
> @mck1117 would rusEfi console tooth logger be a good enough source for such a trace?

TunerStudio's tooth logger should work great. I haven't tried, but if the console also replicates that functionality, then it should work too.

mi-hol, 2023-01-29:
Found a different phonic wheel shape in my test mule engine.
Seems a simpler pattern 
[image]

mck1117, 2023-01-29:
Oh that's a much better option - this is just 30-2 skipped tooth (an unusual number of teeth, but we fully support any reasonable integer number of teeth).