rusefi/rusefi #4745: Ford SENT ETB bench test in NJ

rusefillc, 2022-11-06:
Forking https://github.com/rusefi/rusefi/issues/4688#issuecomment-1301606584 into a separate ticket since separate enough

rusefillc, 2022-11-06:
@dron0gus 

[image]

[image]

rusefillc, 2022-11-07:
Digital recording added https://github.com/rusefi/rusefi-hardware/tree/main/SENT-box/SENT-recordings

that stuff uses legacy Logic 1.x software

https://support.saleae.com/logic-software/legacy-software/older-software-releases

rusefillc, 2022-11-11:
Looks like https://github.com/rusefi/rusefi/pull/4769 is more than just unit test wonder of there is a chance of Ford SENT progress? 

https://github.com/rusefi/rusefi/pull/4769

rusefillc, 2022-11-11:
@dron0gus fresh data, this whole snippet is forced-closed - there is progress for sure, but still restarts happening

```
7 Sig1 0xFA6
2022-11-11_17_50_40_472: EngineState: Restarts 49
2022-11-11_17_50_40_475: EngineState: Interval errors 39 short, 61 long
2022-11-11_17_50_40_476: EngineState: Total frames 287 with CRC error 31 (10.801393536%)
2022-11-11_17_50_40_477: EngineState: Sync errors 5
2022-11-11_17_50_40_477: EngineState: --------------------
2022-11-11_17_50_40_477: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_40_790: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_40_790: EngineState: Unit time 512 CPU ticks 3.047619104 uS
2022-11-11_17_50_40_790: EngineState: Total pulses 7355
2022-11-11_17_50_40_791: EngineState: Last valid fast msg Status 0xC Sig0 0x0F8 Sig1 0xF6C
2022-11-11_17_50_40_792: EngineState: Restarts 49
2022-11-11_17_50_40_798: EngineState: Interval errors 71 short, 151 long
2022-11-11_17_50_40_800: EngineState: Total frames 512 with CRC error 49 (9.570312512%)
2022-11-11_17_50_40_801: EngineState: Sync errors 11
2022-11-11_17_50_40_802: EngineState: --------------------
2022-11-11_17_50_40_802: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_40_803: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_40_804: EngineState: Unit time 505 CPU ticks 3.005952358 uS
2022-11-11_17_50_40_804: EngineState: Total pulses 8329
2022-11-11_17_50_40_805: EngineState: Last valid fast msg Status 0x0 Sig0 0x0F8 Sig1 0xF72
2022-11-11_17_50_40_805: EngineState: Restarts 49
2022-11-11_17_50_40_806: EngineState: Interval errors 81 short, 173 long
2022-11-11_17_50_40_806: EngineState: Total frames 578 with CRC error 57 (9.861591360%)
2022-11-11_17_50_40_807: EngineState: Sync errors 14
2022-11-11_17_50_40_808: EngineState: --------------------
2022-11-11_17_50_40_808: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_41_119: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_41_120: EngineState: Unit time 509 CPU ticks 3.029761792 uS
2022-11-11_17_50_41_121: EngineState: Total pulses 11006
2022-11-11_17_50_41_134: EngineState: Last valid fast msg Status 0x0 Sig0 0x0F8 Sig1 0xF63
2022-11-11_17_50_41_137: EngineState: Restarts 49
2022-11-11_17_50_41_138: EngineState: Interval errors 110 short, 231 long
2022-11-11_17_50_41_138: EngineState: Total frames 760 with CRC error 72 (9.473684320%)
2022-11-11_17_50_41_141: EngineState: Sync errors 19
2022-11-11_17_50_41_142: EngineState: --------------------
2022-11-11_17_50_41_143: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_41_225: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_41_226: EngineState: Unit time 506 CPU ticks 3.011904716 uS
2022-11-11_17_50_41_226: EngineState: Total pulses 13147
2022-11-11_17_50_41_227: EngineState: Last valid fast msg Status 0x0 Sig0 0x0F8 Sig1 0xFC0
2022-11-11_17_50_41_233: EngineState: Restarts 49
2022-11-11_17_50_41_234: EngineState: Interval errors 134 short, 261 long
2022-11-11_17_50_41_235: EngineState: Total frames 921 with CRC error 85 (9.229099280%)
2022-11-11_17_50_41_236: EngineState: Sync errors 25
2022-11-11_17_50_41_238: EngineState: --------------------
2022-11-11_17_50_41_238: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_41_554: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_41_555: EngineState: Unit time 506 CPU ticks 3.011904716 uS
2022-11-11_17_50_41_555: EngineState: Total pulses 15906
2022-11-11_17_50_41_557: EngineState: Last valid fast msg Status 0x9 Sig0 0x0F9 Sig1 0xF12
2022-11-11_17_50_41_563: EngineState: Restarts 49
2022-11-11_17_50_41_566: EngineState: Interval errors 162 short, 339 long
2022-11-11_17_50_41_567: EngineState: Total frames 1090 with CRC error 105 (9.633027072%)
2022-11-11_17_50_41_568: EngineState: Sync errors 27
2022-11-11_17_50_41_570: EngineState: --------------------
2022-11-11_17_50_41_572: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_41_663: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_41_664: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-11_17_50_41_664: EngineState: Total pulses 17537
2022-11-11_17_50_41_666: EngineState: Last valid fast msg Status 0x8 Sig0 0x0F9 Sig1 0xF4C
2022-11-11_17_50_41_672: EngineState: Restarts 49
2022-11-11_17_50_41_674: EngineState: Interval errors 176 short, 390 long
2022-11-11_17_50_41_677: EngineState: Total frames 1189 with CRC error 113 (9.503785120%)
2022-11-11_17_50_41_678: EngineState: Sync errors 29
2022-11-11_17_50_41_681: EngineState: --------------------
2022-11-11_17_50_41_683: EngineState: confirmation_sentinfo:8
[... 346 lines cut ...]
2022-11-11_17_50_49_220: EngineState: --------------------
2022-11-11_17_50_49_221: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_49_322: EngineState: WARNING: interpolatelua: why X
2022-11-11_17_50_49_430: EngineState: ---- SENT ch 0 ----
2022-11-11_17_50_49_431: EngineState: Unit time 648 CPU ticks 3.857142912 uS
2022-11-11_17_50_49_431: EngineState: Total pulses 302
2022-11-11_17_50_49_432: EngineState: Last valid fast msg Status 0x0 Sig0 0x0F8 Sig1 0xF8B
2022-11-11_17_50_49_433: EngineState: Restarts 58
2022-11-11_17_50_49_438: EngineState: Interval errors 1 short, 19 long
2022-11-11_17_50_49_439: EngineState: Total frames 11 with CRC error 0 (0.000000000%)
2022-11-11_17_50_49_440: EngineState: Sync errors 0
2022-11-11_17_50_49_440: EngineState: --------------------
2022-11-11_17_50_49_440: EngineState: confirmation_sentinfo:8
2022-11-11_17_50_49_759: EngineState: ---- SENT ch 0 ----
[... cut ...]
2022-11-11_17_50_49_759: EngineState: Total pulses 3641
2022-11-11_17_50_49_759: EngineState: Last valid fast msg Status 0x8 Sig0 0x0F8 Sig1 0xF01
2022-11-11_17_50_49_764: EngineState: Restarts 58
2022-11-11_17_50_49_765: EngineState: Interval errors 28 short, 76 long
2022-11-11_17_50_49_767: EngineState: Total frames 260 with CRC error 23 (8.846154240%)
2022-11-11_17_50_49_767: EngineState: Sync errors 7
2022-11-11_17_50_49_767: EngineState: --------------------
2022-11-11_17_50_49_768: EngineState: confirmation_sentinfo:8

```

rusefillc, 2022-11-11:
now idle, error rate has improved from 13% to 11%

```
8
2022-11-11_17_53_46_042: EngineState: ---- SENT ch 0 ----
2022-11-11_17_53_46_042: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-11_17_53_46_042: EngineState: Total pulses 1966787
2022-11-11_17_53_46_042: EngineState: Last valid fast msg Status 0x0 Sig0 0x231 Sig1 0xD37
2022-11-11_17_53_46_042: EngineState: Slow channels:
2022-11-11_17_53_46_043: EngineState:  ID 1: 0
2022-11-11_17_53_46_048: EngineState:  ID 44: 533
2022-11-11_17_53_46_050: EngineState:  ID 9: 248
2022-11-11_17_53_46_050: EngineState:  ID 5: 66
2022-11-11_17_53_46_051: EngineState:  ID 7: 0
2022-11-11_17_53_46_052: EngineState:  ID 3: 100
2022-11-11_17_53_46_053: EngineState:  ID 146: 2457
2022-11-11_17_53_46_054: EngineState:  ID 41: 992
2022-11-11_17_53_46_055: EngineState:  ID 42: 2192
2022-11-11_17_53_46_056: EngineState:  ID 8: 85
2022-11-11_17_53_46_057: EngineState:  ID 151: 0
2022-11-11_17_53_46_058: EngineState:  ID 147: 1625
2022-11-11_17_53_46_059: EngineState:  ID 43: 1541
2022-11-11_17_53_46_060: EngineState:  ID 10: 3648
2022-11-11_17_53_46_060: EngineState:  ID 35: 772
2022-11-11_17_53_46_060: EngineState:  ID 144: 2858
2022-11-11_17_53_46_060: EngineState: Restarts 60
2022-11-11_17_53_46_060: EngineState: Interval errors 9890 short, 7223 long
2022-11-11_17_53_46_061: EngineState: Total frames 180308 with CRC error 20613 (11.432105056%)
2022-11-11_17_53_46_061: EngineState: Sync errors 3552
2022-11-11_17_53_46_061: EngineState: --------------------
2022-11-11_17_53_46_061: EngineState: confirmation_sentinfo:8
2022-11-11_17_53_46_376: EngineState: ---- SENT ch 0 ----
2022-11-11_17_53_46_376: EngineState: Unit time 508 CPU ticks 3.023809432 uS
2022-11-11_17_53_46_378: EngineState: Total pulses 1969949
2022-11-11_17_53_46_383: EngineState: Last valid fast msg Status 0x0 Sig0 0x231 Sig1 0xD2B
2022-11-11_17_53_46_385: EngineState: Slow channels:
2022-11-11_17_53_46_386: EngineState:  ID 1: 0
2022-11-11_17_53_46_387: EngineState:  ID 44: 533
2022-11-11_17_53_46_388: EngineState:  ID 9: 248
2022-11-11_17_53_46_388: EngineState:  ID 5: 66
2022-11-11_17_53_46_388: EngineState:  ID 7: 0
2022-11-11_17_53_46_388: EngineState:  ID 3: 100
2022-11-11_17_53_46_388: EngineState:  ID 146: 2457
2022-11-11_17_53_46_388: EngineState:  ID 41: 992
2022-11-11_17_53_46_388: EngineState:  ID 42: 2192
2022-11-11_17_53_46_388: EngineState:  ID 8: 85
2022-11-11_17_53_46_388: EngineState:  ID 151: 0
2022-11-11_17_53_46_388: EngineState:  ID 147: 1625
2022-11-11_17_53_46_388: EngineState:  ID 43: 1541
2022-11-11_17_53_46_389: EngineState:  ID 10: 3648
2022-11-11_17_53_46_389: EngineState:  ID 35: 772
2022-11-11_17_53_46_389: EngineState:  ID 144: 2858
2022-11-11_17_53_46_389: EngineState: Restarts 60
2022-11-11_17_53_46_389: EngineState: Interval errors 9903 short, 7229 long
2022-11-11_17_53_46_389: EngineState: Total frames 180606 with CRC error 20638 (11.427083968%)
2022-11-11_17_53_46_389: EngineState: Sync errors 3559
2022-11-11_17_53_46_389: EngineState: --------------------
2022-11-11_17_53_46_389: EngineState: confirmation_sentinfo:8
2022-11-11_17_53_46_483: EngineState: ---- SENT ch 0 ----
2022-11-11_17_53_46_483: EngineState: Unit time 509 CPU ticks 3.029761792 uS
2022-11-11_17_53_46_483: EngineState: Total pulses 1971280
2022-11-11_17_53_46_485: EngineState: Last valid fast msg Status 0x8 Sig0 0x230 Sig1 0xD93
2022-11-11_17_53_46_492: EngineState: Slow channels:
2022-11-11_17_53_46_496: EngineState:  ID 1: 0
2022-11-11_17_53_46_499: EngineState:  ID 44: 533
2022-11-11_17_53_46_500: EngineState:  ID 9: 248
2022-11-11_17_53_46_503: EngineState:  ID 5: 66
2022-11-11_17_53_46_504: EngineState:  ID 7: 0
2022-11-11_17_53_46_506: EngineState:  ID 3: 100
2022-11-11_17_53_46_507: EngineState:  ID 146: 2457
[... 598 lines cut ...]
2022-11-11_17_53_51_645: EngineState:  ID 146: 2457
2022-11-11_17_53_51_645: EngineState:  ID 41: 992
2022-11-11_17_53_51_645: EngineState:  ID 42: 2192
2022-11-11_17_53_51_646: EngineState:  ID 8: 85
2022-11-11_17_53_51_647: EngineState:  ID 151: 0
2022-11-11_17_53_51_648: EngineState:  ID 147: 1625
2022-11-11_17_53_51_648: EngineState:  ID 43: 1541
2022-11-11_17_53_51_648: EngineState:  ID 10: 3648
2022-11-11_17_53_51_648: EngineState:  ID 35: 772
2022-11-11_17_53_51_648: EngineState:  ID 144: 2858
2022-11-11_17_53_51_648: EngineState: Restarts 60
2022-11-11_17_53_51_648: EngineState: Interval errors 10230 short, 7436 long
2022-11-11_17_53_51_654: EngineState: Total frames 186066 with CRC error 21215 (11.401867872%)
2022-11-11_17_53_51_654: EngineState: Sync errors 3682
2022-11-11_17_53_51_654: EngineState: --------------------
2022-11-11_17_53_51_654: EngineState: confirmation_sentinfo:8
2022-11-11_17_53_51_734: EngineState: WARNING: Sensor fault: Tps1 has timed out
2022-11-11_17_54_00_344: EngineState: WARNING: Injector 0 fault: open_load 
2022-11-11_17_54_00_349: EngineState: WARNING: Injector 1 fault: open_load 
2022-11-11_17_54_00_349: EngineState: WARNING: Injector 2 fault: open_load 
2022-11-11_17_54_00_350: EngineState: WARNING: Injector 3 fault: open_load 
2022-11-11_17_54_00_891: EngineState: WARNING: interpolatelua: why X
2022-11-11_17_54_01_875: EngineState: WARNING: Sensor fault: Tps1 has timed out

```

rusefillc, 2022-11-11:
random middle position

```
2022-11-11_17_59_03_690: EngineState: ---- SENT ch 0 ----
2022-11-11_17_59_03_691: EngineState: Unit time 508 CPU ticks 3.023809432 uS
2022-11-11_17_59_03_691: EngineState: Total pulses 116530
2022-11-11_17_59_03_691: EngineState: Last valid fast msg Status 0xC Sig0 0x813 Sig1 0x7CA
2022-11-11_17_59_03_696: EngineState: Slow channels:
2022-11-11_17_59_03_696: EngineState:  ID 146: 2457
2022-11-11_17_59_03_697: EngineState:  ID 1: 0
2022-11-11_17_59_03_703: EngineState:  ID 144: 2858
2022-11-11_17_59_03_706: EngineState:  ID 147: 1625
2022-11-11_17_59_03_707: EngineState:  ID 43: 1541
2022-11-11_17_59_03_709: EngineState:  ID 148: 17
2022-11-11_17_59_03_712: EngineState:  ID 35: 775
2022-11-11_17_59_03_713: EngineState:  ID 149: 0
2022-11-11_17_59_03_715: EngineState:  ID 8: 84
2022-11-11_17_59_03_717: EngineState:  ID 44: 533
2022-11-11_17_59_03_719: EngineState:  ID 9: 248
2022-11-11_17_59_03_720: EngineState:  ID 145: 2387
2022-11-11_17_59_03_721: EngineState: Restarts 64
2022-11-11_17_59_03_721: EngineState: Interval errors 590 short, 320 long
2022-11-11_17_59_03_721: EngineState: Total frames 10778 with CRC error 965 (8.953423488%)
2022-11-11_17_59_03_721: EngineState: Sync errors 221
2022-11-11_17_59_03_721: EngineState: --------------------
2022-11-11_17_59_03_721: EngineState: confirmation_sentinfo:8
2022-11-11_17_59_03_721: EngineState: WARNING: interpolatelua: why X
2022-11-11_17_59_04_234: EngineState: WARNING: Sensor fault: Tps1 has timed out
2022-11-11_17_59_04_675: EngineState: ---- SENT ch 0 ----
2022-11-11_17_59_04_675: EngineState: Unit time 511 CPU ticks 3.041666748 uS
2022-11-11_17_59_04_675: EngineState: Total pulses 127547
2022-11-11_17_59_04_675: EngineState: Last valid fast msg Status 0x0 Sig0 0x813 Sig1 0x7FF
2022-11-11_17_59_04_675: EngineState: Slow channels:
2022-11-11_17_59_04_675: EngineState:  ID 146: 2457
2022-11-11_17_59_04_677: EngineState:  ID 1: 0
2022-11-11_17_59_04_680: EngineState:  ID 144: 2858
2022-11-11_17_59_04_680: EngineState:  ID 147: 1625
2022-11-11_17_59_04_681: EngineState:  ID 43: 1541
2022-11-11_17_59_04_683: EngineState:  ID 148: 17
2022-11-11_17_59_04_684: EngineState:  ID 35: 775
2022-11-11_17_59_04_685: EngineState:  ID 149: 0
2022-11-11_17_59_04_686: EngineState:  ID 8: 84
2022-11-11_17_59_04_686: EngineState:  ID 44: 533
2022-11-11_17_59_04_687: EngineState:  ID 9: 248
2022-11-11_17_59_04_688: EngineState:  ID 145: 2387
2022-11-11_17_59_04_689: EngineState: Restarts 64
2022-11-11_17_59_04_689: EngineState: Interval errors 645 short, 353 long
2022-11-11_17_59_04_690: EngineState: Total frames 11796 with CRC error 1060 (8.986097344%)
2022-11-11_17_59_04_690: EngineState: Sync errors 237
2022-11-11_17_59_04_691: EngineState: --------------------
2022-11-11_17_59_04_692: EngineState: confirmation_sentinfo:8
2022-11-11_17_59_05_554: EngineState: ---- SENT ch 0 ----
2022-11-11_17_59_05_555: EngineState: Unit time 507 CPU ticks 3.017857074 uS
2022-11-11_17_59_05_555: EngineState: Total pulses 137368
2022-11-11_17_59_05_557: EngineState: Last valid fast msg Status 0x0 Sig0 0x814 Sig1 0x79D
2022-11-11_17_59_05_564: EngineState: Slow channels:
2022-11-11_17_59_05_566: EngineState:  ID 146: 2457
2022-11-11_17_59_05_567: EngineState:  ID 1: 0
2022-11-11_17_59_05_571: EngineState:  ID 144: 2858
2022-11-11_17_59_05_574: EngineState:  ID 147: 1625
2022-11-11_17_59_05_576: EngineState:  ID 43: 1541
2022-11-11_17_59_05_577: EngineState:  ID 148: 17
2022-11-11_17_59_05_577: EngineState:  ID 35: 775
2022-11-11_17_59_05_577: EngineState:  ID 149: 0
2022-11-11_17_59_05_577: EngineState:  ID 8: 84
2022-11-11_17_59_05_578: EngineState:  ID 44: 533
2022-11-11_17_59_05_578: EngineState:  ID 9: 248
2022-11-11_17_59_05_578: EngineState:  ID 145: 2387
2022-11-11_17_59_05_578: EngineState: Restarts 64
2022-11-11_17_59_05_578: EngineState: Interval errors 697 short, 384 long
[... 41 lines cut ...]
2022-11-11_17_59_06_886: EngineState:  ID 44: 533
2022-11-11_17_59_06_891: EngineState:  ID 9: 248
2022-11-11_17_59_06_894: EngineState:  ID 145: 2387
2022-11-11_17_59_06_895: EngineState: Restarts 64
2022-11-11_17_59_06_896: EngineState: Interval errors 771 short, 434 long
2022-11-11_17_59_06_898: EngineState: Total frames 14046 with CRC error 1285 (9.148511888%)
2022-11-11_17_59_06_899: EngineState: Sync errors 283
2022-11-11_17_59_06_899: EngineState: --------------------
2022-11-11_17_59_06_900: EngineState: confirmation_sentinfo:8

```

So for sig1+sig2 we get 10% error easy

idle
0x230+DB2=0xFE2
0x230+D0B=0xF3B    5% error

full closed
FB+F6C=0x1067=4199   3% error

random position
0x81c+0x7a9=0xFC5

let me add new configuration setting sentErrorRate

rusefillc, 2022-11-12:
master with https://github.com/rusefi/rusefi/pull/4774

forced closed
```
2022-11-12_17_53_44_564: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_44_565: EngineState: Unit time 508 CPU ticks 3.023809432 uS
2022-11-12_17_53_44_565: EngineState: Total pulses 72891
2022-11-12_17_53_44_565: EngineState: Last valid fast msg Status 0x0 Sig0 0x0FA Sig1 0xF35
2022-11-12_17_53_44_566: EngineState: Slow channels:
2022-11-12_17_53_44_566: EngineState:  ID 1: 0
2022-11-12_17_53_44_569: EngineState:  ID 9: 248
2022-11-12_17_53_44_569: EngineState:  ID 41: 992
2022-11-12_17_53_44_570: EngineState:  ID 146: 2457
2022-11-12_17_53_44_570: EngineState:  ID 5: 66
2022-11-12_17_53_44_571: EngineState: Restarts 38
2022-11-12_17_53_44_571: EngineState: Interval errors 462 short, 424 long
2022-11-12_17_53_44_571: EngineState: Total frames 6427 with CRC error 707 (11.000466346%)
2022-11-12_17_53_44_572: EngineState: Total slow channel messages 9 with crc6 errors 0 (0.000000000%)
2022-11-12_17_53_44_573: EngineState: Sync errors 184
2022-11-12_17_53_44_573: EngineState: --------------------
2022-11-12_17_53_44_574: EngineState: confirmation_sentinfo:8
2022-11-12_17_53_45_232: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_45_232: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-12_17_53_45_232: EngineState: Total pulses 1069
2022-11-12_17_53_45_232: EngineState: Last valid fast msg Status 0x0 Sig0 0x0FB Sig1 0xF24
2022-11-12_17_53_45_233: EngineState: Restarts 43
2022-11-12_17_53_45_233: EngineState: Interval errors 10 short, 40 long
2022-11-12_17_53_45_237: EngineState: Total frames 57 with CRC error 5 (8.771929728%)
2022-11-12_17_53_45_237: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-12_17_53_45_238: EngineState: Sync errors 2
2022-11-12_17_53_45_239: EngineState: --------------------
2022-11-12_17_53_45_240: EngineState: confirmation_sentinfo:8
2022-11-12_17_53_45_792: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_45_792: EngineState: Unit time 509 CPU ticks 3.029761792 uS
2022-11-12_17_53_45_792: EngineState: Total pulses 7302
2022-11-12_17_53_45_792: EngineState: Last valid fast msg Status 0x0 Sig0 0x0FB Sig1 0xF1B
2022-11-12_17_53_45_793: EngineState: Restarts 43
2022-11-12_17_53_45_797: EngineState: Interval errors 57 short, 239 long
2022-11-12_17_53_45_797: EngineState: Total frames 432 with CRC error 45 (10.416666976%)
2022-11-12_17_53_45_798: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-12_17_53_45_799: EngineState: Sync errors 12
2022-11-12_17_53_45_800: EngineState: --------------------
2022-11-12_17_53_45_800: EngineState: confirmation_sentinfo:8
2022-11-12_17_53_45_904: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_45_904: EngineState: Unit time 510 CPU ticks 3.035714388 uS
2022-11-12_17_53_45_904: EngineState: Total pulses 9328
2022-11-12_17_53_45_904: EngineState: Last valid fast msg Status 0x0 Sig0 0x0FB Sig1 0xFE7
2022-11-12_17_53_45_905: EngineState: Restarts 43
2022-11-12_17_53_45_909: EngineState: Interval errors 74 short, 293 long
2022-11-12_17_53_45_910: EngineState: Total frames 565 with CRC error 63 (11.150442128%)
2022-11-12_17_53_45_912: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-12_17_53_45_913: EngineState: Sync errors 17
2022-11-12_17_53_45_914: EngineState: --------------------
2022-11-12_17_53_45_915: EngineState: confirmation_sentinfo:8
2022-11-12_17_53_46_240: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_46_240: EngineState: Unit time 709 CPU ticks 4.220238208 uS
2022-11-12_17_53_46_240: EngineState: Total pulses 12302
2022-11-12_17_53_46_240: EngineState: Last valid fast msg Status 0xC Sig0 0x0FB Sig1 0xFA8
2022-11-12_17_53_46_241: EngineState: Restarts 43
2022-11-12_17_53_46_241: EngineState: Interval errors 96 short, 418 long
2022-11-12_17_53_46_242: EngineState: Total frames 715 with CRC error 79 (11.048951148%)
2022-11-12_17_53_46_242: EngineState: Total slow channel messages 0 with crc6 errors 0 (NaN%)
2022-11-12_17_53_46_246: EngineState: Sync errors 23
2022-11-12_17_53_46_247: EngineState: --------------------
2022-11-12_17_53_46_247: EngineState: confirmation_sentinfo:8
2022-11-12_17_53_46_351: EngineState: ---- SENT ch 0 ----
2022-11-12_17_53_46_351: EngineState: Unit time 509 CPU ticks 3.029761792 uS
2022-11-12_17_53_46_351: EngineState: Total pulses 14074
2022-11-12_17_53_46_351: EngineState: Last valid fast msg Status 0x0 Sig0 0x0FB Sig1 0xFB5
2022-11-12_17_53_46_352: EngineState: Restarts 43
[... 976 lines cut ...]
2022-11-12_17_55_29_137: EngineState: Slow channels:
2022-11-12_17_55_29_140: EngineState:  ID 1: 0
2022-11-12_17_55_29_142: EngineState:  ID 150: 0
2022-11-12_17_55_29_145: EngineState:  ID 6: 3
2022-11-12_17_55_29_145: EngineState:  ID 146: 2457
2022-11-12_17_55_29_145: EngineState:  ID 10: 3648
2022-11-12_17_55_29_145: EngineState:  ID 41: 992
2022-11-12_17_55_29_145: EngineState:  ID 42: 2192
2022-11-12_17_55_29_145: EngineState:  ID 43: 1541
2022-11-12_17_55_29_145: EngineState:  ID 35: 777
2022-11-12_17_55_29_145: EngineState:  ID 44: 533
2022-11-12_17_55_29_145: EngineState:  ID 145: 2387
2022-11-12_17_55_29_145: EngineState:  ID 7: 0
2022-11-12_17_55_29_145: EngineState:  ID 144: 2858
2022-11-12_17_55_29_145: EngineState:  ID 148: 17
2022-11-12_17_55_29_145: EngineState:  ID 3: 100
2022-11-12_17_55_29_145: EngineState:  ID 8: 85
2022-11-12_17_55_29_146: EngineState: Restarts 75
2022-11-12_17_55_29_146: EngineState: Interval errors 5037 short, 3189 long
2022-11-12_17_55_29_146: EngineState: Total frames 88606 with CRC error 9377 (10.582804672%)
2022-11-12_17_55_29_146: EngineState: Total slow channel messages 117 with crc6 errors 6 (5.128205296%)
2022-11-12_17_55_29_146: EngineState: Sync errors 2138
2022-11-12_17_55_29_149: EngineState: --------------------
2022-11-12_17_55_29_152: EngineState: confirmation_sentinfo:8
```

dron0gus, 2022-11-13:
Ok. Now we have 100% success with data captured with logic analyzer and same SENT decoder.
That makes me think about HW problems.
May be analog input hacked for SENT is not a good solution?
Can you make a oscillogram photo from this point?
[image]

rusefillc, 2022-11-13:
[image]
[image]
[image]

dron0gus, 2022-12-21:
@rusefillc are we happy? Closing?

rusefillc, 2022-12-21:
Happy!!!

closed 2022-12-21 by rusefillc