rusefi/rusefi #4748: Lua tab "Burn to ECU" button causes MRE to hang up

rusefillc, 2022-11-07:
MRE
snapshot version of firmware
go to Lua tab, hit "Burn to ECU"

actual result: MRE blue LED no longer blinks
expected result: happy MRE

additional info: "writeconfig" text command works
that makes me wonder if we have a stack issue with different stack sizes on different threads?

rusefillc, 2022-11-07:
@mck1117 @dron0gus any availability?

dron0gus, 2022-11-07:
I did not observe such issue while working on SENT and LUA on MRE. Can check with latest greatest master today.

mck1117, 2022-11-07:
can you share your script so I can try a repro? I have debugger set up always.

rusefillc, 2022-11-07:
@dron0gus @mck1117 looks like script does not matter

I get this both with

```

function onTick()
end

```

and

```

tpsSensor = Sensor.new("Tps1")

setTickRate(100)

function setTwoBytes(data, offset, value)
	value = math.floor(value)
	data[offset + 2] = value >> 8
	data[offset + 1] = value & 0xff
end

R52_RPM = 0x316

data = { 0, 0, 0, 00, 0, 0 , 0, 0 }

rpm = 7000

function onTick()
	rpm = rpm + 30
	if rpm > 8000 then
		rpm = 0
	end

	setTwoBytes(data, 2, rpm * 6.39)

	txCan(1, R52_RPM, 0, data)

	-- 1 ABS and warm-up light but no tach
	-- 3 blinker and alive
	-- 4 ABS and alive
	--
	status2 = 2

	-- status3 = 0xC0
	status3 = 0x0

	status4 = 0x0
	-- status4 = 0xC0
	-- with ABS with blinker with warm-up light
	-- status4 = 0xFF

	txCan(1, 0x61f, 0, { 0, 0, status2, status3, status, 0 , 0, 0 })

	sentValue = getSentValue(1)
	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)

	tpsSensor : set(tpsValue )
-- 	print('sentValue ' ..sentValue..' tps ' ..tpsValue)
end

```

rusefillc, 2022-11-07:
PS: sometimes same "Burn to ECU" just does nothing. Looks like the important thing is to make any change to the script or just give console some time to start?

rusefillc, 2022-11-12:
Similar issue is now happening in TS as well
Steps to reproduce:

Start with https://rusefi.com/online/view.php?msq=1292 tune - that one I am able to modify in TS
Enable SENT on MRE pin 23, Burn
After that any consecutive burn via TS like changing ETB PID would hang up MRE

dron0gus, 2022-11-13:
https://github.com/rusefi/rusefi/pull/4776

closed 2022-11-13 by rusefillc