rusefi/rusefi #4760: RPM correction/multiplier for Accel Enrich

ElDominio, 2022-11-08:
add a 2D RPM correction table to transient enrich, to be able to lower the amount of enrich applied at upper RPMs (since engines in motion usually need much less enrichment). 

A table like this
[image]

maybe even with 4 cells is enough. 1.0 at 1000RPM, .9 at 3000rpm, .5 at 5000RPM, 0 at 7000RPM.

mck1117, 2022-11-08:
how much time do you spend at 7k rpm while below 30c coolant?

ElDominio, 2022-11-08:
no coolant just showed that to show what a 2D table looks like. Coolant would be replaced by RPM

mck1117, 2022-11-08:
yes I understand the request - I'm asking why you think it's necessary

ElDominio, 2022-11-08:
dding 3ms on a transient at 5.500rpm usually just makes an engine lug

mck1117, 2022-11-08:
oh duh the snip of a random table confused me

ElDominio, 2022-11-08:
[image]

ElDominio, 2022-11-16:
Also, a CLT-based correction on enrich (more enrich added when colder)  would also be good. Some ecus have a "Cold/Hot" temp, where the Hot temp is the normal, and the cold temp uses a multiplier.

Example:
Cold = 100F ; Multiplier = 1.3
Hot = 150F

(up to 100F, it uses 1.3 as a multiplier, then interpolates between 100F and 150F to become 1x multiplier.)

ElDominio, 2022-12-13:
@mck1117 @rusefillc  any progress?

rusefillc, 2022-12-13:
Some progress

[image]

commit 743bdad7 (2022-12-15): RPM correction/multiplier for Accel Enrich fix #4760
commit a0c2689a (2022-12-15): RPM correction/multiplier for Accel Enrich fix #4760
closed 2022-12-15 by rusefillc