rusefi/rusefi #4772: HW CI is not reliable enough

rusefillc, 2022-11-12:
Humans are fucking expensive, we commit a lot. 20 seconds waste of flaky test multiplies.

rusefillc, 2022-11-12:
for example https://github.com/rusefi/rusefi/actions/runs/3353650060/jobs/5556605696#step:16:1340

[image]

rusefillc, 2022-11-12:
@mck1117 any chance you could help?

https://github.com/rusefi/rusefi/actions/runs/3451774157/jobs/5761229627#step:16:1154

[image]

rusefillc, 2022-11-16:
https://github.com/rusefi/rusefi/actions/runs/3481584396/jobs/5822865985#step:16:1151 @mck1117

rusefillc, 2022-11-20:
@nmschulte this one is a functional defect

not sure if proteus HW_CI setup has external wires like discovery setup has

https://github.com/rusefi/rusefi/actions/runs/3505546092/jobs/5871917337

mck1117, 2022-11-20:
They both have external wires. The comments in the relevant Java files tell what mods are needed.

mck1117, 2022-11-20:
Those fails all look like the same thing - comms time out, which may mean that the ECU is locked up. What's the failure rate like?

rusefillc, 2022-11-20:
i want to say 5%-ish?

rusefillc, 2023-01-06:
@mck1117 still 5% failure rate

https://github.com/rusefi/rusefi/actions/runs/3858743431/jobs/6577597618

rusefillc, 2023-01-11:
@mck1117 https://github.com/rusefi/rusefi/actions/runs/3888361986/jobs/6635598767

rusefillc, 2023-01-29:
@mck1117 https://github.com/rusefi/rusefi/actions/runs/4036174871/jobs/6938589005#step:18:726

mck1117, 2023-02-20:
if we increase that 500ms timeout, does it get better?

rusefillc, 2023-02-20:
@mck1117 what do we increase to 500ms exactly? Everything seems to be at least 500ms already?

[image]

mck1117, 2023-02-20:
[image]

rusefillc, 2023-02-20:
Sorry did not read it right

mck1117, 2023-02-20:
but it is strange because just above that it says 5000ms

rusefillc, 2023-02-20:
@mck1117 i believe different method invocations come with different timeouts

rusefillc, 2024-11-22:
https://github.com/rusefi/rusefi/actions/runs/11978464036

```
2024-11-22_14_03_18_263<EOT>: Sleeping 5 seconds
I 241122 140318.265 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: Emulating position sensor(s). RPM=6000
I 241122 140318.265 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: confirmation_rpm 6000:8
2024-11-22_14_03_23_264<EOT>: Sleeping 40 seconds
I 241122 140403.264 [main] IoUtil - Sending command [threadsinfo]
I 241122 140403.264 [ECU Commands Queue] BinaryProtocol - Sending [threadsinfo]
I 241122 140403.346 [ECU Commands Queue] IoUtil - Got confirmation in 82ms
I 241122 140403.346 [main] IoUtil - Command [threadsinfo] executed in 82
Ran 22 tests total.
Ignored 0 tests.
Failed 1 tests.
testV12(com.rusefi.f4discovery.HighRevTest): Failed Got 3998.0 while trying to stay at 6000 after 3 seconds
java.lang.IllegalStateException: Failed Got 3998.0 while trying to stay at 6000 after 3 seconds
Test failed: testV12(com.rusefi.f4discovery.HighRevTest) because Failed Got 3998.0 while trying to stay at 6000 after 3 seconds
Exception:
	at com.rusefi.functional_tests.EcuTestHelper.lambda$static$0(EcuTestHelper.java:28)
	at com.rusefi.functional_tests.EcuTestHelper.assertRpmDoesNotJump(EcuTestHelper.java:57)
	at com.rusefi.f4discovery.HighRevTest.runHighRevTest(HighRevTest.java:31)
	at com.rusefi.f4discovery.HighRevTest.testV12(HighRevTest.java:19)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:59)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:56)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.BlockJUnit4ClassRunner$1.evaluate(BlockJUnit4ClassRunner.java:100)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:366)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:103)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:63)
	at org.junit.runners.ParentRunner$4.run(ParentRunner.java:331)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:79)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:329)
	at org.junit.runners.ParentRunner.access$100(ParentRunner.java:66)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:293)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:413)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:27)
	at org.junit.runners.ParentRunner$4.run(ParentRunner.java:331)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:79)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:329)
	at org.junit.runners.ParentRunner.access$100(ParentRunner.java:66)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:293)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:413)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:115)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:105)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:94)
	at com.rusefi.CmdJUnitRunner.runHardwareTest(CmdJUnitRunner.java:26)
	at com.rusefi.CmdJUnitRunner.runHardwareTestAndExit(CmdJUnitRunner.java:10)
	at com.rusefi.HwCiF4Discovery.main(HwCiF4Discovery.java:40)
Have we left any non-daemon threads?
Name            	 State           	 Priority        	 isDaemon
Common-Cleaner  	 TIMED_WAITING   	 8               	 true
Finalizer       	 WAITING         	 8               	 true
main            	 RUNNABLE        	 5               	 false
ECU Commands Queue 	 WAITING         	 5               	 true
```