rusefi/rusefi #4773: rusEFI console: Seconds of nothing between "connect" and console

rusefillc, 2022-11-12:
We have no UI for a few seconds between "connect" and UI

we shall do nicer

dron0gus, 2022-11-13:
Few? Hold my beer
```
...
I 221113 180600.022 [AWT-EventQueue-0] ConsoleUI - Console 20221016
I 221113 180600.023 [AWT-EventQueue-0] ConsoleUI - Hardware: F4
I 221113 180600.023 [AWT-EventQueue-0] LinkManager - LinkManager: Starting ttyACM1
ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
ANTLR Tool version 4.5 used for code generation does not match the current runtime version 4.9.3
line 672:94 mismatched input ')' expecting {'(', '[', '+', '-', '*', '/', '%', '^', '&', '|', '<', '>', '==', '!=', '<=', '>=', AndAnd, OrOr, '++', '--', ',', '->*', '->', '?', '.', '.*'}
line 133:22 no viable alternative at input '__attribute__((weak))void'
line 182:4 no viable alternative at input 'fail("HW_CHECK_ALWAYS_STIMULATE required to have self-stimulation")#endifint'
line 182:4 missing ';' at 'int'
line 699:28 mismatched input ')' expecting {IntegerLiteral, CharacterLiteral, FloatingLiteral, StringLiteral, BooleanLiteral, PointerLiteral, UserDefinedLiteral, '(', '[', '+', '-', '*', '/', '%', '^', '&', '|', '<', '>', '==', '!=', '<=', '>=', AndAnd, OrOr, '++', '--', ',', '->*', '->', '?', '.', '.*'}
I 221113 180607.993 [AWT-EventQueue-0] MessagesCentral - postMessage ConsoleUI: COMPOSITE_OFF_RPM=300
I 221113 180607.995 [AWT-EventQueue-0] AuthTokenPanel - Auth token from settings: 
I 221113 180608.646 [AWT-EventQueue-0] AuthTokenPanel - Ignoring java.awt.datatransfer.UnsupportedFlavorException: Unicode String
I 221113 180609.863 [AWT-EventQueue-0] Launcher - onWindowOpened
I 221113 180609.865 [AWT-EventQueue-0] StreamConnector - StreamConnector: connecting
I 221113 180609.865 [AWT-EventQueue-0] StreamConnector - scheduleOpening
I 221113 180609.866 [communication executor1] StreamConnector - scheduleOpening>openPort
I 221113 180609.883 [communication executor1] SerialIoStream - Using com.fazecast.jSerialComm 2.9.2
I 221113 180609.883 [communication executor1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM1
I 221113 180609.918 [communication executor1] BinaryProtocol - Got rusEFI 2022.11.12.mre_f4.snap_28755 signature
I 221113 180609.926 [communication executor1] BinaryProtocol - current_configuration.rusefi_binary Local cache CRC e79a463b

I 221113 180609.931 [communication executor1] BinaryProtocol - rusEFI says tune CRC32 0xe79a463b -409319877

I 221113 180609.932 [communication executor1] BinaryProtocol - rusEFI says tune CRC16 0x463b 17979

I 221113 180609.932 [communication executor1] BinaryProtocol - Got configuration from controller.
...
```
9.9 seconds!

ElDominio, 2022-11-14:
Can that log show like a nice status bar? that would be nice

rusefillc, 2022-11-27:
Things should be 4 seconds quicker now, follow-up ticket https://github.com/rusefi/rusefi/issues/4842

commit 0fb3bc9e (2022-11-27): rusEFI console: Seconds of nothing between "connect" and console fix #4773
closed 2022-11-27 by rusefillc