rusefi/rusefi #4778: New boost control methods

ElDominio, 2022-11-13:
I think boost control should have a few changes implemented.
First, open lood should have selcteable Y axis between TPS and MAP. Using MAP allows to do pseudo closed loop boost ( I KNOW IT IS NOT TRUE CLOSED LOOP) but allows for quick somewhat reliable results.

There should be a semi-closed loop boost mode (this is how many, many ECU's do) that uses a base duty lookup table, then uses PID to do the final adjustments to be as close as possible to the actual boost target.

I'll copy paste what a certain manual says about boost control and the three "stages" they use:
```

Closed Loop Control

Closed loop control involves applying a duty cycle to the wastegate solenoid based on predefined values, tables and the current boost pressure. This method compares the current boost pressure to the target boost pressure and corrects the duty cycle accordingly to achieve the desired boost.

More Specifically operation of the closed loop boost control system has been divided into three separate stages (Stage 1, Stage 2 and Stage 3). Each stage has a specific purpose, and it is important to setup the transition between stages and the control within each stage correctly. Details of each stage are outlined as follows:

 

Stage 1

Stage 1 is the first stage after closed loop boost control becomes active. Stage 1 applies a fixed duty cycle to the wastegate solenoid, typically 80 - 90%. The purpose of this stage is to spool the turbo as fast as possible by holding the wastegate firmly closed. This stage remains active until Stage 2 becomes active. Please note that when MAP and engine speed (RPM) levels are falling, stage 1 will only re-activate once the MAP or engine speed (RPM) lockout level has been reached.

 

Stage 2

Stage 2 becomes active once boost pressure comes within a certain range of the target pressure (set using the Stage 2 ON setting). Stage 2 is the first stage of closed loop control. Stage 2 uses proportional and rate of change (P and D) control along with a base duty cycle table. The purpose of this stage is to control the approach to the Boost Target pressure. The duty cycle applied to the wastegate solenoid is the duty cycle from the base table plus the duty cycle from the PD controller. This stage remains active until Stage 3 becomes active.

 

Stage 3

Stage 3 becomes active once boost pressure comes within a certain range of the target pressure (set using the Stage 3 On setting). Stage 3 is the final stage of closed loop control. Stage 3 uses proportional and offset error (P and I) control along with a fixed base duty cycle. Depending on the Base DC Mode setting, the base duty cycle is either fixed from the final value in Stage 2 or continually updating from the base table. The duty cycle applied to the wastegate solenoid is the base duty cycle plus the duty cycle from the PI controller. This stage remains active until boost pressure is outside its allowed range.

 

```

[image]

ElDominio, 2024-02-23:
This is mee still wanting MAP as y axis

rusefillc, 2024-05-08:
current state
[image]

rusefillc, 2024-05-08:
boostControlOpenLookYAxis

@ElDominio would it be OK for us to make it much more flexibility not just TPS vs MAP?

[image]

rusefillc, 2024-05-08:
kind of related to https://github.com/rusefi/rusefi/pull/6480

ElDominio, 2024-05-09:
Sure

kifir23917, 2024-05-10:
```
	expected<float> xAxisValue = readGppwmChannel(m_config->rpmAxis);
	expected<float> yAxisValue = readGppwmChannel(m_config->loadAxis);
```

kifir23917, 2024-05-13:
- Zero
- TPS -> % 
- MAP KPa
- CLT C
- IAT C
- Fuel Load ?
- Ignition Load ?
- Aux Temp 1 C

kifir23917, 2024-05-13:
[image]

Am I correct that we do not want to update numbers on Y axis depending on the value selected in Y Axis drop-down list?

rusefillc, 2024-05-13:
> Am I correct that we do not want to update numbers on Y axis depending on the value selected in `Y Axis` drop-down list?

correct, touching the axis values is out of scope.

mck1117, 2024-05-14:
JFYI if you put MAP on the Y axis, it turns it in to a second closed loop table. It's by definition not open loop if you do that.

ElDominio, 2024-05-14:
I know; it's not PID but it does work oaky for most quarter mile applications, and definitely easier to explain to people. True closed loop is better but it's sometimes not what I need in the moment

closed 2024-05-15 by rusefillc

rusefillc, 2024-05-26:
see also #5027

commit f4b3b00d (2024-05-15): interpret value from openLoopMap according to boostOpenLoopYAxis constant (fixes #4778)