rusefi/rusefi #4785: Something with Lua tickrate

rusefillc, 2022-11-15:
```
i = 0
setTickRate(100)
start = Timer.new()
start : reset()
function onTick()
	i = i + 1
	sinceStart = start : getElapsedSeconds()
	print ("actual rate " ..(i / sinceStart) )
end

2022-11-14_20_36_19_836: EngineState: LUA: actual rate 100.2669677
2022-11-14_20_36_19_838: EngineState: LUA: actual rate 100.2582321
2022-11-14_20_36_19_841: EngineState: LUA: actual rate 100.2655792
2022-11-14_20_36_19_849: EngineState: LUA: actual rate 100.2643280
2022-11-14_20_36_19_849: EngineState: LUA: actual rate 100.2635269
2022-11-14_20_36_19_851: EngineState: LUA: actual rate 100.2626190
2022-11-14_20_36_19_852: EngineState: LUA: actual rate 100.2607879

```

mck1117, 2022-11-15:
is there a concern? reminder that the scheduler has limited precision about sleeps

rusefillc, 2022-11-15:
No issue.

For whatever reason it seemed to me that full script had performance issue but does not seem to be the case

```

tpsSensor = Sensor.new("Tps1")
tpsSensor : setRedundant(true)

ppsSensor = Sensor.new("AcceleratorPedal")
ppsSensor : setRedundant(true)

Tps1Primary = Sensor.new("Tps1Primary")
AcceleratorPedalPrimary = Sensor.new("AcceleratorPedalPrimary")

setTickRate(100)
-- huh? with 100Hz tick rate why do we need such huge timeout?
ppsSensor : setTimeout(1000)
tpsSensor : setTimeout(1000)
restartEtb()

function setTwoBytes(data, offset, value)
	value = math.floor(value)
	data[offset + 2] = value >> 8
	data[offset + 1] = value & 0xff
end

-- CAN Rx Channels
IDRIVE_BUTTONS = 0x267
IDRIVE_ROTARY = 0x264
-- CAN Tx Channels
IDRIVE_BRIGHTNESS = 0x202
IDRIVE_HEARBEAT = 0x560
IDRIVE_ROTARY_INIT = 0x273
canRxAdd(IDRIVE_ROTARY) -- rotation
canRxAdd(IDRIVE_BUTTONS) -- buttons

brighness = 0

R52_RPM = 0x316

data = { 0, 0, 0, 00, 0, 0 , 0, 0 }

rpm = 7000

i = 0

-- One time rotary initialisation

txCan(1, IDRIVE_ROTARY_INIT, 0, { 0x1D, 0xE1, 0x00, 0xF0, 0xFF, 0x7F, 0xDE, 0x04 })

clicks = 5

function onCanRx(bus, id, dlc, data)
	-- 	print('got CAN id=' ..id ..' dlc=' ..dlc)

	if id == IDRIVE_BUTTONS then
		-- 		print('got buttons @4=' ..data[4] ..' @6=' ..data[6])

		if data[4] == 1 or data[4] == 2 then
			print('got hold or release @4=' ..data[4] ..' @6=' ..data[6])

			if data[6] == 2 and brighness > 0 then
				brighness = brighness - 1
				print('brighness ' ..brighness)
			end

			if data[6] == 4 and brighness < 15 then
				brighness = brighness + 1
				print('brighness ' ..brighness)
			end
		end
	end

	if id == IDRIVE_ROTARY then
		if data[5] == 0x80 then
			clicks = data[4]
			print('rotary turned right ' ..data[4] ..' clicks')
		end

		if data[5] == 0x7f then
			print('rotary turned left ' ..255 - data[4] ..' clicks')
		end
	end

end

i = 0
setTickRate(100)
start = Timer.new()
start : reset()
function onTick()
	i = i + 1
	sinceStart = start : getElapsedSeconds()
	print ("actual rate " ..(i / sinceStart) )

	-- wake up, need to send once every 1.5 second but we can send more often because why not
	txCan(1, IDRIVE_HEARBEAT, 0, { 3 })
	txCan(1, IDRIVE_BRIGHTNESS, 0, { brighness * 16 + 0xd })

	rpm = getSensor("RPM")

	if i == 3 then
		print('Postponed restart')
		restartEtb()
		-- 		i = 0
	end
	-- 	print('Period ' ..i)

	setTwoBytes(data, 2, rpm * 6.39)

	txCan(1, R52_RPM, 0, data)

	-- 1 ABS and warm-up light but no tach
	-- 3 blinker and alive
	-- 4 ABS and alive
	--
	status2 = 2

	-- status3 = 0xC0
	status3 = 0x0

	status4 = 0x0
	-- status4 = 0xC0
	-- with ABS with blinker with warm-up light
	-- status4 = 0xFF

	txCan(1, 0x61f, 0, { 0, 0, status2, status3, status, 0 , 0, 0 })

	sentValue = getSentValue(1)
	-- 	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)
	tpsValue = interpolate(0.11, 100, 0.89, 0, sentValue)

	if tpsValue == tpsValue then
		tpsSensor : set(tpsValue )
		Tps1Primary : set(tpsValue )
		-- 		print('sentValue ' ..sentValue..' tps ' ..tpsValue)
	else
--		print('sentValue is Nan')
	end

	ppsSensor : set(5 * clicks)

end

2022-11-14_20_40_11_599: EngineState: LUA: actual rate 88.2001495
2022-11-14_20_40_11_600: EngineState: LUA: actual rate 88.2007141
2022-11-14_20_40_11_698: EngineState: LUA: actual rate 88.1998672
2022-11-14_20_40_11_698: EngineState: LUA: actual rate 88.2004013
2022-11-14_20_40_11_698: EngineState: LUA: actual rate 88.2007293
2022-11-14_20_40_11_698: EngineState: LUA: actual rate 88.2011795
2022-11-14_20_40_11_699: EngineState: LUA: actual rate 88.2015075
2022-11-14_20_40_11_700: EngineState: LUA: actual rate 88.2018508
2022-11-14_20_40_11_705: EngineState: LUA: actual rate 88.2020416
2022-11-14_20_40_11_707: EngineState: LUA: actual rate 88.2025070
2022-11-14_20_40_11_708: EngineState: LUA: actual rate 88.2030639
2022-11-14_20_40_11_805: EngineState: LUA: actual rate 88.2034912
2022-11-14_20_40_11_805: EngineState: LUA: actual rate 88.2037277
2022-11-14_20_40_11_805: EngineState: LUA: actual rate 88.2042770
2022-11-14_20_40_11_805: EngineState: LUA: actual rate 88.2046203
2022-11-14_20_40_11_806: EngineState: LUA: actual rate 88.2048111
2022-11-14_20_40_11_807: EngineState: LUA: actual rate 88.2053833
2022-11-14_20_40_11_808: EngineState: LUA: actual rate 88.2057342
2022-11-14_20_40_11_813: EngineState: LUA: actual rate 88.2061538
2022-11-14_20_40_11_815: EngineState: LUA: actual rate 88.2065963
2022-11-14_20_40_11_817: EngineState: LUA: actual rate 88.2069397
2022-11-14_20_40_11_916: EngineState: LUA: actual rate 88.2073364
2022-11-14_20_40_11_916: EngineState: LUA: actual rate 88.2073822
2022-11-14_20_40_11_916: EngineState: LUA: actual rate 88.2081451
2022-11-14_20_40_11_916: EngineState: LUA: actual rate 88.2084732
2022-11-14_20_40_11_917: EngineState: LUA: actual rate 88.2089080
2022-11-14_20_40_11_918: EngineState: LUA: actual rate 88.2090911
2022-11-14_20_40_11_919: EngineState: LUA: actual rate 88.2093887
2022-11-14_20_40_11_925: EngineState: LUA: actual rate 88.2100067
2022-11-14_20_40_11_926: EngineState: LUA: actual rate 88.2101974

```

mck1117, 2022-11-15:
One possible change to make is to schedule every N ms, instead of sleeping N ms after executing the script.

If the period is 10ms, but your script takes 3ms to execute, you'd actually get 77hz instead of the desired 100hz. We also run Lua on the lowest priority thread in the system, so as ECU load goes up, your script gets slower.

Using locals may help your script perf: for example replace `tpsSensor = Sensor.new("Tps1")` with `local tpsSensor = Sensor.new("Tps1")`. Without `local`, it's doing a lookup by name every time you access the value. With `local`, it's simply looking in a fixed (decided at compile time) stack slot.

rusefillc, 2022-11-16:
```
tpsSensor = Sensor.new("Tps1")
tpsSensor : setRedundant(true)

ppsSensor = Sensor.new("AcceleratorPedal")
ppsSensor : setRedundant(true)

Tps1Primary = Sensor.new("Tps1Primary")
AcceleratorPedalPrimary = Sensor.new("AcceleratorPedalPrimary")

setTickRate(100)
-- huh? with 100Hz tick rate why do we need such huge timeout?
ppsSensor : setTimeout(1000)
tpsSensor : setTimeout(1000)
restartEtb()

function setTwoBytes(data, offset, value)
	value = math.floor(value)
	data[offset + 2] = value >> 8
	data[offset + 1] = value & 0xff
end

-- CAN Rx Channels
IDRIVE_BUTTONS = 0x267
IDRIVE_ROTARY = 0x264
-- CAN Tx Channels
IDRIVE_BRIGHTNESS = 0x202
IDRIVE_HEARBEAT = 0x560
IDRIVE_ROTARY_INIT = 0x273
canRxAdd(IDRIVE_ROTARY) -- rotation
canRxAdd(IDRIVE_BUTTONS) -- buttons

R52_RPM = 0x316

data = { 0, 0, 0, 0, 0, 0 , 0, 0 }

rpm = 7000

i = 0

-- One time rotary initialisation

txCan(1, IDRIVE_ROTARY_INIT, 0, { 0x1D, 0xE1, 0x00, 0xF0, 0xFF, 0x7F, 0xDE, 0x04 })

clicks = 5

activity = Timer.new()

function onCanRx(bus, id, dlc, data)
	-- 	print('got CAN id=' ..id ..' dlc=' ..dlc)

	if id == IDRIVE_BUTTONS then
		-- 		print('got buttons @4=' ..data[4] ..' @6=' ..data[6])

		if data[4] == 1 or data[4] == 2 then
			print('got hold or release @4=' ..data[4] ..' @6=' ..data[6])
		end
	end

	if id == IDRIVE_ROTARY then
		if data[5] == 0x80 then
			clicks = data[4]
			print('rotary turned right ' ..data[4] ..' clicks')
			activity : reset()

		end

		if data[5] == 0x7f then
			clicks = 0
			print('rotary turned left ' ..255 - data[4] ..' clicks')
			activity : reset()

		end
	end

end

i = 0
setTickRate(100)
start = Timer.new()
start : reset()

activity : reset()

function onTick()
	i = i + 1
	sinceStart = start : getElapsedSeconds()
		print ("actual rate " ..(i / sinceStart) )

	-- wake up, need to send once every 1.5 second but we can send more often because why not
	txCan(1, IDRIVE_HEARBEAT, 0, { 3 })
	txCan(1, IDRIVE_BRIGHTNESS, 0, { 0xd })

	etbDisable = activity : getElapsedSeconds() > 5

--	setEtbDisabled(etbDisable)

	rpm = getSensor("RPM")

	if i == 3 then
		print('Postponed restart')
		restartEtb()
		-- 		i = 0
	end
	-- 	print('Period ' ..i)

	setTwoBytes(data, 2, rpm * 6.39)

	txCan(1, R52_RPM, 0, data)

	txCan(1, 0x61f, 0, { 0, 0, 0, 0, 0, 0 , 0, 0 })

	sentValue = getSentValue(1)
	-- 	tpsValue = interpolate(0.11, 0, 0.89, 100, sentValue)
	tpsValue = interpolate(0.11, 100, 0.89, 0, sentValue)

	if tpsValue == tpsValue then
		tpsSensor : set(tpsValue )
		Tps1Primary : set(tpsValue )
		-- 		print('sentValue ' ..sentValue..' tps ' ..tpsValue)
	else
		-- 		print('sentValue is Nan')
	end

	ppsSensor : set(15 * clicks)

end

```

```
2022-11-15_20_54_09_444: EngineState: LUA: actual rate 2.5019350
2022-11-15_20_54_09_778: EngineState: LUA: actual rate 2.5008774
2022-11-15_20_54_10_223: EngineState: LUA: actual rate 2.4998381
2022-11-15_20_54_10_670: EngineState: LUA: actual rate 2.4988324
2022-11-15_20_54_11_005: EngineState: LUA: actual rate 2.4978626
2022-11-15_20_54_11_450: EngineState: LUA: actual rate 2.4969215
2022-11-15_20_54_11_894: EngineState: LUA: actual rate 2.4960084
2022-11-15_20_54_12_231: EngineState: LUA: actual rate 2.4951069
2022-11-15_20_54_12_677: EngineState: LUA: actual rate 2.4941811
2022-11-15_20_54_13_125: EngineState: LUA: actual rate 2.4933510
2022-11-15_20_54_13_460: EngineState: LUA: actual rate 2.4925244
2022-11-15_20_54_13_908: EngineState: LUA: actual rate 2.4917662
2022-11-15_20_54_14_355: EngineState: LUA: actual rate 2.4909901
```

this is caused by disconnecting BMW CAN device

rusefillc, 2022-11-16:
@mck1117 I assume txCan method is blocking? Does txCan take forever with no devices on the CAN bus?

rusefillc, 2022-11-16:
here's one confusing code comment: why 100ms?

```
CanTxMessage::~CanTxMessage() {
    // 100 ms timeout
    canTransmitTimeout(&CAND1, CAN_ANY_MAILBOX, &m_frame, TIME_IMMEDIATE);
}
```

rusefillc, 2022-11-16:
I agree that actual canTransmitTimeout needs a timeout

I am just suggesting that Lua has it's own queue/thread/else not to block in such case?

Or let me state root cause: At the moment my ETB PID depends on if CAN is connected or not. How can I feed data into TPS at same high rate regardless of CAN connected or not?

rusefillc, 2022-11-16:
i guess I can do horrible logic of checking if CAN bus is alive only once per second?

mck1117, 2022-11-16:
> At the moment my ETB PID depends on if CAN is connected or not.

IMO Lua should not be in the loop for your ETB....

rusefillc, 2022-11-16:
Absolutely, Lua should not be in the loop of my ETB

At the moment my *prototyping* ETB PID depends on if CAN is connected or not.