rusefi/rusefi #4788: Please make gagues in Freedom units

ElDominio, 2022-11-16:
Temp, Pressure and Distance be available in Imperial units

mck1117, 2022-11-16:
This is a way huger task than we're willing to tackle right now, I think

rusefillc, 2022-11-16:
We used to have it on .ini side, it's just boring labor for each gauge manually right?

rusefillc, 2022-11-16:
I.e. if we can find commit which killed it we can re-enable for some gauges?

But scales for curves would stay in C :( do we have no cheap solution for for axises?

ElDominio, 2022-11-16:
But I had a client do it in the INI lol

[image]

[image]

mck1117, 2022-11-16:
> I think cheapest solution for axes is to build a coolant temp sensor archive (which we should have with the amount of tunes on rusefi online) and forget about writing in F

We already have presets in tunerstudio - trivial to add more if you have suggestions.

rusefillc, 2022-11-16:
?

I mean "cranking by clt" curve

rusefillc, 2022-11-21:
#607 would be a typical outcome :(

rusefillc, 2023-11-09:
https://www.youtube.com/watch?v=JYqfVE-fykk

rusefillc, 2024-02-06:
@ElDominio got a new complaint in Spanish

ElDominio, 2024-02-07:
[image]
 
another one

rusefillc, 2025-08-09:
@ElDominio we now have freedom temperature

@FDSoftware let's make progress in terms of freedom PSI pressure?

ElDominio, 2025-08-10:
Definitely showing freedom units:

[image]

FDSoftware, 2025-08-13:
> [@ElDominio](https://github.com/ElDominio) we now have freedom temperature
> 
> [@FDSoftware](https://github.com/FDSoftware) let's make progress in terms of freedom PSI pressure?

next step now that we have psi?

rusefillc, 2025-08-29:
@FDSoftware defect report https://github.com/rusefi/rusefi/issues/8521

rusefillc, 2025-08-30:
@FDSoftware time to close or do we still have any work pending?

ElDominio, 2025-08-31:
Will test tomorrow, please ping me at about 4pm

rusefillc, 2025-08-31:
@FDSoftware please address

```
AuxT1Gauge
AuxT2Gauge
ambientTempGauge
lowFuelPressureGauge
fuelTempGauge
acPressureGauge
egt1Gauge to egt8Gauge
```

FDSoftware, 2025-08-31:
@ElDominio 
> i tried copy pasting my MAP estimate:
> when i paste, i see this
> 
> are the Z values in PSI? no negative values so i dont think so. I switched back to metric and it works fine

The Z's in map-estimate are in PSI, but I think TS is not smart enough to handle the case of copying a table in KPA and pasting it over one in PSI.

> VE Y axis is still KPA?

Hmm, yep, good catch, I'm going to have to do some magic with bitStringValue to fix this case, since I only have to modify the scale/translate for kpa/psi but not for maf/%/lua/etc.

ElDominio, 2025-09-01:
> @ElDominio
>
> The Z's in map-estimate are in PSI, but I think TS is not smart enough to handle the case of copying a table in KPA and pasting it over one in PSI.

Is it absolute PSI or gauge? Imperial usually uses gauge

closed 2025-09-25 by rusefillc