rusefi/rusefi #4789: simulator - "WARNING: out-of-order coil off Coil 1" in snapshot_20221117_044326_mre_f4_rusefi

mi-hol, 2022-11-17:
snapshot_20221117_044326_mre_f4_rusefi
```
2022-11-17_11_20_36_541: ConsoleUI: COMPOSITE_OFF_RPM=300
2022-11-17_11_20_38_967: EngineState: Length should be positive
2022-11-17_11_20_38_968: EngineState: Emulating position sensor(s). RPM=1200
2022-11-17_11_20_39_007: EngineState: LUA loading script length: 21...
2022-11-17_11_20_39_012: EngineState: LUA script loaded successfully!
2022-11-17_11_20_39_018: EngineState: Stimulator: updating trigger shape: 2/1 55
2022-11-17_11_20_39_024: EngineState: WARNING: No MAP values
2022-11-17_11_20_39_028: EngineState: WARNING: out-of-order coil off Coil 1
```

snapshot_20220914_125452_mre_f4_rusefi does not show this warning

Maybe I'd need to make an additional setting?

mi-hol, 2022-11-17:
likely due to https://github.com/rusefi/rusefi/issues/3887 console changes to "not connected"
and TunerStudio (MS) Beta Lite!  v3.1.08.15  shows
[image]
Console
[image]

mi-hol, 2022-11-20:
issue persists with snapshot_20221117_045321_stm32f407_rusefi

mck1117, 2022-11-20:
does the engine sniffer look correct when you run the engine?

mi-hol, 2022-11-20:
> does the engine sniffer look correct when you run the engine?

not to understand what you mean, I can't run the real Benelli engine yet, therefore I tried with simulator only.
[image]

command used in powerShell 7.3:
PS C:\Program Files\rusEFI\snapshot_20221117_045321_stm32f407_rusefi\console> java -jar rusefi_console.jar

java process then outputs below message continuously
```
java.io.FileNotFoundException: logs\rusEFI_gauges_2022-11-20_23_41_46_115.msl (Access is denied)
        at java.base/java.io.FileOutputStream.open0(Native Method)
        at java.base/java.io.FileOutputStream.open(FileOutputStream.java:298)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:237)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:126)
        at java.base/java.io.FileWriter.<init>(FileWriter.java:66)
        at com.rusefi.sensor_logs.PlainTextSensorLog.startSensorLogFile(PlainTextSensorLog.java:48)
        at com.rusefi.sensor_logs.PlainTextSensorLog.startIfNeeded(PlainTextSensorLog.java:38)
        at com.rusefi.sensor_logs.PlainTextSensorLog.writeSensorLogLine(PlainTextSensorLog.java:90)
        at com.rusefi.sensor_logs.SensorLogger.lambda$init$0(SensorLogger.java:129)
        at com.rusefi.core.SensorCentral.setValue(SensorCentral.java:57)
        at com.rusefi.core.ISensorHolder.grabSensorValues(ISensorHolder.java:24)
        at com.rusefi.core.SensorCentral.grabSensorValues(SensorCentral.java:34)
        at com.rusefi.binaryprotocol.BinaryProtocol.requestOutputChannels(BinaryProtocol.java:607)
        at com.rusefi.binaryprotocol.BinaryProtocol$2$1.run(BinaryProtocol.java:224)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:829)
java.io.FileNotFoundException: logs\rusEFI_gauges_2022-11-20_23_41_46_120.mlg (Access is denied)
        at java.base/java.io.FileOutputStream.open0(Native Method)
        at java.base/java.io.FileOutputStream.open(FileOutputStream.java:298)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:237)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:126)
        at com.rusefi.sensor_logs.BinarySensorLog.writeSensorLogLine(BinarySensorLog.java:53)
        at com.rusefi.sensor_logs.BinarySensorLogRestarter.writeSensorLogLine(BinarySensorLogRestarter.java:48)
        at com.rusefi.sensor_logs.SensorLogger.lambda$init$0(SensorLogger.java:129)
        at com.rusefi.core.SensorCentral.setValue(SensorCentral.java:57)
        at com.rusefi.core.ISensorHolder.grabSensorValues(ISensorHolder.java:24)
        at com.rusefi.core.SensorCentral.grabSensorValues(SensorCentral.java:34)
        at com.rusefi.binaryprotocol.BinaryProtocol.requestOutputChannels(BinaryProtocol.java:607)
        at com.rusefi.binaryprotocol.BinaryProtocol$2$1.run(BinaryProtocol.java:224)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:829)
I 221120 234146.589 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: No MAP values
I 221120 234146.589 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: No MAP values
I 221120 234146.590 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: out-of-order coil off Coil 1
I 221120 234146.590 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: out-of-order coil off Coil 1
java.io.FileNotFoundException: logs\rusEFI_gauges_2022-11-20_23_41_46_668.msl (Access is denied)
```

mi-hol, 2022-11-20:
[image]

rusefillc, 2022-11-21:
I bet those are part of a known but undocumented false positive drama. I kid you not: I have a reference to this on my physical white board.

Just created a ticket since could not locate one #4815

mi-hol, 2022-11-21:
A "false positive" is a more cosmetic issue, the real issue is "no connection" for TunerStudio.
This issue prevents all activities to prepare a tune for my Benelli in advance :(

[image]

mi-hol, 2022-11-21:
> Let's please create a separate ticket and focus on communication issue.

there is already https://github.com/rusefi/rusefi/issues/3887
and I [attached all logs already here](https://github.com/rusefi/rusefi/issues/4789#issuecomment-1318534035).
Have they been analysed?

Happy to create an new issue but what else besides the provided infos would be needed in a new issue?

rusefillc, 2022-11-21:
#3887 is about never connecting to real hardware
@mi-hol I believe your issues is not very related to #3887 since yours is simulator (TCP vs serial) and you seem to be able to connect _somtimes_

Please create a separate ticket and attach logs there, I will probably edit this ticket later deleting unrelated content.

mi-hol, 2022-11-21:
> you seem to be able to connect _sometimes_

this is a false impression from my view.
Created https://github.com/rusefi/rusefi/issues/4817

rusefillc, 2022-11-21:
Oh I know why I am so confused! I am so confused because we are mixing rusEFI console via TCP with TS via virtual driver.

mi-hol, 2023-08-19:
should be fixed

closed 2023-08-19 by mi-hol