rusefi/rusefi #4791: simulator does not compile

rusefillc, 2022-11-18:
Did GHA just upgrade GCC? No? What's going on?

previous build

[image]

[image]

nmschulte, 2022-11-18:
FYI, `master` (717a51095c67964b99fdc828c650c7e463cd162b) fails to build w/ GCC 12:
```
Compiling hal_can_lld.cpp
simulator/can/hal_can_lld.cpp: In function ‘void can_lld_stop(CANDriver*)’:
simulator/can/hal_can_lld.cpp:143:14: error: ‘remove’ is not a member of ‘std’; did you mean ‘move’?
  143 |         std::remove(instances.begin(), instances.end(), canp);
      |              ^~~~~~
      |              move
make: *** [../firmware/ChibiOS/os/common/startup/SIMIA32/compilers/GCC/rules.mk:130: build/obj/hal_can_lld.o] Error 1
Simulator compilation failed
```

rusefillc, 2022-11-18:
@nmschulte that's separate from current github action, right?

nmschulte, 2022-11-18:
Yep, just trying to reproduce locally to diagnose -- working through GCC 9 and 11 now.

nmschulte, 2022-11-18:
I am able to build successfully with GCC 9, GCC 10, and GCC 11, and GCC 12 fails due to a different issue (`std::remove` is removed w/ the `-std=c++2a` option, I guess; maybe this can be more specific value if so).

However, I had to alter the Makefile to do so on my system, as it expects `gcc` and `g++` to be the correct binaries, ignoring any specified `CC`/`CXX`/`CPP`/`LD`, which I suppose is bad form.  No idea what's changed about the build env. except that `ubuntu-latest` must have changed which GCC is "default."

rusefillc, 2022-11-18:
@chuckwagoncomputing any idea where would GHA announce planned changes/changelog? At the moment we have no idea what's going on, bug or feature :(

nmschulte, 2022-11-18:
Note that the failing build is firmware, not simulator.

rusefillc, 2022-11-18:
@nmschulte huh? on github it says simulator fails?

[image]

mck1117, 2022-11-18:
In GHA do we specify explicit ubuntu version, or latest? we can try specifying previous version for now

nmschulte, 2022-11-18:
`ubuntu-latest` is specified, and between the two simulator runs it seems to have changed from `20.04` to `22.04`; I'll put up a PR about this after testing.

nmschulte, 2022-11-18:
https://github.blog/changelog/2022-11-09-github-actions-ubuntu-latest-workflows-will-use-ubuntu-22-04/

rusefillc, 2022-11-18:
@nmschulte so https://github.com/rusefi/rusefi/pull/4793 would be a work-around right?

Do we know root cause, what's special about latest ubuntu?

nmschulte, 2022-11-18:
Yes, #4793 should resolve the problem from what I've gathered and tested.

I still don't understand exactly what's off, but it seems the GHA `ubuntu-latest` change that seems to have landed; the only notable difference so far is that 22.04 has GCC 11 installed.  My tests w/ Docker and this version of Ubuntu haven't yet turned up anything useful.

nmschulte, 2022-11-19:
Alright, I was able to reproduce this issue finally.  The key is that the _Firmware at GHA_ workflow builds simulator with `OS="Windows_NT"`: MinGW.  This succeeds on Ubuntu 20.04, but not Ubuntu 22.04.  Building with GCC on Linux seems to not suffer at all.

chuckwagoncomputing, 2022-11-19:
That's so a binary is created that people can run on Windows.

rusefillc, 2022-11-19:
So OS="Windows_NT" is a functional requirement

and the second fun fact: more than once we saw Windows GCC settings being different from unix gcc behavior

nmschulte, 2022-11-19:
Yep; I was able to recreate the issue w/ the MinGW GCC 11 (triggered by that env var); I didn't test the changes at all though, just that the build succeeded.  It's interesting that only MinGW exposed the linker issues, though: not sure what's different about the incantation / configuration of MinGW vs regular GCC, or what changed between MinGW GCC 9 vs MinGW GCC 11.

rusEFI could use more unit and integration tests -- maybe good to throw a fuzzer and [memory] safety analyzer at it too (e.g. Valgrind)

rusefillc, 2022-11-19:
@nmschulte https://github.com/rusefi/rusefi/blob/a14345aaac72304c9f7068173a5e470fd4704f30/.github/workflows/build-unit-tests.yaml#L70

mck1117, 2022-11-20:
we also run unit tests & simulator with asan enabled - it's detected plenty of real bugs

PR #4793 (2022-11-18): GHA/build-firmware: pin agents to Ubuntu 20.04
resolves #4791
commit cb2477be (2022-11-18): GHA/build-firmware: pin agents to Ubuntu 20.04 (#4793)

resolves #4791
closed 2022-11-21 by rusefillc