rusefi/rusefi #4798: Remove/make injector duty limit configurable

ElDominio, 2022-11-18:
I have customers complaining about random cuts, after  seeing a VIDEO of injector duty, it goes over 100%, triggering the duty limit. Can we have an option that when duty is >96%, the injector outputs go static(stuck to ground without closing).

If the drivers can support this, I believe this is an easy way to let them lean out and blow their shit up, which is not our fault since the ECU is reporting that it is the current behavior.

rusefillc, 2022-11-18:
I wonder if those users are running a recent version with https://github.com/rusefi/rusefi/pull/4592 improvement

the https://github.com/rusefi/rusefi/pull/4592 with nice indicators was supposed to have helped?

ElDominio, 2022-11-18:
Indicators work, but people swear their fuel system is enough and the ECU is holding them back

ElDominio, 2022-11-26:
This is very problematic in cars where transient goes over for a split second and the limit stays engaged with the throttle at WOT (fuel is cut until car is no longer WOT). This is happening with a drift car when the user stabs the throttle close to rev limit and transient fueling makes duty go over 100% for a split second. Is there any progress on this? @mck1117 @rusefillc

rusefillc, 2022-11-26:
transient and short term 100% duty looks like a great argument to remove the limit

mck1117, 2022-11-26:
> transient and short term 100% duty looks like a great argument to remove the limit

there has been some extensive offline discussion about this - something like only engaging the limiter after some period of time is probably the right solution. Allow very brief over 100% moments, but still cut if you're far over 100% for a "long time".

It also needs some fixes to injection scheduling to happen first.

rusefillc, 2022-11-26:
> It also needs some fixes to injection scheduling to happen first.

absolutely, things would break if condition is just removed

ElDominio, 2022-11-26:
Again with my theorycrafts, couldn't the fueling algorithm just "lock" when injector duty reaches say 96%, stop calculating additional fuel and justkeep the duty cycle locked at 96%, while UI reports unreal duty? (tag: Norman says)

mck1117, 2022-11-27:
Even more evil is that we could display actual duty * 1.042 so that 96% duty displays as 100% 😈

But we shouldn't do that, there are real solutions here too.

ElDominio, 2023-04-01:
please jesus do something about this, every time i run into this at the dyno i waste valuable dyno time

rusefillc, 2023-04-01:
@ElDominio my estimation is that doing something is relatively expensive. It's NOT one line change, proper transition to 100% and back would require effort. If not done properly there is a chance to mess things up badly.

ElDominio, 2023-04-01:
Snip of me complaining on Slack
[image]

ElDominio, 2023-04-02:
a big warning in TS somewhere that says this limit has been reached would be swell, like the coolant temp warning that blacks out the entire screen. how easy to implement that?

ElDominio, 2023-07-29:
This just boned me in a parallel ECU.

Duty limit makes the engine go lean if the user requests more fuel than 100%, where the injectors controlled are supplemental (not the main ones), and it makes it go extremely lean.

mck1117, 2023-07-29:
add that to the list of reasons a piggyback ecu is a hack

ElDominio, 2023-07-29:
As true as that might be

People see videos like this
https://youtu.be/1mmhkPTe6F0

And buy an ECU, and this time around they bought it from me. I was tasked with giving them customer support (since the guy was tuning it on his own) because of this and ran into the issue.

rusefillc, 2024-01-08:
https://github.com/rusefi/rusefi/commit/6ff8e67acd4175ca22271d6472b875e8b1c1c1c4

closed 2024-01-08 by rusefillc

davepmo, 2024-12-23:
Hi Guys,
I'm a new user, so bear with me.  I have a Proteus V7 board running on an PNP adapter board that I made.  Application is an Arctic cat  1100 turbo snowmobile with factory staged injectors.  FYI, Thanks for adding staged injectors, this is working.  However, I'm running into this 100% duty cycle issue also.  I see in the forums this was discussed and marked as completed on Sept 10, 2023 but it's unclear what the fix was, or if there is some interaction with the staged injectors going 100%.

In tunerstudio the settings I see are maximum allowed duty cycle (I set to 110% in an attempt to get around this), and there is a timeout value which will only allow a max of 5 seconds.  Sure enough, if I go over 5 seconds it shuts them down. 

We race a "stock" class on stock turbo and fuel system.  So the wastegate is bolted shut for max boost and we run both sets of injectors at 100% all day long, and have been for years.  Everyone is. 

I need to be able to run both sets of injectors at 100% with no limitations.  Any suggestions on where to start to resolve this are welcome.

Thanks a lot,
Dave.
 
I see the code below, do I need to edit this and recompile to get around this?  Is there a better workaround?

	// Cut at 110% instantly
	engineConfiguration->maxInjectorDutyInstant = 110;
	// Cut at 96% after 0.5 second
	engineConfiguration->maxInjectorDutySustained = 96;
	engineConfiguration->maxInjectorDutySustainedTimeout = 0.5f;

ElDominio, 2024-12-23:
What is your calculated injector dc while running?

Modauser, 2024-12-23:
Hello ElDominio,
Thanks for the reply.  Here are more details:

I designed an engine simulator pcb and some software where I simulate the engine, and the software measures the injector pulse widths and timing at each load point from the OEM ECU at defined conditions.  So I have a full map of what the OEM ECU is doing, and I'm backing in the VE table in tunerstudio so that the injector pulse widths match at each load point OEM to Proteus.  I also am able to measure temp compensation, timing, and accelerator pump multiplier tables the same way.  

I've done this before many times and it works great, I have an almost complete copy of the OEM map into a standalone when done, and it only takes an hour or so to derive the VE load tables in a standalone, it's that automated.  The engine will run great and only needs minor tweaking on the dyno then in the wild afterwards.  FYI, I've tried to use this info to calc VE, but there are too many factors to get it dead on, turns out it is just faster to tweak the VE table by hand to match outputs OEM to standalone. 

This OEM map runs 100% in 10 or so load cells under max boost.

To specifically answer your question,   I have it set so it's trying to do 100%. I have not experimented with running slightly under, but I need all the fuel I can get at these load points.   I can understand if I had the VE table so it was commanding 200% duty cycle, it's not (I'll double check this when I get home).  I looked in the forums and found this was an issue, so I thought it might be faster asking for some help.   

 I just need some guidance so I can run 100% duty cycle without timing out.

Thanks again for the quick reply.  Any suggestions?

commit 26b351ba (2024-01-08): Remove/make injector duty limit configurable fix #4798