rusefi/rusefi #4801: Add Boost Control Tool Tips

OSprey65, 2022-11-19:
Under boost control PID, add something like the post below to make it more understandable. Also in boost control target table, make reference to the cell units being KPa.
[Re: Boost control tuning](https://rusefi.com/forum/viewtopic.php?p=45745#p45745)
[Post](https://rusefi.com/forum/viewtopic.php?p=45745#p45745) by [mck1117](https://rusefi.com/forum/memberlist.php?mode=viewprofile&u=683) » Fri Jul 01, 2022 5:39 am

The rusEFI values are all in understandable units:
P = % per kPa error
I = % per kPa-seconds
D = % per kPa/sec

So:
If the boost error is 5kpa, and your P is set to 2, it will make a correction of 10%.
If the boost error has been 5kPa for 2 second (10 kPa-seconds) and I is set to 0.1, it will make a correction of 1%
If the boost is changing at 10kPa/s and D is set to 0.5, it will make a correction of 5%.

Those three terms are then added together, and limited by the PID min/max settings.

That is then added to the value from the open loop table, and sent to PWM the valve.

OSprey65, 2022-11-19:
As a matter of fact, this should be done for the idle PID too, and while we're at it the alternator PID too. Basically give someone a rough idea of how the system should respond by looking at the PID values

rusefillc, 2022-11-20:
@OSprey65 legit proposal for PID tooltips improvement but unfortunately above trivial level of effort needed due to PID tooltips being shared between all PIDs, a change to how we handle meta info would be needed

"boost control target table, make reference to the cell units being KPa." is a related but separate proposal. For that can you please help by providing an example of a table/dialog where it's done better? i.e. what text are you looking to add where exactly?

rusefillc, 2022-11-20:
@ElDominio remember you wanted me to show the better approach to table comments?

OSprey65, 2022-11-20:
I'll make the boost KPa into a seperate issue with added specifics

OSprey65, 2022-11-20:
@rusefillc  Unfortunate about the PID tool tips, I think it'd be a worthwhile addition though as the PIDs are already difficult to work with as someone who's tuned racing drone PIDs for the past fes years and I imagine someone without any control in their background would greatly benifit from the descriptions of how they work

rusefillc, 2022-11-20:
Problem statement: PID is not for humans
Desired solution: PID should be hidden from humans, we shall have more amazing magic like ETB auto-tune

mck1117, 2022-11-20:
> PID should be hidden from humans, we shall have more amazing magic like ETB auto-tune

And unfortunately the world isn't there yet. We can at least make it easier for users.

rusefillc, 2022-11-20:
Just added help into Boost PID dialog https://github.com/rusefi/rusefi/commit/6c567d44b718ab1b3c5d2fa36b63e9e6cca79f1a

rusefillc, 2022-11-20:
See also https://github.com/rusefi/rusefi/issues/4803

OSprey65, 2022-11-20:
Awesomw work guys!