rusefi/rusefi #4803: Boost is it actually "open+closed"?

rusefillc, 2022-11-20:
https://rusefi.com/forum/viewtopic.php?p=45745#p45745 makes me wonder if we have an issue with option name

boost control mode does not say "open + closed"

[image]

like for instance idle control says

[image]

So is the forum wrong or is the mode name wrong?

mck1117, 2022-11-20:
Agree, boost should be "open + closed" like idle

rusefillc, 2022-11-20:
@ElDominio FYI

no logic change here, only UI mode name correction

closed 2022-11-20 by rusefillc

ElDominio, 2022-11-20:
Just saw, open loop boost duty is used as base then PID is applied on top? That's perfect.

Now, the base duty table should use target KPA on the Y axis 

a.k.a. for 150kpa you need 20% duty
for 180 kpa you need 27% duty
etc

mck1117, 2022-11-20:
> Now, the base duty table should use target KPA on the Y axis

Ehhhh. That also depends on throttle position. My car will make 200 kpa at both 60% throttle or 100% throttle, but it needs much more gate duty at 60%, so there isn't a clear mapping between gate duty and resulting MAP.

ElDominio, 2022-11-20:
I'm just saying because this is pretty much standard practice on most aftermarket ECU's i've worked with

[image]
[image]

commit dbfb52cd (2022-11-20): Boost is it actually "open+closed"? fix #4803