rusefi/rusefi #4812: Release firmware link goes to NOT FOUND

BeerMoneyMotorsports, 2022-11-21:
link used from 96/97 hellen wiki "release firmware"  https://github.com/rusefi/rusefi/releases/download/20221117_release/rusefi_bundle_hellenNA8_96.zip

rusefillc, 2022-11-21:
Looks like bad manual release upload

https://github.com/rusefi/rusefi/releases/tag/20220914_release says 44 files

https://github.com/rusefi/rusefi/releases/tag/20221117_release says 16 files

mck1117, 2022-11-21:
It looks like in fact the latest release is broken - 2/3 of builds are just missing?

mck1117, 2022-11-21:
besides, did we test that pre-release on real car(s) before shipping?

rusefillc, 2022-11-21:
Missing files were added

@mck1117 my process is to announce RC and promote into release couple of days later unless I hear reasons. With the resources I have that's the best I can do.

Something something LTS, something something no one is getting paid, something something real life.

closed 2022-11-21 by rusefillc