rusefi/rusefi #4814: provide checksum/crypto-hash of release images

nmschulte, 2022-11-21:
To mitigate the concern of corrupt (or malicious) rusEFI images being flashed to devices, provide alongside the downloads checksums or cryptohashes (a la SHA512SUM or SHA256SUM files) so users can verify their images (`.zip`s).

A convenient mechanism to validate the images would go a long way as well ...
- a simple script to run `sha512`
- a more complex integration into the flashing process to verify

As well, it sounds a good idea to verify the user experience about the actual flashing is sufficient to inform users of failed flashes, and to ensure the need for irrecoverability of the currently flashed image (e.g. a disclosure/notice if one doesn't exist; an easy to opt-in automatic read/backup of the existing image; an A/B style boot ... probably not feasible given space constraints, but worth considering).

rusefillc, 2022-11-21:
So new files into full bundles and matching files into "autoupdate" files? https://rusefi.com/build_server/autoupdate/

And then later validate local files prior to firmware update?

please https://xyproblem.info/ :)

nmschulte, 2022-11-21:
> please https://xyproblem.info/ :)

I tried to cover that in the first line:

> To mitigate the concern of corrupt (or malicious) rusEFI images being flashed to devices

It's not clear what the actual problem encountered was, but one hypothesis is that the downloaded `.zip` was corrupted, or its contents corrupted after unzipping, or the flashing went awry.

Providing a hash alongside the `.zip`s covers the first scenario: user thinks they've obtained a correct (reliable), trusted*, `.zip`, but actually it's not: corrupted yet somehow working.  This checksum needs to be supplied _alongside_ the `.zip`, to allow verifying the `.zip`.  As well, by providing it alongside the `.zip`, it provides some trust that the `.zip` you've obtained is from the source you believe it is -- providing the file _inside_ the `.zip` doesn't afford that, as anyone can bundle a corresponding has with their (potentially malicious) `.zip`.

Second, providing a hash/checksum of the actual flash ROM _inside_ the `.zip` allows to verify and avoid corrupted ROMs after unzipping (e.g. bad file system, failing hard disk, w/e).

Third, validating the UX about what happens when a flash of a ROM fails flashing verification should be done, to ensure if something goes awry the user is made aware, and can re-flash, or flash a backup (which may be an automatic opt-in feature; to e.g. create a backup before flashing).

mck1117, 2022-11-21:
> Providing a hash alongside the .zips covers the first scenario

Corrupt zip that unzips properly is extremely unlikely, and I don't really expect any real users to actually check this

nmschulte, 2022-11-21:
> Corrupt zip that unzips properly is extremely unlikely, and I don't really expect any real users to actually check this

FWIW, I always check this when downloading software to run, but I agree most users probably don't [know to] care.  This would protect against someone distributing malicious images should that ever occur, though.

rusefillc, 2022-11-21:
Autoupdate zip files are unnzipped by rusEFI code not manually but I agree, does not sound realistic

Can we please take trust/signing out of scope?

Sounds like we agree on hash/checksum of the actual flash ROM inside the .zip

Sounds like validating the UX about what happens when a flash of a ROM fails is a related but separate ticket

nmschulte, 2022-11-21:
Fair enough, but the trust is something I would appreciate.

I think starting at the bottom, checking into the UX about failures is kosher and helps the user recover and avoid bad flashes.

mck1117, 2022-11-21:
#3 is by far the most useful/likely case, the bits were probably delivered to the user's PC correctly, but mis-flashed to the ECU for whatever reason