rusefi/rusefi #4816: 4chan fails to auto-calibrate specific VAG ETB

rusefillc, 2022-11-21:
4chan rev D with JLC assembled TLE9201
start with default tune - ETB is oscillating trying to shut down or something? PCB lights are cycling, looks like some sort of shutdown of either 6A questionable wall power supply or TLE9201 protection? Same questionable wall power supply is happy to drive a much larger Ford SENT ETB - log of oscillation https://rusefi.com/online/view.php?log=1101

Now pressing "auto calibrate" getting un unhappy TPS calibration https://rusefi.com/online/view.php?msq=1310 log https://rusefi.com/online/view.php?log=1102

[image]

[image]

rusefillc, 2022-11-21:
Oh and ETB does open completely during auto-calibration

also I have at least two units of same kind, here's another piece

[image]

mck1117, 2022-11-21:
I think video would be handy here

rusefillc, 2022-11-21:
@mck1117 https://youtu.be/Pwmy6Y5feR4

rusefillc, 2022-11-26:
Same exact behavior with Proteus

mck1117, 2022-11-26:
Do you have a TS log? We should be logging the raw voltage.

mck1117, 2022-11-26:
> weird not inverted TPS calibration here? with a 4 wire TPS it's not like I could wire things in a wring way?

“not inverted” is pretty normal, lots of throttles have both signals with the same sign but different slope magnitude and offset.

rusefillc, 2022-11-26:
> Do you have a TS log? We should be logging the raw voltage.

Two 4chan logs linked all the way on top, do you mean Proteus logs? I sure can grab those but behavior is 200% the same exact.

rusefillc, 2022-11-27:
This issue here is a combination of #4832 and shitty power supply where power supply was shutting down before TLE9201 would overload.
[image]

[image]

@mck1117 feel free to close unless you have ideas for improvements?

mck1117, 2022-11-27:
One suggestion for the humans (not the code) is to bench test things like this with a small lead acid battery in parallel with your PSU while testing. For example set PSU to 13v, max current limit, with lead acid in parallel. The PSU will keep the battery fully charged, and the battery will supply the big current peaks necessary for ETB stuff when the PSU can't keep up/supply enough. I think I used one of the alarm/UPS style batteries like this: https://www.amazon.com/ExpertPower-EXP1270-Rechargeable-Lead-Battery/dp/B003S1RQ2S

closed 2022-11-27 by mck1117

rusefillc, 2022-11-27:
> One suggestion for the humans (not the code) is to bench test things like this with a small lead acid battery 

thank you, ordered!