rusefi/rusefi #4817: TunerStudio Lite can not connect to simulator (virtual serial)

mi-hol, 2022-11-21:
[image]

rusefillc, 2022-11-21:
windows 3.11? windows 10? redhat?

what virtual serial driver are you using?

mi-hol, 2022-11-21:
[image]
[image]

rusefillc, 2022-11-21:
RPM gauge shows 1200 so it was able to get RPM info so it was able to connect

looks like it's loosing connection at some point?

rusEFI console logs are not very relevant here, TS logs could provide extra value

overall virtual simulator option anything is low priority since proper way is to use commercial edition of TS with TCP/IP connector

mi-hol, 2022-11-21:
is this the log required?
Did not see a Help topic regarding "Get support in case of issues"
[TunerStudioAppDebug.txt](https://github.com/rusefi/rusefi/files/10059562/TunerStudioAppDebug.txt)

rusefillc, 2022-11-21:
TunerStudioAppDebug.txt is the right file name but looks like attached file is size zero. This might happen if TS is still running while [TunerStudioAppDebug.txt](https://github.com/rusefi/rusefi/files/10059562/TunerStudioAppDebug.txt) was being uploaded? Maybe try closing it.

Also please "Comm Debug Log" checkbox is helpful

[image]

mi-hol, 2022-11-21:
Create New Project
[image]
[image]

[image]

DebugPackage:
[TunerStudioDebugPackage.zip](https://github.com/rusefi/rusefi/files/10059848/TunerStudioDebugPackage.zip)

rusefillc, 2022-11-21:
Does progress bar moves slowly?

maybe try increasing timeout from 3000 to 15000

in your file like C:\Users\Dell2019\Documents\TunerStudioProjects\dron_mre_f4\projectCfg (with your project name)

edit mainController.ini file

change 3000 to 15000 in
```
	blockReadTimeout	= 3000; Milliseconds general timeout
```

mi-hol, 2022-11-21:
> Does progress bar moves slowly?

I'm lacking experience to make a judgement.
What time in seconds would be fast versus slow? 

>  change 3000 to 15000

```
;	blockReadTimeout	= 3000; Milliseconds general timeout
	blockReadTimeout	= 15000; Milliseconds general timeout
```
changed progress bar speed to show ~ every 3% but got stuck at 72% and shows still same error just after a longer waiting time
[image]

rusefillc, 2022-11-21:
#794 might be relevant

mi-hol, 2022-11-21:
no luck with
```
;	blockingFactor = 750 ; max chunk size
	blockingFactor = 256 ; max chunk size
```
[TunerStudioDebugPackage.zip](https://github.com/rusefi/rusefi/files/10060948/TunerStudioDebugPackage.zip)

rusefillc, 2022-11-21:
We need to go around this issue :(

options are
* commercial TS
* some stm32 real hardware dev board
* another virtual serial driver maybe?

mi-hol, 2022-11-22:
> We need to go around this issue :(

one last check please, is the configuration of virtual serial (VS) correct?
the VS logs contain:
```
11/21/2022 11:03:37 PM: Demo (COM100): Virtual port COM100 is opened
11/21/2022 11:03:37 PM: Demo (COM100): Virtual Port: PARITY=None, STOPBITS=1, DATABITS=8
11/21/2022 11:03:37 PM: Demo (COM100): Virtual Port: 0x1, 0x80000040, 0x800, 0x200
11/21/2022 11:03:37 PM: Demo (COM100): Virtual Port: HANDSHAKE = Unknown
11/21/2022 11:03:37 PM: Demo (COM100): Virtual Port: BAUDRATE=115200
11/21/2022 11:03:40 PM: Demo (COM100): Write settings of remote port failed
11/21/2022 11:03:57 PM: Demo (COM100): Virtual port COM100 is closed
11/21/2022 11:18:24 PM: Demo (COM100): Connection was lost localhost:29002
```
is **HANDSHAKE = Unknown** correct?