rusefi/rusefi #4818: Full Optimized – High Speed

rusefillc, 2022-11-22:
We shall support "Full Optimized – High Speed" TS mode

mck1117, 2022-11-22:
What is full optimized mode? I looked in to all the modes enumerated in the PDF and if you're logging lots of fields (like we are), the mode we're using is most efficient.

rusefillc, 2022-11-22:
``11.2.3 Full Optimized – High Speed`` yes only helpful while focusing on a small subset

mck1117, 2022-11-22:
What scenario do we think this gives an advantage?

mck1117, 2022-11-22:
Remember that we're unlikely to exceed 1khz over USB due to protocol limitations.

rusefillc, 2022-11-22:
I believe @dron0gus wants to explore it for Bluetooth for primary and/or wideband

rusefillc, 2022-11-24:
Gauges refresh rate more than doubled with java protocol simulator TcpServerSandbox.java and TCP/IP connector

java protocol simulator, TCP/IP connector, without scatteredGetEnabled

[image]

java protocol simulator, TCP/IP connector, with scatteredGetEnabled

[image]

rusefillc, 2022-11-24:
@dron0gus current size is 22Kish and protocol limitation for offsets is 8K :(

mck1117, 2022-11-24:
> @dron0gus current size is 22Kish and protocol limitation for offsets is 8K :(

Aren’t those offsets in to the output channels though, which are only ~1k?

rusefillc, 2022-11-24:
Oooops, my mistake

dron0gus, 2022-11-28:
[image]
AlphaX over JDY33.
Should Console@USB and TS@BT work simultaneously?

rusefillc, 2022-11-28:
> Should Console@USB and TS@BT work simultaneously?

yes absolutely, I would expect any combination of connectors to work in parallel