rusefi/rusefi #4821: Nissan: probably hardware issue but maybe we need firmware progress as well

rusefillc, 2022-11-23:
rawPPS primary and secondary show different level of noise

q: do we want software to respond in any different way or just should address relevant RC filter?

[image]

relevant log https://rusefi.com/forum/viewtopic.php?p=46624#p46624

Qwerty-OFF, 2022-11-23:
Not same
[image]

mck1117, 2022-11-23:
> want software to respond in any different way

https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem Says software cannot help us, and we MUST fix the hardware.

rusefillc, 2024-05-01:
many 350z are happy, closing

closed 2024-05-01 by rusefillc