rusefi/rusefi #4823: VE Blend tables is an ignition adder?

ElDominio, 2022-11-23:
[image]

Can I add VE or Ignition here? If it is ignition, why is it under VE?
If it's VE, why can i write negative numbers?

[image]

I  suppose "bias" is voltage and "param" is how much the blend table is blended to main table? if so, "voltage" and "blend factor" is a better name to use

mck1117, 2022-11-23:
First bit is probably a typo if it in fact opens a dialog that then says VE. 

You can put negative values in the table because maybe whatever you’re doing makes the engine less efficient: suppose it’s VVT, so in part of the operating range it removes VE, and part it adds.

ElDominio, 2022-11-23:
I'm an idiot lol

PR #4835 (2022-11-27): ve blend typo, explanation on dialog
fix #4823
closed 2022-11-27 by rusefillc