rusefi/rusefi #4825: Toyota 36-2 with 4-1 cam trigger problems

ElDominio, 2022-11-25:
Car runs fine semi-sequentially, but once cam is activated RPM's go crazy.

[image]

Sorry for low speed logging the car is in the other side of the island and i'm waiting for my shop, once I have it i'll try to get the car in person

https://rusefi.com/online/view.php?log=1110

mck1117, 2022-11-25:
will take a look once back in the country, if you can grab a logic analyzer trace once the car is nearby that would be super helpful

ElDominio, 2022-11-29:
i'll try to get a scope as soon as i get the new shop

rusefillc, 2022-11-29:
> the new shop

Shall we all it "Maserati shop"?

ElDominio, 2023-01-09:
[displaylogictrue 36minus2.zip](https://github.com/rusefi/rusefi/files/10377190/displaylogictrue.36minus2.zip)

crank to run
36-2, crank not inverted, cam inverted

setting crank to inverted causes issues while starting, but there seems to be an extra tooth in trigger logger. setting to crank invert to TRUE after start gets rid of the extra tooth

VVT position going crazy

ElDominio, 2023-01-13:
Connected jumper cable between sensor ground and VR-

Signal is significantly cleaner, but not very reliable.
Scopes of this at 
[bridgeoscope.zip](https://github.com/rusefi/rusefi/files/10414013/bridgeoscope.zip)

We will remove this bridge and run a seperate CMP+/- wire tor esolve issues, but I think VVT errors were due to sharing the VR

Also, noticed I duplicated this issue #4930

ElDominio, 2024-02-23:
definitely not RusEFI's fault. i hate VR sensors sometimes

closed 2024-02-23 by ElDominio