rusefi/rusefi #4826: Target AFR table Y axis stuck at 0

BeerMoneyMotorsports, 2022-11-26:
When updating firmware, an issue was found where the Target AFR tables "dot" does not move up or down the y axis.  It is stuck at zero.  It will follow X axis as expected.  Below is screenshot where Y axis is MAP and should be near 100.

Old Firmware
snapshot_20221011_031129_hellenNA6_rusefi
New Firmware
snapshot_20221117_044339_hellenNA6_rusefi

Firmware downloaded from https://github.com/rusefi/rusefi/wiki/Hellen64-Miata-NA6-94

[image]

BeerMoneyMotorsports, 2022-11-26:
I confirmed that firmware that it has happened at some point between this _snapshot_20221011_031129_hellenNA6_rusefi_ and this _snapshot_20221109_195113_hellenNA6_rusefi_

rusefillc, 2022-11-26:
Does "fuel: target AFR" gauge work?

BeerMoneyMotorsports, 2022-11-26:
Reads 0

[image]

mck1117, 2022-11-26:
I disagree with that title change - the "fuel: target AFR" gauge does NOT have anything to do with TS rendering the dot on that table. The dot gets controlled by `afrTableYAxis`. BOTH fields seem to be broken, which isn't really surprising since they're set next to each other in code.

rusefillc, 2022-11-26:
@mck1117 looks like a lot of concurrent edits are happening

BeerMoneyMotorsports, 2022-11-26:
Quick bench test shows improvement.  I will confirm on the lemon.  
[image]

rusefillc, 2022-11-26:
@mck1117 do you by any chance understand what commit already in master has fixed what? because I do not. Was it https://github.com/rusefi/rusefi/commit/23698be5f869f4204242944026e7de036f39c57c or https://github.com/rusefi/rusefi/commit/55062b1ef6d35e7b9c9480520acc8d1b0513d9d0 or else?

mck1117, 2022-11-26:
It looks like somehow `fuel_computer_s* getLiveDataAddr()` was returning nullptr

rusefillc, 2022-11-26:
@mck1117 hold on, you do not know how exactly we have fixed that?

Damn :(

rusefillc, 2022-11-27:
Time to close at least this one in light of https://github.com/rusefi/rusefi/commit/2b51f3798b2dbbbb409e42d23871653d0d6f3f55 ?

mck1117, 2022-11-27:
> @mck1117 hold on, you do not know how exactly we have fixed that?

> Time to close at least this one in light of https://github.com/rusefi/rusefi/commit/2b51f3798b2dbbbb409e42d23871653d0d6f3f55 ?

I'd like to actually debug why it was failing the old way once I'm at a real desk. Currently at 35000 feet/535mph/somewhere over the southern tip of greenland, and unable to attach a debugger to the simulator/ECU and figure out what was actually wrong.

rusefillc, 2022-11-27:
that's a crowded place no way to focus!

[image]

mck1117, 2022-11-27:
The real problem is that the sim doesn't run on a mac ☹️

rusefillc, 2022-11-27:
you have JNI unit tests?

rusefillc, 2022-11-27:
Dad mode: https://github.com/rusefi/rusefi/issues/4843

rusefillc, 2022-11-29:
@mck1117 do you agree that it's time for a release?
@mck1117 do you agree that we shall close this prior to a release?

mck1117, 2022-11-30:
> I'd like to actually debug why it was failing the old way once I'm at a real desk.

HAHAHAHAHAHA

the problem is that the `fragments` array is constructed before the `fuelComputer` of `Engine` was set. So of course it's null.

closed 2022-11-30 by mck1117