rusefi/rusefi #4832: ETB overheats due to constant isTpsError true/false/true/false jitter

rusefillc, 2022-11-27:
Proteus
default settings
VAG ETB
https://rusefi.com/online/view.php?log=1111

evil combination over-heats TLE9201 on Proteus while consuming around 5A

[image]

rusefillc, 2022-11-27:
@mck1117 @ElDominio we can definitely detect that condition of isTpsError jitter

Question: how do we handle it? Turn throttle off for 10 seconds? what if we are in the middle of a turn? take whatever TPS value and apply it for next 1 second? else?

ElDominio, 2022-11-27:
I downloaded the log, I can see isTpsError  is 1 for maybe 0.01 seconds at a time; is duty being set to 0 when this happens? On most ETB systems I have seen, errors are summed and when it goes over a certain amount of errors for a certain amount of time (or even, only if the error state is true for more than a second), THEN ETB is considered failed and ETB duty set to zero, probably for this reason

mck1117, 2022-11-27:
what if we just change the defaults so they're invalid for all throttles? Correct me if I'm wrong, but the loop we get here is

1. default throttle calibration is sometimes valid for this throttle
2. ECU tries to control throttle
3. driver chip overheats
4. user is unable to autocalibrate because chip may be overtemp'd
5. user stuck with bad defaults

mck1117, 2022-11-27:
>  evil combination over-heats TLE9201 on Proteus while consuming around 5A

How did you power Proteus in this scenario?

ElDominio, 2022-11-27:
But i can do calibration just fine? Or are you talking about PID calibration?

ElDominio, 2022-11-30:
Hello, this is to report that the problem was indeed PID settings, and i used vales that seem logical (P close to 1, i close to 10-15, D close to .1) and the TLS is cool to the touch. I suggest PID autocal uses old vales to start, so it is more of an aid than a "we can do this automatigically"

But in the end, the problem is no more. Thanks everyone

mck1117, 2022-11-30:
can you create a log of good/bad PIDs?

rusefillc, 2022-11-30:
@ElDominio here we have "ETB overheats due to constant isTpsError true/false/true/false jitter: this seems obvious enough to disable ETB at least on non running engine" which is NOT something you have experienced

PID drama would be https://github.com/rusefi/rusefi/issues/4833

ElDominio, 2022-11-30:
Very bad PID settings -> https://rusefi.com/online/view.php?log=1103 ; https://rusefi.com/online/view.php?msq=1312

Working PID (driver is only warm to the touch) -> https://rusefi.com/online/view.php?msq=1321

rusefillc, 2022-11-30:
it works! https://rusefi.com/online/view.php?log=1113

closed 2022-11-30 by rusefillc

nmschulte, 2023-01-02:
FWIW, while I was testing Toyota ETCS-i / partial redundant TPS, I ran into a situation where the ETB was trying to command a TPS value that it could not achieve, and caused huge current draw.  I don't recall the exact situation, but it had to do with a certain `tpsSecondaryMaximum` (partial redundancy cut-over) causing a stepwise discrepancy when switching to partial redundancy, preventing the TPS from ever reading the commanded value.  If I re-create it I'll share more detail, but wanted to capture it here for now.  Thinking now, I _don't_ think the TPS was erroring in my case, however.

commit 421ac16f (2022-11-30): ETB input jitter protection fix #4832