rusefi/rusefi #4839: warning in trigger_input_adc.h

mck1117, 2022-11-27:
Probable bug due to use of an uninitialized field:

```
In file included from ../firmware/hw_layer/digital_input/trigger/trigger_input_adc.cpp:11:
../firmware/hw_layer/digital_input/trigger/trigger_input_adc.h:43:23: warning: field 'adcDefaultThreshold' is uninitialized when used here [-Wuninitialized]
        float adcThreshold = adcDefaultThreshold;
                             ^
../firmware/hw_layer/digital_input/trigger/trigger_input_adc.cpp:14:31: note: in implicit default constructor for 'TriggerAdcDetector' first required here
/*static*/ TriggerAdcDetector trigAdcState;
                              ^
../firmware/hw_layer/digital_input/trigger/trigger_input_adc.h:19:7: note: during field initialization in the implicit default constructor
class TriggerAdcDetector {
      ^
```

mck1117, 2022-11-27:
Not totally clear what the intent was here, so assigned to @andreika-git instead of fixing.

nmschulte, 2022-11-27:
I think it's unrelated, but know that I touched this module while correcting issues exposed by MinGW GCC 11: https://github.com/rusefi/rusefi/pull/4799/files#diff-21bd59bb8a80929bc76a4dbb4d2496c75007f7f60ef012f4cdc4dd22405eb90a

mck1117, 2022-11-27:
@nmschulte that looks ok, the offending line is older than your PR

nmschulte, 2023-01-06:
I think these warnings have gone away?