rusefi/rusefi #4843: sim doesn't run on a mac

rusefillc, 2022-11-27:
@mck1117 says that sim doesn't run on a mac

rusefillc, 2022-11-27:
@mck1117 does it not compile or does it not run? is that a ChibiOS problem or rusEFI problem?

mck1117, 2022-11-27:
nowhere close to compiling

I wouldn't mind "macos sim" being simply out of scope - I'm not sure we really care

rusefillc, 2022-11-27:
@nmschulte @chuckwagoncomputing any interest? Worst case scenario this could be investigated using a github action.

nmschulte, 2022-11-27:
I don't have any Apple hardware so cannot (will not) help.

However ... possibly the current sim can run in a container on MacOS?

mck1117, 2022-11-27:
A container would probably work, but again, what do we think the use case is here? How much do we actually care?

Running in a container would exclude its use for any “regular” user, limiting its use to the one weirdo (me) who sometimes does rusEFI dev work on a Mac mostly while traveling.