rusefi/rusefi #4847: Auto-reboot after DFU software upload

rusefillc, 2022-11-27:
I've noticed forever ago but cannot find ticket

Some boards would auto-reboot after DFU firmware upload and some would not, what the heck?!

For instance same exact 4chan firmware, two boards from two different assembly batches, same hellen MCU-144-0.6 module, same stm32f407zgt6 part number just from different factories

[image]

rusefillc, 2022-11-27:
black board auto-reboots after DFU

[image]

rusefillc, 2022-11-27:
green board does not auto-reboot

[image]

dron0gus, 2022-11-28:
CCM RAM?
[image]

rusefillc, 2022-11-28:
@dron0gus says to try
```dfu-util --alt 0 --download rusefi.dfu --reset```
since it might print bootloader version

dron0gus, 2022-11-28:
Black board. Chip label is the same as on @rusefillc photo:
```
$ dfu-util --alt 0 --download deliver/rusefi.dfu -s :leave
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
file contains 1 DFU images
Please note that the next version of dfu-util will automatically
set alternate interfaces based on the DfuSe file images!
parsing DFU image 1
Target name: EncedoKey
image for alternate setting 0, (1 elements, total size = 567764)
parsing element 1, address = 0x08000000, size = 567756
Erase   	[=========================] 100%       567756 bytes
Erase    done.
Download	[=========================] 100%       567756 bytes
Download done.
Done parsing DfuSe file
Transitioning to dfuMANIFEST state
```
Reboots after flashing. 
TODO: add `-s :leave` to `flash_dfu.sh`

Simple test and way to get bootloader version: restart to DFU mode with buttons and run:
```
$ dfu-util --alt 0 -v -s :leave
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.21 (11156)
dfu-util: Cannot open DFU device 20b1:000a
dfu-util: Cannot open DFU device 20b1:000a
dfu-util: Cannot open DFU device 0a5c:21e6
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
**DFU mode device DFU version 011a**
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Memory segment at 0x08000000   4 x 16384 = 65536 (rew)
Memory segment at 0x08010000   1 x 65536 = 65536 (rew)
Memory segment at 0x08020000   7 x 131072 = 917504 (rew)
DfuSe command mode
Transitioning to dfuMANIFEST state
```

dron0gus, 2022-11-28:
Some MRE with STM32F427VGT6
```
$ dfu-util --alt 0 --download deliver/rusefi.dfu -s :leave
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
dfu-util: Patching string descriptor 4 length (was 166, received 112)
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
file contains 1 DFU images
Please note that the next version of dfu-util will automatically
set alternate interfaces based on the DfuSe file images!
parsing DFU image 1
Target name: EncedoKey
image for alternate setting 0, (2 elements, total size = 515404)
parsing element 1, address = 0x08000000, size = 12168
Erase   	[=========================] 100%        12168 bytes
Erase    done.
Download	[=========================] 100%        12168 bytes
Download done.
parsing element 2, address = 0x08008000, size = 503220
Erase   	[=========================] 100%       503220 bytes
Erase    done.
Download	[=========================] 100%       503220 bytes
Download done.
Done parsing DfuSe file
Transitioning to dfuMANIFEST state
```
Reboots and blinks.

rusefillc, 2022-11-28:
@dron0gus do you have any boards which do not auto-reboot?

dron0gus, 2022-11-29:
Linux scripts were fixed to run application after flashing https://github.com/rusefi/rusefi/pull/4851

dron0gus, 2022-11-29:
@rusefillc nope. Just tested MRE R0.4.7-2L with STM32F407 - also flashed and starts blinking.
BUT I'm under linux and using dfu-util tool.