rusefi/rusefi #4859: Better ETB duty ROC length default

rusefillc, 2022-12-01:
@ElDominio please play with the "rate of change averaging length" parameter while having good and bad PID settings

looking for some logs and looking to decide what's the best length and if we can identify "good" and "very bad" thresholds?

(Rota typo already fixed)

[image]

[image]

rusefillc, 2022-12-01:
This is follow-up to #4833

ElDominio, 2022-12-02:
Today I tested a bunch of values, and I think something around 50 is good. It allows you to se momentary spikes well since they don't get averaged over time (usually TunerStudio's "ghost" arrow will show the spike), while giving you an idea if you're close to being correct

rusefillc, 2022-12-02:
Plz dome logs?