rusefi/rusefi #4861: Grab digital inputs through LUA

ElDominio, 2022-12-01:
LUA can grab analog inputs, but pins declared as digital don't show in the LUA Analog Inputs (as they shouldn't), but I wanted to grab a switch state from LUA to do stuff. How can I do this?

rusefillc, 2022-12-01:
At the moment we do not have this

We have not relevant https://github.com/rusefi/rusefi/wiki/Lua-Scripting#getdigitalindex

We have horrible option https://github.com/rusefi/rusefi/wiki/Lua-Scripting#readpinpinname which would be ugly

ElDominio, 2023-01-03:
Nope sorry

andreika-git, 2023-07-09:
We have at least three options IMHO:

1) use the existing getDigital(0..7) function and the new "Lua Digital Inputs" config dialog - and thus break compatibility with current getDigital() implementation which returns the state of 4 pre-defined inputs (clutch up/down, break pedal and AC button)

2) use the existing getDigital(4..11) function - but leave the first 4 indices as reserved for existing pre-defined inputs, and rename the added configurable aux inputs to "Digital #4"..."Digital #11", in order to support both old and new behavior.

3) add a new lua function getAuxDigital(0..7) similar to existing getAuxAnalog(), and maybe rename TS dialog to "Lua Aux Digital Inputs" (we already have a name inconsistency with 'aux' word for analog inputs)?

@ElDominio @rusefillc Can you please decide which one is better?

rusefillc, 2023-07-09:
Let's go with option 3

rusefillc, 2023-07-11:
@andreika-git please update Lua-Scripting wiki page?

rusefillc, 2023-11-22:
@ElDominio seems to work

closed 2023-11-22 by rusefillc