rusefi/rusefi #4862: Working disconnect button in Console

ElDominio, 2022-12-01:
Put a disconnect/connect button in Console so we can switch between Tunerstudio and Console without having to reopen console every time.

mck1117, 2022-12-01:
why not close it

ElDominio, 2022-12-01:
Closing it means having to open, connect every time (about 15 seconds)

A disconnect button keeps the program open and ready to just connect, we save one click and a few alt tabs of having to go to Explorer to click it

rusefillc, 2022-12-01:
#220 is technically kind of close
also "about 15 seconds" was recently improved a bit

Not saying that "disconnect" is not an option - just saying that it's a pile of small improvements and root cause is the fucking "switch between console and TS back and forth" :(

ElDominio, 2023-04-10:
please; switching between console and TS for writing Lua scripts is very aggravating

ElDominio, 2023-07-28:
please this is so bad when writing stuff

commit aadf3cf1 (2023-07-29): Working disconnect button in Console fix #4862
closed 2023-07-29 by rusefillc