rusefi/rusefi #4864: Get ETB target position in Lua

ElDominio, 2022-12-02:
Can there be a sensor type for this? So I can alter final TPS with math from Lua. 

Basically add TagetTPS to this https://github.com/rusefi/rusefi/blob/master/firmware/controllers/sensors/sensor_type.h

mck1117, 2022-12-02:
do you want the unadjusted number, or the number after lua adjustment?

ElDominio, 2022-12-02:
before, so it doesn't cause weird feedback loops lol (lua adds, then grab modified, adds again, keeps adding forever)

mck1117, 2022-12-02:
> before, so it doesn't cause weird feedback loops

That's what I thought, but I figured I'd ask anyway 🙂

rusefillc, 2022-12-03:
@ElDominio thank you for your interest in advanced rusEFI Lua

Here is what I think is the best way to do it: https://github.com/rusefi/rusefi/wiki/Lua-Scripting#getoutputname

Step 1: make sure that you see a live data point matching your desire, create a ticket to add a live data point

Step 2: we need https://github.com/rusefi/rusefi/issues/4867 and https://github.com/rusefi/rusefi/issues/4866

Step 3: @ElDominio make a video explaining all this magic Live Data including! https://github.com/rusefi/rusefi_documentation/issues/258

Bonus point: https://github.com/rusefi/rusefi/issues/4868

ElDominio, 2024-02-23:
idk i probably figured out a way to do what i wanted without this

closed 2024-02-23 by ElDominio