rusefi/rusefi #4865: Epic: daily drive Nissan

rusefillc, 2022-12-02:
I would love to start running RE while I am driving around for errands

rusefillc, 2022-12-02:
Depends on https://rusefi.com/forum/viewtopic.php?f=4&t=2468

mck1117, 2022-12-02:
reminder to me to upload the nissan WBO board (I do indeed have it)

rusefillc, 2022-12-02:
@mck1117 please upload the Nissan WBO board :)

mck1117, 2022-12-05:
> @mck1117 please upload the Nissan WBO board :)

https://github.com/mck1117/denso-wideband

rusefillc, 2024-06-29:
Depends on https://github.com/rusefi/rusefi/issues/2278

rusefillc, 2024-06-29:
TCU subset

[image]

rusefillc, 2024-06-29:
rusEFI so close

```
ID=505,Type=D,Length=8,Data=20001A5E
ID=561,Type=D,Length=8,Data=E08000E0D4C34C9E
ID=563,Type=D,Length=8,Data=79A200180E000001
ID=573,Type=D,Length=8,Data=0000000E01380079
ID=574,Type=D,Length=8,Data=0040FF4500D600A2
ID=593,Type=D,Length=8,Data=10000001006F1052
ID=595,Type=D,Length=8,Data=014AFF
ID=721,Type=D,Length=7,Data=0084000031F801
ID=734,Type=D,Length=8,Data=0F080200196507A8
ID=1361,Type=D,Length=8,Data=7DDB00A0000280FF
ID=1408,Type=D,Length=8,Data=008240
```

rusefillc, 2024-06-29:
```
ID=505,Type=D,Length=8,Data=20001A5E
ID=561,Type=D,Length=8,Data=E08000E0D4C34C9E
ID=563,Type=D,Length=8,Data=79A200180E000001
ID=573,Type=D,Length=8,Data=0000000E01380079
ID=574,Type=D,Length=8,Data=0040FF4500D600A2
ID=593,Type=D,Length=8,Data=1000000100EF1230
ID=595,Type=D,Length=8,Data=0351FF000000B102
ID=721,Type=D,Length=7,Data=0084000031F801
ID=734,Type=D,Length=8,Data=0F080200196507A8
ID=1361,Type=D,Length=8,Data=7DDB00A0000280FF
ID=1408,Type=D,Length=8,Data=008240
```