rusefi/rusefi #4874: 2JZ 12+1 Trigger issues at ccranking speeds

ElDominio, 2022-12-06:
https://rusefi.com/online/view.php?log=1114

In this log, you can see syncgapratio is very erratic while starting, while it is close to 1 when running (no missing tooth)

What is this due to? this makes the car very hard to start.

https://rusefi.com/online/view.php?log=1115

In this log, it never started and the RPM was always reading incorrectly.

ElDominio, 2022-12-06:
https://rusefi.com/online/view.php?log=1118 sync loss at upper RPM

mck1117, 2022-12-06:
why is the VR threshold so high?

[image]

ElDominio, 2022-12-06:
I set it to 1 and it still has problems, I scoped it and the voltage was easily over 1.5v in idle;

Can voltage go DOWN with RPM?

mck1117, 2022-12-06:
There is an analog filter that will reduce amplitude somewhat with RPM, but it shouldn't do a whole lot.

I'd try setting it to 0.3v or something (or slightly more than the minimum required to avoid trigger errors at idle) and see what happens.

rusefillc, 2022-12-06:
@mck1117 do we need a ticket to actually implement the ADC3 analog_input logic?

mck1117, 2022-12-06:
> @mck1117 do we need a ticket to actually implement the ADC3 analog_input logic?

#4876

mck1117, 2022-12-09:
analog trigger pattern looks very good

[image]

mck1117, 2022-12-09:
@ElDominio is going to try and get digital logic analyzer of cam+crank inputs to ECU when back home

mck1117, 2022-12-13:
onboard trigger scope says both crank+cam should be set "invert = true"

[image]

rusefillc, 2023-03-03:
@ElDominio did https://github.com/rusefi/alphax-4chan/issues/85 help? Can we test with improved hardware?

ElDominio, 2024-02-23:
god I hate VR part 3. Probably fixed with better conditioning

closed 2024-02-23 by ElDominio