rusefi/rusefi #4888: yaml file less hard work: template for pin ID

rusefillc, 2022-12-11:
in our .yaml files we seem to often use pin ID in ts_name

```
  - pin: E31
    class: outputs
    function: MIL Check Engine
    ts_name: E31 Check Engine
```

all boards are somewhat same so new yaml is created from older sections of existing boards

Would be great to reduce manual labor

can we use template for pin ID to reduce manual labor?

```
  - pin: E31
    class: outputs
    function: MIL Check Engine
    ts_name: --- Check Engine
```

rusefillc, 2022-12-11:
@chuckwagoncomputing what is the level of effort here? Is there a chance in December or not really?

chuckwagoncomputing, 2022-12-11:
This is what https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/skeleton is for, is it not?

rusefillc, 2022-12-11:
even on the skeleton we have "13" twice

```
  - pin: 13
    id: D6
    class: outputs
    color: yellow
    function: Alternator (see JP2 jumper note)
    ts_name: 13 - GP Out 6
    type: gp_high
```

chuckwagoncomputing, 2022-12-16:
Given that the template exists, I suggest the title of this issue be changed to "Connector template YAML has problems" or something like that.

rusefillc, 2022-12-16:
I guess what I am asking for is https://en.wikipedia.org/wiki/Template_processor? The mental test I apply is "do I have to type same thing twice or not". In this case pin ID seems to be something I type twice again and again

Definitely not a show stopper

chuckwagoncomputing, 2023-01-01:
I'm not sure I'm understanding. Could you provide a short example (e.g. 1 pin worth) of template file, input file, and output file?
Example:
template
```
id: E14
class: event_inputs
ts_name: Digital 3
type: din
```
input
```
id: E14
function: Digital trigger/switch input for instance Start Button
```
output
```
id: E14
class: event_inputs
ts_name: Digital 3
type: din
function: Digital trigger/switch input for instance Start Button
```

rusefillc, 2023-01-02:
the main use case is having pin ID at the beginning of ts_name

at the moment that could only be achieved by manually synching pin ID with ts_name prefix

suggestion is for https://github.com/rusefi/rusefi/commit/0ecb3746e79b43d0f5e066437c5f144157d773a5 different input to provide same output as before, i.e. ``___`` to be automatically replaced by pin id

chuckwagoncomputing, 2023-01-02:
What about just automatically putting the id in front of every ts_name, and not requiring the `___`? Are there cases where we don't want this?

rusefillc, 2023-01-02:
On the one hand overall yes just having ID in front of every ts_name looks reasonable

On the other hand who knows what corner case we are forgetting about. Also what about all existing files, would we have ID twice until we (manually?) get all 12+ files to new convention?

closed 2023-02-12 by rusefillc