rusefi/rusefi #4896: Toyota ETCS-i APS (ETB APP) [and TPS]

nmschulte, 2022-12-15:
_First, according to the documentation and references, this [APS] situation isn't intended in normal/stock operation, so this issue serves at the moment to simply document the situation: manual actuation on the bench, beyond the designed/intended [APS] ranges.  However, TPS suffers similarly!_

Toyota's ETCS-i (ETB) APS (APP) outputs behave similarly to some (e.g. certain Ford; https://github.com/rusefi/rusefi/blob/master/firmware/controllers/sensors/redundant_ford_tps.cpp) TPS outputs: one signal reaches max before the other when actuated across the entire range.  In this [bench] situation, the VPA2 (redundant/secondary APP) signal reaches max (~5V) before VPA (primary APP).  This causes rusEFI to signal an error (again, when using the full mapping/range, on the bench).

Additionally, the ETCS-i's TPS output (VTA1 and VTA2) behaves similarly, though this is a typical situation.  This would be ideal to support via rusEFI.

[image]

I will check soon (<24hr hopefully) the APS output ranges in-vehicle, though the documentation notes the typical/allowable range.  Ideally rusEFI would like to support this situation as well.

----

[image]
[image]

----

_Proteus 0.6 - rusEFI 2022.12.08.proteus_f7.snap_1513_

mck1117, 2022-12-15:
Here's a log with raw voltages of "actuated via "hand pedal" until stop, then opened the plate by hand, fluttered it a bit, released back to stop, and then released "hand pedal" to closed"

[2022-12-15_02.11.02.mlg.zip](https://github.com/rusefi/rusefi/files/10240017/2022-12-15_02.11.02.mlg.zip)

mck1117, 2022-12-15:
I believe @nmschulte is going to try and reproduce that log with a real throttle cable/pedal/car soon, as that log was taken on the bench, possibly actuating the PPS further than is possible in the real car.

mck1117, 2022-12-18:
[image]

closed 2023-01-10 by rusefillc