rusefi/rusefi #4899: RusEFI console keeps resetting on latest

ElDominio, 2022-12-16:
[image]

I keep seeing "Not Connected" pop up for a microsecond, I can't upload Lua stuff either.

mck1117, 2022-12-16:
@nmschulte sees this too I think

ElDominio, 2022-12-16:
Reverting to 12-01 fixed the issue FYI

rusefillc, 2022-12-16:
probably antilag livedata, need to make this area more robust

nmschulte, 2022-12-17:
`git bisect` reveals 52884c64cb531081db2c3937f8ee29c5de6d128a (_SD card tooth log (#4897)_) is the commit causing my issue on Proteus (F7, v0.6), but it seems this code-path is skipped if not using "trigger" (SD) vs "normal" (mgl) setting.  The board seems to stop communicating until reset, however, TS seems to not suffer and my [limited] tuning experience seems to proceed without issue:

```
I 221217 035049.404 [AWT-EventQueue-1] IniFileModel - Section [Tools]
I 221217 035049.427 [AWT-EventQueue-1] BinaryProtocolServer - BinaryProtocolServer on 2390
I 221217 035049.429 [AWT-EventQueue-1] StatusConsumer - ServerSocket 2390 created. Feel free to point TS at IP Address 'localhost' port 2390
debug mode 22
I 221217 035051.579 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035052.581 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035053.582 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035054.578 [communication executor1] IncomingDataBuffer - composite log header: timeout 5000ms. Got only 0byte(s) while expecting 2
I 221217 035054.583 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035055.583 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035056.584 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035057.585 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035058.586 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035059.578 [communication executor1] IncomingDataBuffer - text header: timeout 5000ms. Got only 0byte(s) while expecting 2
I 221217 035059.581 [communication executor1] ConnectionWatchdog - ConnectionWatchdog.reconnectTimer restarting: 1000
I 221217 035059.582 [Commands Queue] BinaryProtocol - Sending [set date 2022-12-17T03:50:49.289]
I 221217 035059.588 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035059.589 [communication executor1] SerialIoStream - ttyACM0: Closing port...
I 221217 035059.617 [text pull1] BinaryProtocol - Port shutdown: Stopping text pull
I 221217 035059.805 [communication executor1] SerialIoStream - ttyACM0: Closed port.
I 221217 035059.806 [communication executor1] LinkManager - restart isPortAvailableAgain=true
I 221217 035059.807 [communication executor1] LinkManager - LinkManager: Starting ttyACM0
I 221217 035059.807 [communication executor1] StreamConnector - StreamConnector: connecting
I 221217 035059.808 [communication executor1] StreamConnector - scheduleOpening
I 221217 035059.808 [communication executor1] StreamConnector - scheduleOpening>openPort
class com.rusefi.io.LinkManager$5: Opening port: ttyACM0
I 221217 035059.808 [communication executor1] BufferedSerialIoStream - [BufferedSerialIoStream] openPort ttyACM0
I 221217 035101.590 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035102.591 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
I 221217 035103.592 [AWT-EventQueue-1] ConnectionWatchdog - restart already pending...
```

----

1) It seems the switch to mailbox `fetch` w/ `TIME_INFINITE` timeout in `GetToothLoggerBuffer` never times out, but switching this back to non-blocking `fetchI` doesn't resolve the issue, likely because ...
2) `sdTriggerLogger` infinite-loops over the call anyway.

I guess using "normal"/MLG logging `sdTriggerLog which avoids the code path avoids the issue, but I guess "trigger"/SD is the default.  As well, I haven't found how to set these values or issue commands (like `date`) from TS, and so this makes me stuck.  I guess the first defined value in `rusefi_config.txt` is the default, and switching the order would mitigate the issue until the logic is fixed.

mck1117, 2022-12-17:
Trigger logging mode should not be the default, and besides, that SD card thread will never run if you don't have a card present.

rusefillc, 2022-12-17:
text pull seems broken on real firmware
text pull works in simulator

https://github.com/rusefi/rusefi/commit/2e98d31671ac5f7758f6ff0bb173461901bd78d2 is my attempt to add HW CI

@mck1117 is it reasonable to suspect some changes around shared big buffer?

rusefillc, 2022-12-17:
Damn HW CI does not seem to fail

yet when I execute under IDEA locally issues seem to be around text pull :(

mck1117, 2022-12-17:
Big buffer didn't touch text pull. Is it an issue with whatever console does with `TS_GET_COMPOSITE_BUFFER_DONE_DIFFERENTLY`? I think that's the only part I touched.

rusefillc, 2022-12-17:
Yes, at the moment I am focusing on TS_GET_COMPOSITE_BUFFER_DONE_DIFFERENTLY

rusefillc, 2022-12-18:
@mck1117 
```
msg_t msg = filledBuffers.fetch(&buffer, TIME_INFINITE);
```

mck1117, 2022-12-18:
oops it blocks - let me PR fix

PR #4903 (2022-12-18): nonblocking for TS, blocking for SD card
fix #4899
closed 2022-12-18 by rusefillc