rusefi/rusefi #4900: There should be seperate setBoostAdd and setWgDutyAdd functions

ElDominio, 2022-12-17:
Changing target without changing base duty makes it depend only on PID, besides I use open loop a lot without PID. Have seperate functions  to control separate parts of boost control.

rusefillc, 2022-12-19:
@ElDominio please help me understand you.

Are you looking to add brand new functions?

Are you looking to remove any functions?

Are you looking to change behavior of any of the current functions? what functions exactly would you like to change?

ElDominio, 2022-12-19:
Add new function that modifies only duty cycle, not boost target

rusefillc, 2022-12-19:
Renaming setBoostAdd to setBoostTargetAdd
Renaming setBoostMult to setBoostTargetAdd

Adding setBoostDutyAdd