rusefi/rusefi #4910: Accel enrich activating while trying to control ETB

ElDominio, 2022-12-19:
https://rusefi.com/online/view.php?log=1130

This log shows when i rev and let go, since TP moves quite quickly while the control loop is trying to control throttle (for a split sec) you can see injection time going to extreme values for a split second, exactly at 1.7s, 2s, 6s, and 6.4s. This is making the car have a rich spot on decel

mck1117, 2022-12-19:
I only see TPS AE from the initial throttle kick:

That yellow blip is 0.4 seconds, exactly as configured on the accel enrich page.

[image]

ElDominio, 2022-12-19:
Interesting, then why is my injection time jumping to about 18ms on decel? I see that there is no tps fuel added

ElDominio, 2023-01-02:
[image]

This is what I mean. Why is flow rate spiking so massively?