rusefi/rusefi #4914: Reduce scratchBuffer size

dron0gus, 2022-12-26:
Livedata is divided in fragments.
No need to copy ALL fragment to `scratchBuffer` at once and transfer in one shot.
So `scratchBuffer` need to be as big as biggest fragment.
```
send header
while (fragment) {
 copy to scratchBuffer
 incremental crc
 send
 fragment++
}
send tail (crc)
```

rusefillc, 2022-12-26:
Same scratchBuffer used for all other commands which rely on larger size in order to have simpler code?

dron0gus, 2022-12-27:
Actually only LiveData requires such a big buffer.
Also some other users do unnecessarily copy to this buffer and one `write` to `tsChannel` instead of few `write`s from source.

dron0gus, 2022-12-27:
Dreaming about stm32h7 and its MDMA controller with chained list support and scatter/gather support.

mck1117, 2022-12-27:
> Dreaming about stm32h7 and its MDMA controller with chained list support and scatter/gather support.

It's just not worth it - too much complication so such a tiny performance gain.

mck1117, 2022-12-27:
> Actually only LiveData requires such a big buffer.

We could increase the blocking factor in TS to the size of a full page, then handle the required buffer size internally. Not a bad idea.

dron0gus, 2022-12-27:
> It's just not worth it - too much complication so such a tiny performance gain.

LiveData request can be handled almost without CPU intervention.

mck1117, 2022-12-27:
> LiveData request can be handled almost without CPU intervention.

There's no way to automatically CRC multiple blocks like that. The CPU still has to walk the linked list. You could save on the actual copy with DMA, but you could do that on an STM32F4 too.

It's also just not very much CPU. Have you run the perf tracer to look at how much time we're spending handling it?

mck1117, 2022-12-27:
using the hardware DMA controller would be the greatest opportunity for improvement by far.

dron0gus, 2022-12-27:
> using the hardware CRC controller would be the greatest opportunity for improvement by far.

https://github.com/rusefi/rusefi/pull/2909 :)

dron0gus, 2022-12-27:
BTW, USB (at least bulk) has some data delivery and integrity protection. Is there any way to disable CRC in TS protocol?

rusefillc, 2022-12-27:
One disables crc by editing .ini

mck1117, 2022-12-27:
> On F103 gathering livedata reply using highSpeedOffsets

Wellllll....highSpeedOffsets has to walk the table for every field that gets copied in, instead of once for all values. Pretty obvious performance bottleneck there.

mck1117, 2022-12-27:
@dron0gus is there an actual performance improvement we're trying to find on the rusEFI side, or are we hunting for optimization for the sake of optimization?

dron0gus, 2022-12-27:
@mck1117 some optimization was done for F103.
While working on F103 I found few places where optimization can be done for RE code.
Currently there is no need for this.
But this is where we can save some RAM. Also sending livedata by fragments will reduce peak CPU load.

So this issue is just a king of reminder.

mck1117, 2022-12-27:
Did some looking at wideband issue....

> On F103 gathering livedata reply using highSpeedOffsets created so much CPU load so in rev3 heater stops oscillating. :)

On wideband the tunerstudio thread is currently at `NORMALPRIO + 1`, which is way too high...

mck1117, 2022-12-27:
Some performance metrics, on F4 @ 168mhz, for the first 750 bytes of output channels (TS command with offset 0 count 750):

|operation|microseconds|
|----|----|
|`copyRange(scratchBuffer + 3, getLiveDataFragments(), offset, count);`|35|
|`uint32_t crc = crc32(&scratchBuffer[2], size + 1);`|60|
|`write(reinterpret_cast<uint8_t*>(scratchBuffer), size + 7, true);` and `flush();`|15-25|

Total cost to respond to the command is ~110-120us, so even if TS is requesting at 1khz, that spends only ~15% CPU, and it's on a very low priority thread.

dron0gus, 2022-12-27:
This is more about:
`#define BLOCKING_FACTOR 750`
and
`char scratchBuffer[BLOCKING_FACTOR + 30];`
for each `TsChannel`
High CPU load is on F103@48MHz.