rusefi/rusefi #4915: virtual MSD with .ini in fork

rusefillc, 2022-12-27:
@TurboMarian is wondering how to get MSD with .ini working on https://github.com/TurboMarian/rusefi custom boards

mck1117, 2022-12-27:
just generate config and it works?

rusefillc, 2022-12-27:
Let's see how https://github.com/rusefi/rusefi/commit/dd6b3de294fd3ef065e1b90cd2f0b99580c9d972 goes

rusefillc, 2022-12-27:
@TurboMarian note how https://github.com/rusefi/rusefi/commit/20e0f6c3468e069b82383e6a4e1a104c240f4427 adds .ini files based on https://github.com/rusefi/rusefi/commit/dd6b3de294fd3ef065e1b90cd2f0b99580c9d972 I believe that's what you are missing on your branch

TurboMarian, 2022-12-27:
Thank you guys. I done that but still virtual drive is blank (disk 2). Disk 3 is SDCARD in ECU.

[image]

mck1117, 2022-12-27:
Did you run the gen_config script for your board before building the firmware?

mck1117, 2022-12-27:
If the drive is there but blank, it means you didn’t run the script.

rusefillc, 2022-12-27:
@TurboMarian please push into https://github.com/TurboMarian/rusefi for us to be less blind

mck1117, 2022-12-27:
Like this, you should see content added to ramdisk_image.h

[image]

TurboMarian, 2022-12-27:
Thank you guys!
[image]

closed 2022-12-27 by rusefillc