rusefi/rusefi #4918: Mitsubishi 4g93 in trigger list but not in wiki?

ElDominio, 2022-12-29:
What kind of trigger pattern is expected when you choose this?

rusefillc, 2022-12-29:
@ElDominio what do you think about removing that trigger and renaming Miata NA to mention that it's same as Mitsu?

```
	// todo: this really looks to be same as Miata_NA shall we remove?
	TT_MITSUBISHI = 11,

```

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#mazda-miata-na

ElDominio, 2022-12-29:
Maybe call it Miata NA/4G63

ElDominio, 2023-01-02:
4G18 is not an engine; why was the title changed to that?

Difference between Miata NA/4G63 and 4G93 is that 4G93 is separate Cam/Crank sensors, Miata NA is a CAS unit. I think this is enough reason to keep them separate @rusefillc

rusefillc, 2023-01-02:
Google knows 4g13 and for whatever reason we have a reference to 4g18 in the source code I cannot explain that and it's not the most important issue

I see that https://github.com/rusefi/rusefi_documentation/blob/master/Images/triggers/trigger_TT_MITSUBISHI.png image is wrong - circles do not match linear shape

compare to https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#mazda-miata-na

ElDominio, 2023-01-02:
https://rusefi.com/online/view.php?log=1135 -> log of what's happening, it runs for a split second and has trigger error. Changing to 4G69 trigger has perfect syncgapratio but RPM = 0

[4g93start.zip](https://github.com/rusefi/rusefi/files/10333658/4g93start.zip) -> logaic analyzer log

[osclog.csv](https://github.com/rusefi/rusefi/files/10333660/osclog.csv) -> ts oscope

rusefillc, 2023-01-03:
@ElDominio 

[image]

does the file open for you? using what version of what app?

ElDominio, 2023-01-03:
[image]

rusefillc, 2023-01-03:
yes works with 1.2.40!

rusefillc, 2023-01-03:
@ElDominio this is now two separate tickets mixed together

glad to see logic analyzer!

you seem to describe two separate cranking attempts with different settings but only one log?

https://rusefi.com/online/view.php?msq=1352 shows crank as primary input and cam as seconday

[image]

please clarify if physical wires are flipped (c3-crank goes to cam or crank? c7-cam goes to cam or crank on actual vehicle?) since cam has to be primary input

ElDominio, 2023-01-03:
Pretty sure we wired crank to crank+(c3) and cam to c7 and i flipped it in settings

rusefillc, 2023-01-03:
I am confused. Please confirm what tune matches what log. Is this a case of tune not matching log?

ElDominio, 2023-01-03:
I don't really remember but I do know flipping it didn't work. 

The only setting that has at least tried to fire up is Miata NA, both inverted, with Primary set to C7 and Secondary set to C3. But it loses sync quickly and shuts off (which is the behavior seen in the most recent log)

rusefillc, 2023-01-03:
trigger documentation now show gaps in light grey for instance "gap 1.90"

[image]

[image]

note that one is waiting for only one gap and the other waits for two. note the gap sequence

please post links to four logs
mitsu trigger, cam not inverted
mitsu trigger, cam inverted
miata NA trigger, cam not inverted
miata NA trigger, cam inverted

secondary channel inverted or not would be a separate story, for now let's focus on top cam wheel since that's the one where we see gaps nicely

I guess for secondary inverted or not we will be troubleshooting instant RPM @mck1117 ?

rusefillc, 2023-01-03:
PS: look at logs 
[image]

ElDominio, 2023-01-03:
using mitsu trigger
[2022-12-29_14.04.55.mlg.zip](https://github.com/rusefi/rusefi/files/10339982/2022-12-29_14.04.55.mlg.zip)

syncgapratio is content but no RPM

mck1117, 2023-01-03:
> syncgapratio is content but no RPM

syncgapratio is NOT content - look at the sync parameters on the image

[image]

@ElDominio try these parameters in trigger gap override, if the engine starts, please record SD trigger log of cranking+start+run so we can hardcode better trigger shape

[image]

rusefillc, 2023-01-03:
@mck1117 you seem to be suggesting miata gaps but note the difference between images - mitsu has gaps painted only on falls
```
void initializeMitsubishi4g18(TriggerWaveform *s) {
	s->initialize(FOUR_STROKE_CAM_SENSOR, SyncEdge::Fall);
```

vs

```
void initializeMazdaMiataNaShape(TriggerWaveform *s) {
	s->initialize(FOUR_STROKE_CAM_SENSOR, SyncEdge::Both);
```

mck1117, 2023-01-03:
Whatever the gaps are, look at the log - they don't match the number in the image, but WILL sync with the miata numbers (should mitsu also be rising edge??)

rusefillc, 2023-01-03:
miata numbers are "BOTH" numbers. miata "BOTH" numbers would not work for mitsu "Fall" would they?

mck1117, 2023-01-03:
https://rusefi.com/online/view.php?log=1135 shows some gap ratios, are those not the real gap ratios measured on the real engine?

ElDominio, 2023-01-04:
That log was made on real engine yes

ElDominio, 2023-01-04:
Engine does not turn on with those settings, it seems to try but the moment it does it loses sync and never runs

rusefillc, 2023-01-04:
I am glad to be right!

Now that you've executed @mck1117's suggestion please consider my request for data as mentioned above

mck1117, 2023-01-04:
> it seems to try but the moment it does it loses sync and never runs

grab a log of that?

rusefillc, 2023-01-04:
Related to #4935

ElDominio, 2023-01-04:
> > it seems to try but the moment it does it loses sync and never runs
> 
> grab a log of that?

It's  in the "miatanainverted.mlg" but it only shows 204RPM (which it did go over)

rusefillc, 2023-01-05:
these two look like winners to me

[image]

[image]

note how one is inverted another is not. either we have magic or trigger definitions are inverted.

@ElDominio please confirm that timing light was used to confirm TDC location. I wonder if bad TDC was causing engine to kick

rusefillc, 2023-01-05:
[image]

[image]

look like loosers?

ElDominio, 2023-01-05:
> these two look like winners to me
> 
> [image]
> 
> [image]
> 
> note how one is inverted another is not. either we have magic or trigger definitions are inverted.
> 
> @ElDominio please confirm that timing light was used to confirm TDC location. I wonder if bad TDC was causing engine to kick
> 

Timing only confirmed on miata NA/INVERTED

mck1117, 2023-01-05:
@rusefillc you can look at `triggerToothAngleError` to see if teeth are arriving at the correct time

ElDominio, 2023-01-06:
@mck1117  @rusefillc  anything else I need to get?

rusefillc, 2023-01-06:
@ElDominio what's the current problem statement? does the car start/run/happy?

ElDominio, 2023-01-06:
start yes
run no
happy definitely no

i have no more requests for logs so idk what else to do

rusefillc, 2023-01-06:
plz call?

why does it not run in your opinion? do you have a log where it cranks and starts? why does it not start/run in the logs which look happy in terms of trigger sync?

For your information without tunes I was assuming that ignition/injection was off or something. As is the process is pretty frustrating :( Let's brainstorm how can we make this less spread out in time. Just the logs are not giving at least me the full enough picture, please consider posting to youtube more often. No, pixelated whatsapp videos would not be same :(

rusefillc, 2023-01-07:
[image]

[image]

ElDominio, 2023-01-07:
[4g93startnosparkplugs.zip](https://github.com/rusefi/rusefi/files/10367134/4g93startnosparkplugs.zip)

ElDominio, 2023-01-07:
Runs on 4G93 only cam

Issues on 4G93 both fronts

log with crank inverted
[4g63 both front only crank invert.zip](https://github.com/rusefi/rusefi/files/10367297/4g63.both.front.only.crank.invert.zip)

log with crank not inverted
[4g63 both front no invert.zip](https://github.com/rusefi/rusefi/files/10367298/4g63.both.front.no.invert.zip)

log with crank not inverted, with injectors enabled
[4g93bothfront none inverted.zip](https://github.com/rusefi/rusefi/files/10367299/4g93bothfront.none.inverted.zip)

ElDominio, 2023-01-10:
fun fact:

this only works for revE 4chan, not any previous one

rusefillc, 2023-01-10:
https://rusefi.com/online/view.php?log=1143

i see crazy gap ratio I suspect noise. open question is 4chan hardware is problem or not, unfortunately unable to get in touch with @mck1117 on #4656 

[image]

my plan:
* I ship more connector parts to @ElDominio asking to make Proteus and 8chan (!!!) harnesses to isolate HW issues
* I will see if I can revive #4656

rusefillc, 2023-01-22:
Let me recap
* Still a major concern for @ElDominio 
* We have since merged https://github.com/rusefi/rusefi/pull/4656 @ElDominio did we test with that firmware fix/hack?
* We suspect HW digital input specific to hall (which probably technically also affects 2chan? but maybe progress in 8chan), we have two vectors of progress here
* Either make a Proteus or 8chan harness
* or try https://github.com/rusefi/alphax-4chan/issues/72 capacitor mods

@mck1117 FYI

rusefillc, 2023-01-23:
I think we have changes so much it's time for a new ticket with only current status

closed 2023-01-23 by rusefillc